platformio/platform-espressif32 #1428: Unable to inspect memory

italocjs, 2024-07-21:
My code was unable to build duo iram overflow (happens on one cpu, not on another, same code).  i trie using project inspection but it returns "invalid params".

How can i fix it? any tips on fixing iram issues? i dont use it on my code, is probably something hidden by esp-idf.  i currently use nimble, but thats all.

[image]

```
[env:esp32dev]
build_type = debug
debug_build_flags = -Os
test_framework = googletest
test_ignore = test_native
platform = espressif32
board = simovatrack130
framework = arduino, espidf
    
board_build.partitions = partitions.csv
build_flags = 
    -D USE_ESP32_ENVIRONMENT
    ; -Wall                ; Enable all warnings
    ; -Wno-error           ; Ensure warnings are not treated as errors

upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
    esp32_exception_decoder
    send_on_enter

lib_deps = 
    adafruit/Adafruit Unified Sensor
    adafruit/Adafruit BusIO
    adafruit/Adafruit ADXL343
    https://github.com/italocjs/DFRobot_LIS.git
    https://github.com/italocjs/geofencing_esp32.git
    https://github.com/italocjs/NuS-NimBLE-Serial
    mprograms/QMC5883LCompass
    adafruit/RTClib
    mikalhart/TinyGPSPlus
    bblanchon/ArduinoJson
```

italocjs, 2024-07-21:
this is a successful verbose build, maybe its helpful
[verbose build.txt](https://github.com/user-attachments/files/16326385/verbose.build.txt)

italocjs, 2024-12-01:
Hello! any suggestion on how to investigate? i stll have the same issue.
running pio check -e esp32dev works, on GUI it does not.

```
italo@italodellg5:~/italolabs/simova/simova_smvtrack$ pio --version
PlatformIO Core, version 6.1.16
italo@italodellg5:~/italolabs/simova/simova_smvtrack$ pio pkg list
Resolving native dependencies...
No packages

Resolving esp32dev dependencies...
Platform espressif32 @ 6.9.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20017.0 (required: platformio/framework-arduinoespressif32 @ 3.20017.0)
├── framework-espidf @ 3.40407.240606 (required: platformio/framework-espidf @ 3.40407.240606)
├── tool-cmake @ 3.16.4 (required: platformio/tool-cmake @ ~3.16.0)
├── tool-esptoolpy @ 1.40501.0 (required: platformio/tool-esptoolpy @ ~1.40501.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── toolchain-esp32ulp @ 1.23800.240113 (required: platformio/toolchain-esp32ulp @ ~1.23800.0)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)

Libraries
├── Adafruit ADXL343 @ 1.6.4 (required: adafruit/Adafruit ADXL343 @ 1.6.4)
│   ├── Adafruit BusIO @ 1.16.1 (required: Adafruit BusIO)
│   └── Adafruit Unified Sensor @ 1.1.14 (required: Adafruit Unified Sensor)
├── Adafruit GFX Library @ 1.11.10 (required: adafruit/Adafruit GFX Library @ 1.11.10)
├── Adafruit SSD1306 @ 2.5.11 (required: adafruit/Adafruit SSD1306 @ 2.5.11)
├── ArduinoJson @ 7.1.0 (required: bblanchon/ArduinoJson @ 7.1.0)
├── DFRobot_LIS @ 1.0.1+sha.5d34332 (required: git+https://github.com/italocjs/DFRobot_LIS.git @ 1.0.1+sha.5d34332)
├── Geofence @ 1.0.0+sha.140dd27 (required: git+https://github.com/italocjs/geofencing_esp32.git @ 1.0.0+sha.140dd27)
├── NimBLE-Arduino @ 1.4.2 (required: NimBLE-Arduino @ 1.4.2)
├── NuS-NimBLE-Serial @ 2.0.4+sha.9ea1cdb (required: git+https://github.com/italocjs/NuS-NimBLE-Serial @ 2.0.4+sha.9ea1cdb)
│   └── NimBLE-Arduino @ 1.4.3 (required: NimBLE-Arduino @ >=1.4.0)
├── QMC5883LCompass @ 1.2.3 (required: mprograms/QMC5883LCompass @ 1.2.3)
├── RTClib @ 2.1.4 (required: adafruit/RTClib @ 2.1.4)
└── TinyGPSPlus @ 1.1.0 (required: mikalhart/TinyGPSPlus @ 1.1.0)

italo@italodellg5:~/italolabs/simova/simova_smvtrack$ 
``` 

Platformio.ini

```
[env:native]
lib_ldf_mode = deep+
test_framework = googletest
test_ignore = test_esp32dev
platform = native
build_unflags = -std=gnu++11
build_flags = -D USE_NATIVE_ENVIRONMENT
    -std=gnu++17
    -Wall                ; Enable all common warnings
    -Wunused-function    ; Warn about unused functions
    
lib_deps = 

[env:esp32dev]
; build_type = debug
; debug_build_flags = -Os
lib_ldf_mode = deep+
test_framework = googletest
test_ignore = test_native
platform = espressif32
board = simovatrack130
framework = arduino, espidf
platform_packages =
    framework-arduinoespressif32 @ 3.20017.0
    framework-espidf @ 3.40407.240606

board_build.partitions = partitions.csv
build_unflags = -std=gnu++11
build_flags = 
    -D USE_ESP32_ENVIRONMENT
    -std=gnu++17
    -Wall                ; Enable all common warnings
    -Wunused-function    ; Warn about unused functions
 
upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
    esp32_exception_decoder
    send_on_enter

lib_deps = 
    adafruit/Adafruit Unified Sensor @ 1.1.14
    adafruit/Adafruit BusIO @ 1.16.1
    adafruit/Adafruit ADXL343 @ 1.6.4
    https://github.com/italocjs/DFRobot_LIS.git @ 1.0.1+sha.5d34332
    https://github.com/italocjs/geofencing_esp32.git @ 1.0.0+sha.140dd27
    https://github.com/italocjs/NuS-NimBLE-Serial @ 2.0.4+sha.9ea1cdb
    mprograms/QMC5883LCompass @ 1.2.3
    adafruit/RTClib @ 2.1.4
    mikalhart/TinyGPSPlus @ 1.1.0
    bblanchon/ArduinoJson @ 7.1.0
    adafruit/Adafruit GFX Library @ 1.11.10
    adafruit/Adafruit SSD1306 @ 2.5.11
    NimBLE-Arduino @ 1.4.2

; extra_scripts = esp32_exception_decoder.py
```

valeros, 2024-12-04:
Hi @italocjs, can you provide a project example to reproduce the issue?

closed 2025-04-25 by stale[bot]