platformio/platform-espressif32 #1435: ESP32S3 using very old toolchain

m-mcgowan, 2024-08-05:
I have a PlatformIO project that I've been working on for several weeks. It's been working fine, but last week (around 1st August) the project stopped compiling. 

```
xtensa-esp32s3-elf-g++: error: unrecognized command line option '-std=gnu++2b'; did you mean '-std=gnu++2a'?
```

This is the relevant part of platformio.ini

```
framework = arduino
platform = https://github.com/platformio/platform-espressif32.git
;platform_packages = espressif/toolchain-xtensa-esp32s3
board = esp32-s3-devkitm-1
build_type = debug
build_flags = ${env:arduino.build_flags}  -DARDUINO_USB_MODE -DARDUINO_USB_CDC_ON_BOOT
```

When building, the toolchain selected is an old version:

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

I have tried uninstalling and reinstalling the latest version of `toolchain-xtensa-esp32s3` but the PIO build continues to use the old version. 

Running `pio pkg update` shows that all dependencies are already up to date.

I added 
```
platform_packages = espressif/toolchain-xtensa-esp32s3
``` 
which then uses the latest version of the toolchain (`toolchain-xtensa-esp32s3@12.2.0+20230208`),  and does build the .elf file, but results in many warnings like:

```/Users/mat/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/port/xtensa/include/freertos/portmacro.h:614:1: warning: ignoring attribute 'section (".iram1.24")' because it conflicts with previous 'section (".iram1.22")' [-Wattributes]
  614 | {
``` 

and during linking, 
```
warning: .pio/build/esp32s3/firmware.elf has a LOAD segment with RWX permissions
```

```
pio --version
PlatformIO Core, version 6.1.15

python --version
Python 3.12.4
```

I've also deleted the ~/.platformio directory and the project's `.pio` directory but no change in behavior.

m-mcgowan, 2024-08-05:
I also have an environment that targets both arduino and esp-idf, which fails with this error:

```
CMake Warning at /Users/mat/.platformio/packages/framework-espidf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Toolchain
  /Users/mat/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
  version 12.2.0 is not the supported version 8.4.0.  Check Getting Started
  documentation or proceed at own risk.

Call Stack (most recent call first):
  /Users/mat/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:6 (gcc_version_check)
  /Users/mat/.platformio/packages/framework-espidf/tools/cmake/build.cmake:334 (include)
  /Users/mat/.platformio/packages/framework-espidf/tools/cmake/build.cmake:543 (__build_process_project_includes)
  /Users/mat/.platformio/packages/framework-espidf/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:3 (project)

CMake Error at /Users/mat/.platformio/packages/framework-espidf/tools/cmake/crosstool_version_check.cmake:37 (message):
  

  Toolchain:
  /Users/mat/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc,
  crosstool-ng version esp-12.2.0_20230208 doesn't match supported version
  esp-2021r2-patch5
```

Jason2866, 2024-08-06:
It is not possible to freely choose the toolchain version.

m-mcgowan, 2024-08-06:
I understand. But then something must have changed in the package, because this was working fine a week ago - I'm building from a known good commit hash.

valeros, 2024-08-07:
Hi @m-mcgowan, unfortunately, the toolchain version is tightly coupled to the framework version, so you cannot change it arbitrarily.

m-mcgowan, 2024-08-07:
I feel there's a miscommunication here and that this issue should be kept open. My build was working fine a week ago, with nothing unusual in the `platformio.ini` file. What used to be a good known build no longer works. 

I have an environment for compiling the code for the arduino framework, and another environment that targets arduino and esp-idf.  Both of these had been working fine, with C++23 support, but no longer do so.  If I knew how to pin the versions of the packages/frameworks to what being selected as default prior to last week that would most likely provide a suitable workaround.

I have found a workaround that doesn't involve pinning versions, which, although far less than ideal, I am happy to share, but I feel this should be addressed at the source, either in this package, or in the espressif Arduino IDF component dependency, or whereever the fix is required. Notwithstanding that, at the very least, I would appreciate some help getting back to a reliable build by pinning versions of frameworks/packages.

I am not very familiar with the inner workings of the PIO frameworks, packages and toolchains, and despite my research of the changes in the past week I was not able to find a solution.

I am working on a project that has been in development for several months, for which I advocated PlatformIO.  My workaround can support continued development, but I will not be able to release the software later this year with that workaround. 

> It is not possible to freely choose the toolchain version.

To be clear, I was not choosing the toolchain version and was just using the latest, which had C++23 (gnu++2b) support. `platformio.ini` has not changed in the meantime, yet the build is now broken with missing support for gnu++2b and the `concepts` headers. Since the source code hasn't changed, I can only assume it's something in the PIO frameworks/packages that has changed, or perhaps in the PIO core.

I would truly appreciate some help from anyone that knows the PIO ecosystem better than I do.

valeros, 2024-08-08:
@m-mcgowan please share here your entire `platformio.ini` file.

brightproject, 2024-08-08:
My [trouble](https://github.com/sivar2311/platformio_boards/issues/2#issuecomment-2276488650).

Jason2866, 2024-08-08:
@brightproject Dont hijack this issue with something unrelated.

m-mcgowan, 2024-08-08:
> @brightproject Dont hijack this issue with something unrelated.

I believe this is the same problem. The old toolchain is now being used, where it wasn't previously.

m-mcgowan, 2024-08-12:
Here's a minimal example:

```
[env:esp32s3]
; This used to work just fine, but no longer recognizes -std=gnu++2b
framework = arduino
platform = espressif32
board = esp32-s3-devkitm-1
build_flags =  -fexceptions -std=gnu++2b 
   -DARDUINO_USB_CDC_ON_BOOT=1
build_unflags = -std=gnu++11

[env:esp32s3_working]
; This works with some caveats:
; Warnings about IRAM sections
; gdb is not found out of the box (workaround is to manually copy from the gdb package to the esp32s3 package)
extends = env:esp32s3
platform_packages = espressif/tool-xtensa-esp-elf-gdb
	espressif/toolchain-xtensa-esp32s3

```

The esp32s3 environment used to work shortly before I opened this issue. The project relies on concepts, so the minimal source to test is
```
#include <concepts>

void setup() {}
void loop() {}
```

The esp32 environment, which used to compile, now fails with

```
xtensa-esp32s3-elf-g++: error: unrecognized command line option '-std=gnu++2b'; did you mean '-std=gnu++2a'?
```
Changing to gnu++2a removes the unrecognized option error, but the concepts header is not present in that version of the toolchain. 
```
src/main.cpp:1:10: fatal error: concepts: No such file or directory
 #include <concepts>
 ```

So, the newer toolchain was being used by default previously, but for some reason has reverted to a much older version without gnu++2b and concepts support.

valeros, 2024-08-12:
@m-mcgowan please share also the beginning of the log  where the configuration header is printed for the broken env `[env:esp32s3]`.

m-mcgowan, 2024-08-13:
> If you want/need pin the platform version. This will use any version found on the system
> 
> ```
> platform = espressif32
> ```
> 
> better do like this (using latest version)
> 
> ```
> platform = espressif32 @ 6.8.1
> ```

Thank you for that suggestion. I did try pinning the platform version, but couldn't return to the previous behavior using any of the released versions. I know this sounds crazy! That's why I'm so puzzled about what has changed.

m-mcgowan, 2024-08-14:
Here's fuller logs from the build:

With the old toolchain:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitm-1.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32-S3-DevKitM-1
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Compiling .pio/build/esp32s3/src/main.cpp.o
Building .pio/build/esp32s3/bootloader.bin
Generating partitions .pio/build/esp32s3/partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
src/main.cpp:1:10: fatal error: concepts: No such file or directory
 #include <concepts>
          ^~~~~~~~~~
compilation terminated.
```

With the new toolchain:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitm-1.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32-S3-DevKitM-1
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Retrieving maximum program size .pio/build/esp32s3_working/firmware.elf
Checking size .pio/build/esp32s3_working/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.5% (used 18032 bytes from 327680 bytes)
Flash: [=         ]   7.0% (used 235345 bytes from 3342336 bytes)
======================== [SUCCESS] Took 1.56 seconds ========================
```

Jason2866, 2024-08-15:
@m-mcgowan To use `toolchain-xtensa-esp32s3 @ 12.2.0+20230208` for Arduino core 2.0.17 is a not valid configuration. The only used and supported toolchain from espressif for Arduino core 2.0.17 is `toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5` using every other toolchain is **not** supported and recommended. Your code needs to be changed to use only features from gcc version 8.4.0.

valeros, 2024-08-15:
@m-mcgowan The default and recommended toolchain for the Arduino framework is v8.4.0 and it's NOT capable of compiling projects with `-std=gnu++2b`. Any custom overrides you make, including manually specifying a newer version of the toolchain (which has a different structure (e.g. missing GDB binaries)), you do at your own risk.

m-mcgowan, 2024-08-19:
Thank you, I understand that.  The point here is that I didn't need any custom overrides or any changes to a simple platformio environment, and was able to target the code for g++2b. It had been working fine up until recently. 

In practice, I have several environments, one targeting just arduino and the other a combination of arduino and esp-idf. Both of these were working (with no toolchain overrides) up until the time I opened this issue. The overrides are merely a workaround so that I'm not blocked until the underlying cause is resolved.

m-mcgowan, 2024-08-19:
> @m-mcgowan To use `toolchain-xtensa-esp32s3 @ 12.2.0+20230208` for Arduino core 2.0.17 is a not valid configuration. The only used and supported toolchain from espressif for Arduino core 2.0.17 is `toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5` using every other toolchain is **not** supported and recommended. Your code needs to be changed to use only features from gcc version 8.4.0 when using Arduino core 2.0.17. Not an Platformio issue. Platformio is setting up the build environment 100% correctly.

I can appreciate this may not be a PlatformIO issue, and could well lie with the arduino core for esp32s3. I can open an issue in the core.  What is puzzling me is that this WAS working perfectly, with no custom overrides, no compiler or linker warnings, but then suddenly stopped working shortly before I opened this issue.

I looked through the recent PIO changes and there were some changes in the espressif package around selecting the arduino toolchain, overriding versions etc. I will try pinning version numbers to see if I can get back to the state that was working previously, which hopefully will shed some light on what has changed.

FunDeckHermit, 2024-08-22:
The latest version of `framework-arduinoespressif32` on the PlatformIO registry is 3.200017 ([LINK](https://registry.platformio.org/tools/platformio/framework-arduinoespressif32/versions)).
This version is 3 months old and corresponds with [Version 2.0.17](https://github.com/espressif/arduino-esp32/releases/tag/2.0.17) of the arduino-esp32 and is based on ESP-IDF 4.4. 

There is no newer version in the registry, so at the moment you're stuck.

m-mcgowan, 2024-08-23:
So puzzling that this has worked previously.  But you're definitely onto something - when I installed the esp32 BSP in Arduino IDE, version 3.0.4, it does use the correct toolchains:

`esp32:xtensa-esp-elf-gdb@12.1_20231023`

Full log below.

So who's chain (no pun intended) do I have to pull to get 3.0.4 updated in the platformIO registry? I'd be happy to help, if someone were to show me the ropes.

```
Tool esp32:mkspiffs@0.2.3 already installed
Tool esp32:mklittlefs@3.0.0-gnu12-dc7f933 already installed
Tool arduino:dfu-util@0.11.0-arduino5 already installed
Downloading packages
esp32:esp32-arduino-libs@idf-release_v5.1-b6b4727c58
esp32:esp-x32@2302
esp32:esp-xs2@2302
esp32:esp-xs3@2302
esp32:xtensa-esp-elf-gdb@12.1_20231023
esp32:esp-rv32@2302
esp32:riscv32-esp-elf-gdb@12.1_20231023
esp32:openocd-esp32@v0.12.0-esp32-20240318
esp32:esptool_py@4.6
esp32:esp32@3.0.4
Installing esp32:esp32-arduino-libs@idf-release_v5.1-b6b4727c58
Configuring tool.
esp32:esp32-arduino-libs@idf-release_v5.1-b6b4727c58 installed
Installing esp32:esp-x32@2302
Configuring tool.
esp32:esp-x32@2302 installed
Installing esp32:esp-xs2@2302
Configuring tool.
esp32:esp-xs2@2302 installed
Installing esp32:esp-xs3@2302
Configuring tool.
esp32:esp-xs3@2302 installed
Installing esp32:xtensa-esp-elf-gdb@12.1_20231023
Configuring tool.
esp32:xtensa-esp-elf-gdb@12.1_20231023 installed
Installing esp32:esp-rv32@2302
Configuring tool.
esp32:esp-rv32@2302 installed
Installing esp32:riscv32-esp-elf-gdb@12.1_20231023
Configuring tool.
esp32:riscv32-esp-elf-gdb@12.1_20231023 installed
Installing esp32:openocd-esp32@v0.12.0-esp32-20240318
Configuring tool.
esp32:openocd-esp32@v0.12.0-esp32-20240318 installed
Installing esp32:esptool_py@4.6
Configuring tool.
esp32:esptool_py@4.6 installed
Replacing platform esp32:esp32@2.0.11 with esp32:esp32@3.0.4
Uninstalling esp32:esp32@2.0.11
Running pre_uninstall script.
Platform esp32:esp32@2.0.11 uninstalled
Uninstalling arduino:dfu-util@0.11.0-arduino5, tool is no more required
Uninstalling esp32:esptool_py@4.5.1, tool is no more required
Running pre_uninstall script.
Tool esp32:esptool_py@4.5.1 uninstalled
Uninstalling esp32:mklittlefs@3.0.0-gnu12-dc7f933, tool is no more required
Uninstalling esp32:mkspiffs@0.2.3, tool is no more required
Uninstalling esp32:openocd-esp32@v0.11.0-esp32-20221026, tool is no more required
Running pre_uninstall script.
Tool esp32:openocd-esp32@v0.11.0-esp32-20221026 uninstalled
Uninstalling esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0, tool is no more required
Running pre_uninstall script.
Tool esp32:riscv32-esp-elf-gcc@esp-2021r2-patch5-8.4.0 uninstalled
Uninstalling esp32:riscv32-esp-elf-gdb@11.2_20220823, tool is no more required
Running pre_uninstall script.
Tool esp32:riscv32-esp-elf-gdb@11.2_20220823 uninstalled
Uninstalling esp32:xtensa-esp-elf-gdb@11.2_20220823, tool is no more required
Running pre_uninstall script.
Tool esp32:xtensa-esp-elf-gdb@11.2_20220823 uninstalled
Uninstalling esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0, tool is no more required
Running pre_uninstall script.
Tool esp32:xtensa-esp32-elf-gcc@esp-2021r2-patch5-8.4.0 uninstalled
Uninstalling esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0, tool is no more required
Running pre_uninstall script.
Tool esp32:xtensa-esp32s2-elf-gcc@esp-2021r2-patch5-8.4.0 uninstalled
Uninstalling esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0, tool is no more required
Running pre_uninstall script.
Tool esp32:xtensa-esp32s3-elf-gcc@esp-2021r2-patch5-8.4.0 uninstalled
Configuring platform.
Platform esp32:esp32@3.0.4 installed
```

dlbogdan, 2024-08-27:
see https://github.com/platformio/platform-espressif32/issues/1225

45gfg9, 2024-08-27:
+1 for needing a newer version toolchain. The current `toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5`, which is gcc 8.4.0, does not even have proper C++20 support. It fails to compile `operator<=>`. I see that I have `toolchain-xtensa-esp32s3` installed in `.platformio/packages`, gcc 12.2.0, but I am not bothered to hack into configurations.

dlbogdan, 2024-08-27:
Use this instead:

https://github.com/pioarduino/platform-espressif32

m-mcgowan, 2024-08-27:
> Use this instead:
> 
> https://github.com/pioarduino/platform-espressif32

Thank you for pointing me to this, and a thank you to the project's maintainers. This is perfect!

m-mcgowan, 2024-08-28:
So, what is really strange, is that the project is now building again as it was a few weeks back. With the same platformio.ini as above, the default espressif32 package is now using the correct compiler versions.

```
[env:esp32s3]
framework = arduino
platform = espressif32
board = esp32-s3-devkitm-1
build_flags =  -fexceptions -std=gnu++2b 
   -DARDUINO_USB_CDC_ON_BOOT=1
build_unflags = -std=gnu++11
```

Build output
```
 *  Executing task: platformio run --environment esp32s3 

Processing esp32s3 (framework: arduino; platform: espressif32; board: esp32-s3-devkitm-1)
-------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
cpp_flags []
LINKFLAGS 
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitm-1.html
PLATFORM: Espressif 32 (2024.8.10+sha.55f67ff) > Espressif ESP32-S3-DevKitM-1
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.4+sha.147836c 
 - tool-esptoolpy @ 4.7.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
```
It's great that it's suddenly working again, but my fear is that it will suddenly stop working. I'm so confused! :-)

45gfg9, 2024-08-28:
Do you have a `dependencies.lock` in your project root?

I was using platform version 6.6.0, `platform = espressif32` in `platformio.ini`, IDF v4.4.6 shown in the lock file, everything was working fine. Then I decided to upgrade to latest platform, so I specified 6.8.1 in the `platformio.ini`, upgraded, then it breaks one of my dependency library. Now I have to pin the version like `platform = espressif32 @ 6.6.0`, otherwise it would go latest again. I don't know how PlatformIO handles this, but if you do have a `dependencies.lock` I suspect that it changed at some time in between.

sivar2311, 2024-08-28:
@m-mcgowan 

But `platform = espressif32` in your platformio.ini contradicts `framework-arduinoespressif32 @ 3.0.4+sha.147836c` in your log !?
Arduino 3.x is not available officially by PlatformIO.

There must be some leftovers on your system. Maybe you have used 
```
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.4
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.4/esp32-arduino-libs-3.0.1.zip
```
or pioarduino before?

m-mcgowan, 2024-08-28:
I hadn't used pioarduino before until today, and it's probably no coincidence that things started working after that. It looks like platformio is picking up the different platform than intended? There is no `dependencies.lock` file in the project. (Just the platformio.ini, and a .ino file)

I don't recognize the `platform_packages`, especially `framework-arduinoespressif32-libs`, so I don't believe I've overridden those at any point.

To try to remove any leftovers, I removed `packages` and `platforms` from `~/.platformio`, and the project's `.pio` directory.  Then the old errors resurfaced.

Thank you for these responses, I feel now there is at least some explanation of what is going on.  I have saved the deleted folders from `~/.platformio` - I'll try to find out which package/platform is making things work for me.

brightproject, 2024-10-04:
If I specify the platform like this

> platform = espressif32

Then the code build is `successful`

> Processing ROUND_DISPLAY_LVGL (platform: 
> espressif32; board: esp32-s3-devkitc-1; framework: arduino)
> --------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
> PLATFORM: Espressif 32 (51.3.4+develop.sha.2656164) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
> HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
> DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.0.4
>  - framework-arduinoespressif32-libs @ 5.1.0+sha.33fbade6b8
>  - tool-esptoolpy @ 4.7.5
>  - tool-mklittlefs @ 3.2.0
>  - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
>  - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
>  - toolchain-riscv32-esp @ 12.2.0+20230208
>  - toolchain-xtensa-esp32s3 @ 12.2.0+20230208

If I specify the platform like this

> espressif32@^6.9.0

Then the code build is `unsuccessful`

> Processing ROUND_DISPLAY_LVGL (platform:
> espressif32@^6.9.0; board: esp32-s3-devkitc-1; framework: arduino)
> --------------------------Verbose mode can be enabled via `-v, --verbose` optio
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
> PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
> HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
> DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
>  - tool-esptoolpy @ 1.40501.0 (4.5.1)
>  - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
>  - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5

 
That's why I often just specify it to get the latest version of the framework:

> platform = https://github.com/pioarduino/platform-espressif32

daef, 2025-01-11:
I'm trying to use https://github.com/maxgerhardt/pio-waveshare-s3 to get a project started and am also running into this issue...: 

    Compiling .pio/build/waveshare_esp32s3_touch_lcd_128/src/main.cpp.o
    *** [.pio/build/waveshare_esp32s3_touch_lcd_128/bootloader.bin] Source `/home/d/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/bin/bootloader_qio_.elf' not found, needed by target `.pio/build/waveshare_esp32s3_touch_lcd_128/bootloader.bin'.
    xtensa-esp32s3-elf-g++: error: unrecognized command line option '-std=gnu++2b'; did you mean '-std=gnu++2a'?
    *** [.pio/build/waveshare_esp32s3_touch_lcd_128/src/main.cpp.o] Error 1

daef, 2025-08-21:
wdym by "move to Arduino3"? all i found was https://github.com/pioarduino/platform-espressif32

Jason2866, 2025-08-21:
@razvanphp This is not working anymore. Many changes in Arduino. Only way to use espressif32 Arduino 3.x is the fork pioarduino.

robertlipe, 2025-11-23:
As discussed, PIOduino is now the volunteer-supported alternative, and it actually runs modern code. 

You're free/encouraged to flee this ecosystem directly and go to CLion or, better yet, ESP-IDF. There's just no substitute if you're making a product for a chip than using the tools from the makers of that chip. There are (actively maintained) equivalents for most of the common libraries. Generally prefer the kits with namespace:espressif (supported by Espressif) but don't flip out that "OMG, I can't use arduino libs" when you can use the libs from https://components.espressif.com instead.

Platformio is "just" an layer atop of the Visual Code editor. There's no magic to it. The one case that it does make uniquely simple is if you're distributing code for a bunch of random embeddded devices and your code is closer to a product (e.g. random internet people expect to build and use it) than a tool or library (where your code would be morphed into existing build systems anyway). That is a case that PlatformIO does well. If you're targeting a single chip (or family of related chips, like "any ESP32" or "any  STM32" or such, you're better off not locking yourself into external projects (especially free ones that don't owe you anything).

valeros, 2026-10-20:
Resolved in https://github.com/platformio/platform-espressif32/releases/tag/v8.0.0

closed 2026-10-20 by valeros