platformio/platform-espressif32 #1437: ESP32-POE-ISO-WROVER board definition

solexious, 2024-08-08:
What kind of issue is this?

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 6.1.15

### Description of problem
Looking to have a board definition made for the ESP32-POE-ISO-WROVER. There is one for the ESP32-POE-ISO but the board i'm targeting uses a different esp32 with different ROM RAM and Ethernet pins etc.

Details can be found on this page (-WROVER isn't the default option on this page): https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

valeros, 2024-08-12:
Hi @solexious, a variant header file with proper pin definitions in the official [Arduino core](https://github.com/espressif/arduino-esp32) is required.

solexious, 2024-08-12:
> Hi @solexious, a variant header file with proper pin definitions in the official [Arduino core](https://github.com/espressif/arduino-esp32) is required.

Ok thanks, will file an issue there.

closed 2024-08-12 by solexious