platformio/platform-espressif32 #1438: Core Arduino library not found with framework "arduino, espidf"

sylque, 2024-08-09:
In order to use the [ESP32 BLE for Arduino](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE) library, you need to copy it to the `lib` folder. This is unexpected, as it's already part of [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32).

Here is the `platformio.ini` file:
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino, espidf
monitor_speed = 115200
upload_speed = 1500000
lib_deps =        ; DOESN'T WORK
;lib_deps = BLE   ; DOESN'T WORK
;lib_deps = https://github.com/sylque/ESP32-BLE-for-Arduino ; WORKS
```

And the `main.cpp` file:
```c++
#include <BLEDevice.h>

extern "C" void app_main() {
  initArduino();
}
```

Result:
```
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating LD script .pio\build\esp32dev\memory.ld
src/main.cpp:2:10: fatal error: BLEDevice.h: No such file or directory
```
Notice that I've created a temporary repo (with a copy of the library) to make testing easier: https://github.com/sylque/ESP32-BLE-for-Arduino. Using `lib_deps = https://github.com/sylque/ESP32-BLE-for-Arduino` allows to compile the project without error, same as copying the library in the `lib` folder.

Please see the attached project: [Issue-BLE-arduino-espidf.zip](https://github.com/user-attachments/files/16563228/Issue-BLE-arduino-espidf.zip)

Jason2866, 2024-08-09:
Maybe the issue is back https://github.com/platformio/platformio-core/issues/4900
The header file is listed as dependency https://github.com/espressif/arduino-esp32/blob/54c4b0c619458c2a1a830d8518bdc0fdaee435bf/libraries/BLE/library.properties#L10

Jason2866, 2024-08-11:
@sylque overlooked. It is an easy fault that can happen. You have forgot to include `Arduino.h`! This is needed since you use the extension `*.cpp`

sylque, 2024-08-11:
@Jason2866 I added `#include <Arduino.h>` to main.cpp, but it doesn't seem to solve the issue. Do you confirm you got the project compiling on your side?

valeros, 2024-08-12:
Hi @sylque, unfortunately, I cannot reproduce the issue. I downloaded your project example, added `#include <Arduino.h>` (without this header file the project won't compile) and executed `pio run`. The `BLE` dependency was correctly found in the framework package and the project compiled successfully.

[image]

Am I missing something?

sylque, 2024-08-12:
Well, that is odd. 

Here is my PIO version:
```
PS C:\LP\Tech\dev\bugs\Issue-BLE-arduino-espidf> pio upgrade
You're up-to-date!
PlatformIO 6.1.15 is currently the newest version available.
```

Here is the full compilation log:
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino, espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2024.7.11) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.2+sha.9100566 
 - framework-espidf @ 3.50104.0 (5.1.4) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.4 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 25 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Generating LD script .pio\build\esp32dev\memory.ld
Compiling .pio\build\esp32dev\app_trace\app_trace.c.o
Compiling .pio\build\esp32dev\app_trace\app_trace_util.c.o
Compiling .pio\build\esp32dev\app_trace\host_file_io.c.o
Compiling .pio\build\esp32dev\app_trace\port\port_uart.c.o
Compiling .pio\build\esp32dev\app_update\esp_ota_ops.c.o
Compiling .pio\build\esp32dev\app_update\esp_ota_app_desc.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_common.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_common_loader.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_clock_init.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_mem.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_random.c.o
Archiving .pio\build\esp32dev\esp-idf\app_update\libapp_update.a
Archiving .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_random_esp32.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_efuse.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\flash_encrypt.c.o
Compiling .pio\build\esp32dev\bootloader_support\src\secure_boot.c.o
Compiling .pio\build\esp32dev\bootloader_support\bootloader_flash\src\bootloader_flash.c.o
Compiling .pio\build\esp32dev\bootloader_support\bootloader_flash\src\flash_qio_mode.c.o
Compiling .pio\build\esp32dev\bootloader_support\bootloader_flash\src\bootloader_flash_config_esp32.c.o
src/main.cpp:3:10: fatal error: BLEDevice.h: No such file or directory

*******************************************************************
* Looking for BLEDevice.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:BLEDevice.h"
* Web  > https://registry.platformio.org/search?q=header:BLEDevice.h
*
*******************************************************************

    3 | #include <BLEDevice.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
==================================================================================== [FAILED] Took 24.05 seconds ====================================================================================
```

Besides, I confirm that **I do not need** to `#include <Arduino.h>` in order to make the project to compile correctly, given that I provide the BLE library path explicitly in `lib_deps` or copy the library to the `lib` folder.

sivar2311, 2024-08-12:
I see
`PLATFORM: Espressif 32 (2024.7.11) > Espressif ESP32 Dev Module` - note the "`2024.7.11`"!
`- framework-arduinoespressif32 @ 3.0.2+sha.9100566` - note the "`3.0.2+sha.9100566`"! 

which looks like a "non official" espressif32 platform!

Please try again using the latest official version
```
platform = espressif32 @ 6.8.1
```

valeros, 2024-08-12:
You're using a fork of the `espressif` dev-platform probably maintained by @Jason2866, so I believe the issue should be opened here https://github.com/Jason2866/platform-espressif32

closed 2024-08-12 by valeros

sylque, 2024-08-12:
Thanks @sivar2311, I confirm using `platform = espressif32 @ 6.8.1` solves the issue. 

@valeros, I never intended to use the fork in this project. I used the fork a few days ago in another project, with a different framework (`framework = arduino`). Any idea why PIO picks the wrong platform for this project?

Jason2866, 2024-08-12:
Using `platform = espressif32` accepts and uses every version. To make sure to have always the same wanted pin the version as shown above

sylque, 2024-08-12:
Thanks @Jason2866 . So there's no way to set up a project in sync with the latest version of https://github.com/platformio/platform-espressif32?

valeros, 2024-08-12:
> do you confirm using platform = espressif32 doesn't ensure the fork you get?

Since the fork and the official platform has the same name, PlatformIO will pick up the first one that satisfies version requirements.

sylque, 2024-08-12:
Couldn't I fork the repo, bump the version to 1000.0.1, insert a malicious error message in a file (like `#error "Missing library safety_first, please download it at https://www.website_were_i_steal_your_money.com"`), and have it deployed to anyone building the [espressif32 examples](https://github.com/platformio/platform-espressif32/tree/develop/examples) and beyond?

Jason2866, 2024-08-13:
No, not really. A fork is only installed when your platform has an entry which is not "Platformio" or "espressif*". Installing a fork does not happen without a entry pointing to the fork. If used as described here and in Platformio documentation there will be never anything else than orig. Platformio

sylque, 2024-08-13:
Sorry @Jason2866 , but I'm lost. In the project I provided (see my first post in this thread), where is the "entry point" to the fork? It only has `platform = espressif32`, so why do I end up compiling with https://github.com/Jason2866/platform-espressif32 instead of https://github.com/platformio/platform-espressif32?

Jason2866, 2024-08-13:
You have used one day my fork. If the version to use is not pinned Platformio uses a version found. To remove not wanted stuff remove hidden folder `.platformio` and let Platformio reinstall. So only original Platformio stuff will be there and used

sylque, 2024-08-13:
Thanks so much, @Jason2866, everything makes sense now.

When @valeros mentioned that 'PlatformIO will pick up the first one that satisfies version requirements,' I initially thought it meant PlatformIO would choose among the various forks available online. However, I now understand it’s actually selecting from the versions already present in the local `.platformio` folder.

This behavior can lead to a poor user experience, as my initial problem clearly demonstrated. 

Furthermore, it makes using `platform = espressif32` in the platformio.ini file unreliable, as you can’t be certain which fork will be used. This is especially problematic in a team environment, where a shared platformio.ini file doesn’t guarantee that everyone will work with the same platform version.

I think this is an unwanted behavior, which should be fixed.

As a workaround, I believe the best practice is to avoid using `platform = espressif32` alone. Instead, we should always pin the platform to a specific version (but what is two forks have the same version?) or explicitly use `platform = https://github.com/platformio/platform-espressif32`.

valeros, 2024-08-13:
@sylque If you open the [official page](https://registry.platformio.org/platforms/platformio/espressif32/installation) for the espressif32 dev-platform in the PlatformIO registry you will see the recommended syntax of declaring a proper platform specification, something like:
```ini
[env:esp32dev]
platform = platformio/espressif32@^6.8.1
...
```
This declarative approach guarantees reproducible builds on any machine.

sylque, 2024-08-13:
Separate issue filed here: https://github.com/platformio/platformio-core/issues/4962