platformio/platform-espressif32 #1440: Successfully upload code to ESP32-S2 custom board, but code is not running (LED not blinking)

nazmibojan, 2024-08-10:
I create a blinking LED code using PIN 15 of ESP32-S2 and the code was running smoothly if I upload the code using Arduino IDE:

```
void setup() {
    pinMode(15, OUTPUT);
}

void loop() {
    digitalWrite(15, HIGH);
    delay(5000);
    digitalWrite(15, LOW);
    delay(5000);
}
```

Below is my configuration in Arduino IDE:

[image]

I implement the same code using PlatformIO but the code is not running (LED was not blinking) although the code was successfully uploaded:

[image]

I build board.json for my custom board:

```
{
    "build": {
        "arduino": {
            "ldscript": "esp32s2_out.ld"
        },
        "core": "esp32",
        "extra_flags": "-DARDUINO_ESP32S2_DEV",
        "f_cpu": "240000000L",
        "f_flash": "40000000L",
        "flash_mode": "qio",
        "mcu": "esp32s2",
        "variant": "esp32s2"
    },
    "connectivity": [
        "wifi"
    ],
    "debug": {
        "openocd_target": "esp32s2.cfg"
    },
    "frameworks": [
        "arduino",
        "espidf"
    ],
    "name": "Espressif Ferbos-ESP32-S2",
    "upload": {
        "flash_size": "2MB",
        "maximum_ram_size": 327680,
        "maximum_size": 2097152,
        "require_upload_port": true,
        "speed": 460800
    },
    "url": "ferbos",
    "vendor": "Espressif"
}
```

and my platform.ini file contains this:

```
[env:ferbos_esp32s2]
platform = espressif32
board = ferbos-esp32-s2
framework = arduino

board_build.mcu = esp32s2
board_build.f_cpu = 240000000L
board_build.f_flash = 40000000L
board_build.flash_mode = qio
board_build.flash_size = 2MB
board_build.partitions = min_spiffs.csv
board_build.psram_type = disabled

upload_protocol = esptool
upload_speed = 921600

monitor_speed = 115200
```

what is the exact problem that I face? is there any misconfiguration in my platformio?

valeros, 2024-08-12:
Hi @nazmibojan

> board_build.psram_type = disabled

If you don't need the option, you simply don't set it in your `platformio.ini` instead of `disabled` which is not a valid value. 

I'm wondering if this warning has something to do with the issue:
[image]

Have you tried to reset your board manually?

Also what do you see in the serial monitor after uploading the firnware?

nazmibojan, 2024-08-12:
HI @valeros ,

>In the code you set the baudrate to 9600 while in the platformio.ini you use 115200. Those values have to be in sync.
I'd also recommend reducing the delay to 500 just for better observability.

Sorry I forgot to tell that I changed my `platform.ini` to this:
[image]

and changed my `board.json` to this:
```
{
    "build": {
        "arduino": {
            "ldscript": "esp32s2_out.ld"
        },
        "core": "esp32",
        "extra_flags": [
            "-DARDUINO_USB_CDC_ON_BOOT=1",
            "-DARDUINO_USB_MODE=0"
        ],
        "f_cpu": "240000000L",
        "f_flash": "40000000L",
        "flash_mode": "dio",
        "mcu": "esp32s2",
        "variant": "esp32s2"
    },
    "connectivity": [
        "wifi"
    ],
    "debug": {
        "openocd_target": "esp32s2.cfg"
    },
    "frameworks": [
        "arduino",
        "espidf"
    ],
    "name": "Espressif Ferbos-ESP32-S2",
    "upload": {
        "flash_size": "2MB",
        "maximum_ram_size": 327680,
        "maximum_size": 2097152,
        "require_upload_port": true,
        "speed": 460800
    },
    "url": "ferbos",
    "vendor": "Espressif"
}
```

About the delay, I'll try to reduce it and get back to you.

nazmibojan, 2024-08-12:
I also curious about flash frequency which the value is 40000000. How can we calculate that from crystall frequency?

nazmibojan, 2024-08-12:
UPDATE

I start to change the framework to espidf in platform.ini by creating new projects and put this code in main.c
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define BLINK_GPIO 15

void app_main(void)
{
    gpio_reset_pin(BLINK_GPIO);
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);

    while (1) {
        gpio_set_level(BLINK_GPIO, 1);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        gpio_set_level(BLINK_GPIO, 0);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
}
```

and the LED is blinking!
I use the same board.json files

nazmibojan, 2024-08-12:
I found the problem!

I change the `platform.ini` to this:
```
[env:ferbos_esp32s2]
platform = espressif32
board = ferbos-esp32-s2
framework = arduino

build_flags = -Iinclude

board_build.mcu = esp32s2
board_build.flash_size = 2MB
board_build.partitions = minimal.csv

upload_speed = 921600
```

The previous code not works because of in the partition table, the size of app0 and app1 are bigger than total flash (2MB) of STM32-S2-FH2

closed 2024-08-12 by nazmibojan