platformio/platform-espressif32 #1443: Esp32c6 does not compile due to missing entry in scons / esp-idf.py

zackees, 2024-08-17:
Hi there, I'm one of the maintainers of FastLED, I noticed that the C6 variant of the esp32 board does not compile.

Here is the repro case:
https://github.com/FastLED/FastLED/actions/workflows/build_esp32c6.yml

Every test *.ino file that attempts to compile using the esp32c6 as a target just immediately stack dumps with a scons exception:

```
Notice:  A new release of pip is available: 24.0 -> 24.2
Notice:  To update, run: python -m pip install --upgrade pip
Reading CMake configuration...
KeyError: 'compileGroups':
  File "/opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/runner/.platformio/platforms/espressif32/builder/main.py", line 359:
    target_elf = env.BuildProgram()
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/opt/hostedtoolcache/Python/3.11.9/x64/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 676:
    return method(*args, **kw)
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/runner/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1563:
    project_defines = get_app_defines(project_config)
  File "/home/runner/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 346:
    return extract_defines(app_config["compileGroups"][0])
 ```

Jason2866, 2024-08-17:
That's the reason for the compile error
```
Error: Invalid boards specified: esp32-c6-devkitc-1
*** Initializing environment for board esp32-c6-devkitc-1 ***
```

Jason2866, 2024-08-17:
I see no variant entry in the boards manifest. Probably the missing `"variant": "esp32c6"` is causing issues.

zackees, 2024-08-17:
Will this fix go in automatically or will it need an official release? Thanks for the fast turn around on this fix.

Jason2866, 2024-08-17:
First it needs to be merged. When you don't want to use the development branch as source for the platform than yes it needs a release.

valeros, 2024-08-19:
Hi @zackees, I'm curious how I can reproduce the issue. The `esp32-c6-devkitc-1` target is covered in our CI and there is no problem with compiling an IDF project for this board. Apparently, you're using a third-party development platform and the issue tracker in this repository is dedicated solely to the official one. 

@Jason2866 I'm not sure the `"variant": "esp32c6"` has anything to do with the issue, since this field is not used in the IDF build script. It may be used in the Arduino build script, but this board doesn't support the Arduino framework at the moment. Could you please elaborate how it could fix the problem?

Jason2866, 2024-08-19:
@valeros The entry was the only thing which is different to my fork which does compile with the board `esp32-c6-devkitc-1` for IDF and Arduino. But yes with IDF this entry is not used and IDF will compile. So the issue is in the user setup.

valeros, 2024-08-19:
Closing as not relevant to the official dev-platform in this repo.

closed 2024-08-19 by valeros