platformio/platform-espressif32 #1445: ESp32 C2 does not compile in linux, but seems to compile in Windows

zackees, 2024-08-17:
Hi, I'm a maintainer for FastLED.

I noticed that the ESP32 C2 board does not compile under our test running in linux, but seems to compile just fine on my windows box when running locally.

Here is the repro case
https://github.com/FastLED/FastLED/actions/workflows/build_esp32c2.yml

However I can totally run the command in windows and it works great:

(note that I'm running using git-bash in windows at the root of the FastLED repo)

```bash
$ ci/ci-compile --boards esp32-c2-devkitm-1 --examples _bugs/1653_S3_Apa102_Hardware_Pins
Changing working directory to C:\Users\niteris\dev\FastLED
Error: Invalid boards specified: esp32-c2-devkitm-1
Error: Invalid examples specified: _bugs/1653_S3_Apa102_Hardware_Pins
*** Initializing environment for board esp32-c2-devkitm-1 ***
Resolving esp32-c2-devkitm-1 dependencies...
Already up-to-date.
Project has been successfully updated!

*** Finished initializing environment for board esp32-c2-devkitm-1 ***
*** Initializing environment for board esp32-c2-devkitm-1 ***
Resolving esp32-c2-devkitm-1 dependencies...
Already up-to-date.
Project has been successfully updated!

*** Finished initializing environment for board esp32-c2-devkitm-1 ***
Finished initializing build_dir for board esp32-c2-devkitm-1

All build directories initialized.

*** Building _bugs/1653_S3_Apa102_Hardware_Pins for board esp32-c2-devkitm-1 ***
*** Building for first example _bugs/1653_S3_Apa102_Hardware_Pins board esp32-c2-devkitm-1 ***
*** Building example _bugs/1653_S3_Apa102_Hardware_Pins for board esp32-c2-devkitm-1 ***
Resolving esp32-c2-devkitm-1 dependencies...
Already up-to-date.
Project has been successfully updated!
Processing esp32-c2-devkitm-1 (platform: espressif32; board: esp32-c2-devkitm-1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c2-devkitm-1.html
PLATFORM: Espressif 32 (2024.8.10+sha.a348f24) > Espressif ESP32-C2-DevKitM-1
HARDWARE: ESP32C2 120MHz, 272KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.4+sha.0732d1b
 - tool-esptoolpy @ 4.7.6
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
Converting 1653_S3_Apa102_Hardware_Pins.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- src
Building in release mode
Compiling .pio\build\esp32-c2-devkitm-1\src\1653_S3_Apa102_Hardware_Pins.ino.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\FastLED.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\bitswap.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\colorpalettes.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\colorutils.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\five_bit_hd_gamma.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\hsv2rgb.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\lib8tion.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\noise.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\platforms\esp\32\clockless_rmt_esp32.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\power_mgt.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\lib1e0\src\wiring.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32-c2-devkitm-1\FrameworkArduino\esp32-hal-adc.c.o
```

zackees, 2024-08-17:
Okay confirmed, you are missing a the c2 json file.

You can probably just copy it from here:

https://github.com/zackees/platform-espressif32/blob/Arduino/IDF5/boards/esp32-c2-devkitm-1.json

Jason2866, 2024-08-17:
@zackees There is no C2 supported.

zackees, 2024-08-18:
Gotcha. Seems like a complicated political battle.

closed 2024-08-19 by valeros