platformio/platform-espressif32 #1449: Add generic support for ESP32-S3 N16R8 with 520K RAM

nicklasb, 2024-08-26:
Hi!

Seems like there are several top-end boards like that no and no support for them, would be great with a generic one. 

However, even if I am hacking another board definition and setting the values I want manually ("maximum_ram_size": 532480), I cannot allocate a large memory area (like ~300), without getting a "dram0_0_seg' overflowed" error.
The .ld scripting seems very complex, is there any other way to allocate a large thing in SRAM (for speed, PSRAM is to slow) like that?
Setting the heap to 320 K doesn't help either for the same reason, obviously, and given all the heavy lifting the ESP32 S3 can do, I would think that it is not an uncommon use case.

valeros, 2024-08-26:
Hi @nicklasb 

> is there any other way to allocate a large thing in SRAM (for speed, PSRAM is to slow) like that?

I'm afraid the issue tracker in this repository is not the right place to ask generic questions about ESP32 capabilities, please  forward your question to https://esp32.com/viewforum.php?f=23

As for generic N16R8-based boards see https://github.com/platformio/platform-espressif32/issues/989

closed 2024-08-26 by valeros

nicklasb, 2024-08-26:
@valeros 
> > is there any other way to allocate a large thing in SRAM (for speed, PSRAM is to slow) like that?
> I'm afraid the issue tracker in this repository is not the right place to ask generic questions about ESP32 capabilities, please forward your question to https://esp32.com/viewforum.php?f=23

Sorry. It seemed to me like those .ld scripts was controlled by PlatformIO in my project, but I will direct it to Espressif instead. 

> As for generic N16R8-based boards see #989

Missed that. Sorry again, and thanks!