platformio/platform-espressif32 #1455: ESP-Matter as managed component

HamzaHajeir, 2024-08-29:
Hi there, 

I'm trying to build an esp-matter project using PlatformIO. I used a simple start project with esp-idf as a framework.

Then a file under `src/` called `idf_component.yml` that configures the managed components was added with content of: 

```
dependencies:
  espressif/esp_matter:
    version: "^1.3.0"
```

This didn't succeed in reaching the compiling stage, therefore copied and editted some other files to meet one of [ESP_Matter's examples](https://github.com/espressif/esp-matter/tree/f4eaa699f0fd8433bc5ca134328934f68f1e1525/examples/managed_component_light), which also failed out with: 

```
pio run
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (51.3.4) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-
ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)                                                                     PACKAGES: 
 - framework-espidf @ 3.50104.0 (5.1.4) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.5 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.43.0") 
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /root/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 13 dependencies:
[1/13] espressif/cbor (0.6.0~1)
[2/13] espressif/esp_diag_data_store (1.0.1)
[3/13] espressif/esp_diagnostics (1.0.2)
[4/13] espressif/esp_encrypted_img (2.1.0)
[5/13] espressif/esp_insights (1.0.1)
[6/13] espressif/esp_matter (1.3.0)
[7/13] espressif/esp_secure_cert_mgr (2.4.1)
[8/13] espressif/jsmn (1.1.0)
[9/13] espressif/json_generator (1.1.2)
[10/13] espressif/json_parser (1.0.3)
[11/13] espressif/mdns (1.4.0)
[12/13] espressif/rmaker_common (1.4.6)
[13/13] idf (5.1.4)
-- Found Python3: /root/.platformio/penv/.espidf-5.1.4/bin/python3 (found version "3.12.3") found components: Interpreter 
-- Project sdkconfig file /root/PlatformIO/Projects/pio-matter/sdkconfig.esp32-s3-devkitc-1
Loading defaults file /root/PlatformIO/Projects/pio-matter/sdkconfig.defaults...
-- Looking for sys/types.h
[... 48 lines cut ...]
  /root/.platformio/packages/framework-espidf@src-29cf07cc82e49157db19869ff2d1be3f/tools/cmake/third_party/GetGitRevisionDescription.cmake:83 (
include)                                                                                                                                         /root/.platformio/packages/framework-espidf@src-29cf07cc82e49157db19869ff2d1be3f/tools/cmake/third_party/GetGitRevisionDescription.cmake:93 (
get_git_head_revision)                                                                                                                           managed_components/espressif__esp_insights/CMakeLists.txt:45 (git_describe)

ESP Insights Project commit: HEAD-HASH-NOTFOUND
CMake Error at /root/.platformio/packages/framework-espidf@src-29cf07cc82e49157db19869ff2d1be3f/tools/cmake/component.cmake:381 (__component_ge
t_property):                                                                                                                                     __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  managed_components/espressif__esp_matter/CMakeLists.txt:452 (idf_component_get_property)

========================================================= [FAILED] Took 13.73 seconds =========================================================

```

From the log, it seems the tools is first trying to fetch `.pio/build/esp32-s3-devkitc-1/esp-idf/espressif__esp_insights/CMakeFiles/git-data/head-ref` which does not exist, but what does exists in the same directory is `HEAD` instead.

However, tried to setup a similar project with ESP-IDF and succeeded. 

Repository to reproduce: (https://github.com/HamzaHajeir/pio-matter), Note it has two commits, represents the two steps described within this issue.

What the issue could it be? How to solve it?

With thanks,
Hamza Hajeir

HamzaHajeir, 2024-08-30:
> Hi @HamzaHajeir, my suspicion is that there is a conflict with other Git repository somewhere on the path to your project.

Hi @valeros, thanks for your input. 

This demo project was created in WSL environment directly before initializing the git repo to share. I can not see anything else.

But it's interesting point that the script was looking for `ref-head` but there's `HEAD` file present instead.

Can you replicate the issue at your end?

valeros, 2024-09-02:
> before initializing the git repo to share

There should be no git repos in the entire path to the PlatformIO project.

I tried your example in WSL, there was no issues with the `grabRef.cmake` file, although it failed to compile with a bunch of different errors. I'm interested how it works with the native IDF development environment. Are you running it in WSL as well? Could you please compile your project in verbose mode using IDF toolchain and attach here the entire log?

HamzaHajeir, 2024-09-02:
> There should be no git repos in the entire path to the PlatformIO project.

I've confirmed no .git directory appears before the Project file, starting from ~. 

> I tried your example in WSL, there was no issues with the `grabRef.cmake` file, although it failed to compile with a bunch of different errors. I'm interested how it works with the native IDF [...]

Yes, I am running both from WSL. 

Here's the log of blink example, with added esp-matter support, as done to the demo: 

```
 *  Executing task: /root/.espressif/tools/cmake/3.24.0/bin/cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=/root/esp-matter-test/build -S=/root/esp-matter-test 

-- Found Git: /usr/bin/git (found version "2.43.0") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 14 dependencies:
[1/14] espressif/cbor (0.6.0~1)
[2/14] espressif/esp_diag_data_store (1.0.1)
[3/14] espressif/esp_diagnostics (1.0.2)
[4/14] espressif/esp_encrypted_img (2.1.0)
[5/14] espressif/esp_insights (1.0.1)
[6/14] espressif/esp_matter (1.3.0)
[7/14] espressif/esp_secure_cert_mgr (2.4.1)
[8/14] espressif/jsmn (1.1.0)
[9/14] espressif/json_generator (1.1.2)
[10/14] espressif/json_parser (1.0.3)
[11/14] espressif/led_strip (2.5.5)
[12/14] espressif/mdns (1.4.0)
[13/14] espressif/rmaker_common (1.4.6)
[14/14] idf (5.3.0)
-- Found Python3: /root/.espressif/python_env/idf5.3_py3.12_env/bin/python3.12 (found version "3.12.3") found components: Interpreter 
-- Project sdkconfig file /root/esp-matter-test/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /root/esp-matter-test/managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, /root/esp-matter-test/managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /root/esp-matter-test/managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, /root/esp-matter-test/managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
Loading defaults file /root/esp-matter-test/sdkconfig.defaults...
warning: unknown kconfig symbol 'ESP_MATTER_NVS_USE_COMPACT_ATTR_STORAGE' assigned to 'y' in /root/esp-matter-test/sdkconfig.defaults
Loading defaults file /root/esp-matter-test/sdkconfig.defaults.esp32...
warning: unknown kconfig symbol 'ESP_MATTER_NVS_USE_COMPACT_ATTR_STORAGE' assigned to 'y' in /root/esp-matter-test/sdkconfig.defaults.esp32
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /root/.espressif/python_env/idf5.3_py3.12_env/bin/python (found version "3.12.3") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp-matter-test" version: 1.0
-- Adding linker script /root/esp-matter-test/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /root/esp-matter-test/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /root/esp/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /root/esp/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /root/esp/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /root/esp/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /root/esp/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /root/esp/v5.3/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /root/esp/v5.3/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
[... 337 lines cut ...]
pherals.ld                                                                                                                                     -- Components: __pio_env app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format es
p_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cbor espressif__esp_diag_data_stor [... cut ...]
t_property):                                                                                                                                     __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  managed_components/espressif__esp_matter/CMakeLists.txt:452 (idf_component_get_property)

========================================================= [FAILED] Took 11.34 seconds =========================================================

```

If I understood it correctly: some of the errors complaining no precense of git repository. Perhaps after it failed in finding the ref-head file.

valeros, 2024-09-02:
I see a bunch of discrepancies in your logs.

First and foremost, you're using a third-party fork of the `espressif32` development platform which is not supported in this repository:

> PLATFORM: Espressif 32 (51.3.4) (https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip)

I believe this leads to IDF version mismatch. In the native IDF project you're using v5.3.0, while in your PlatformIO project IDF version is set v5.1.4: 

> framework-espidf @ 3.50104.0 (5.1.4, https://github.com/pioarduino/esp-idf/releases/download/v5.1.4.240801/esp-idf-v5.1.4.zip)

HamzaHajeir, 2024-09-02:
I've just edited platformio.ini to use latest official release of it `platform = espressif32 @ 6.8.1`.

Here's the verbose build log: 

```
pio run -v
Processing esp32-s3-devkitc-1 (platform: espressif32 @ 6.8.1; board: esp32-s3-devkitc-1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-
ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)                                                                     PACKAGES: 
 - framework-espidf @ 3.50300.0 (5.3.0) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.43.0") 
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /root/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /root/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 13 dependencies:
[1/13] espressif/cbor (0.6.0~1)
[2/13] espressif/esp_diag_data_store (1.0.1)
[3/13] espressif/esp_diagnostics (1.0.2)
[4/13] espressif/esp_encrypted_img (2.1.0)
[5/13] espressif/esp_insights (1.0.1)
[6/13] espressif/esp_matter (1.3.0)
[7/13] espressif/esp_secure_cert_mgr (2.4.1)
[8/13] espressif/jsmn (1.1.0)
[9/13] espressif/json_generator (1.1.2)
[10/13] espressif/json_parser (1.0.3)
[11/13] espressif/mdns (1.4.0)
[12/13] espressif/rmaker_common (1.4.6)
[13/13] idf (5.3.0)
-- Found Python3: /root/.platformio/penv/.espidf-5.3.0/bin/python3 (found version "3.12.3") found components: Interpreter 
-- Project sdkconfig file /root/PlatformIO/Projects/pio-matter/sdkconfig.esp32-s3-devkitc-1
Loading defaults file /root/PlatformIO/Projects/pio-matter/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /root/.platformio/penv/.espidf-5.3.0/bin/python (found version "3.12.3") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "pio-matter" version: 1.0
-- Adding linker script /root/PlatformIO/Projects/pio-matter/.pio/build/esp32-s3-devkitc-1/esp-idf/esp_system/ld/memory.ld
[... 4 lines cut ...]
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Found Python3: /root/.platformio/penv/.espidf-5.3.0/bin/python3 (found version "3.12.3") found components: Interpreter 
-- Components: __pio_env app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format es
p_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump espressif__cbor espressif__esp_diag_data_store espressif__esp_diagnostics e [... cut ...]
CMake Error at /root/.platformio/packages/framework-espidf/tools/cmake/component.cmake:382 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  managed_components/espressif__esp_matter/CMakeLists.txt:452 (idf_component_get_property)

========================================================= [FAILED] Took 15.34 seconds =========================================================

```

valeros, 2024-09-02:
OK, so the `ref-head` issue seems to have gone. I also suspect that the `Matter` component only accepts the `main` component as dependency. You can try to rename the `src` folder to `main` and add the following line at the top of you `platformio.ini`:
```
[platformio]
src_dir = main
```

HamzaHajeir, 2024-09-02:
Now I'm having an error which is outputted for the first build after cleaning the project: 

```
-- Configuring incomplete, errors occurred!
See also "/root/PlatformIO/Projects/pio-matter/.pio/build/esp32-s3-devkitc-1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository: /root/.platformio/packages/framework-espidf/components/openthread/openthread/../../../.git/modules/components/openthread/openthread
ESP Insights Project commit: bd7b483-dirty
CMake Error at main/CMakeLists.txt:8 (set_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "CXX_STANDARD" is not allowed.

CMake Error at main/CMakeLists.txt:9 (target_compile_options):
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
```

The concerning lines mentions:
```cmake
set_property(TARGET ${COMPONENT_LIB} PROPERTY CXX_STANDARD 17)
target_compile_options(${COMPONENT_LIB} PRIVATE "-DCHIP_HAVE_CONFIG_H")
```

The next build: 

```
pio run -v
Processing esp32-s3-devkitc-1 (platform: espressif32 @ 6.8.1; board: esp32-s3-devkitc-1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-
ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)                                                                     PACKAGES: 
 - framework-espidf @ 3.50300.0 (5.3.0) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32s3
Processing 13 dependencies:
[1/13] espressif/cbor (0.6.0~1)
[2/13] espressif/esp_diag_data_store (1.0.1)
[3/13] espressif/esp_diagnostics (1.0.2)
[4/13] espressif/esp_encrypted_img (2.1.0)
[5/13] espressif/esp_insights (1.0.1)
[6/13] espressif/esp_matter (1.3.0)
[7/13] espressif/esp_secure_cert_mgr (2.4.1)
[8/13] espressif/jsmn (1.1.0)
[9/13] espressif/json_generator (1.1.2)
[10/13] espressif/json_parser (1.0.3)
[11/13] espressif/mdns (1.4.0)
[12/13] espressif/rmaker_common (1.4.6)
[13/13] idf (5.3.0)
-- Configuring incomplete, errors occurred!
See also "/root/PlatformIO/Projects/pio-matter/.pio/build/esp32-s3-devkitc-1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /root/.platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  ERROR: Some components (espressif/esp_matter) in the "managed_components"
  directory were modified on the disk since the last run of the CMake.
  Content of this directory is managed automatically.

  If you want to keep the changes, you can move the directory with the
  component to the "components"directory of your project.

  I.E.  for "espressif__esp_matter" run:

  mv
  /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_matter
  /root/PlatformIO/Projects/pio-matter/components/espressif__esp_matter

  Or, if you want to discard the changes remove the ".component_hash" file
  from the component's directory.

  I.E.  for "espressif__esp_matter" run:

  rm
  /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_matter/.component_hash

Call Stack (most recent call first):
  /root/.platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

========================================================= [FAILED] Took 3.55 seconds =========================================================

```

valeros, 2024-09-02:
Hmm, that's probably a conflict in the `managed_components` folder. Try to delete both `.pio` and `managed_components` folders.

HamzaHajeir, 2024-09-02:
> Hmm, that's probably a conflict in the `managed_components` folder. Try to delete both `.pio` and `managed_components` folders.

Same ... 

Before any iteration or trial, I do clean the project alongwith removing managed_components directory, and removing the auto generated sdkconfig.xyz.

Therefore might be some issue in downloaded cache?

Jason2866, 2024-09-02:
The component manager has a changed behaviour. It stops when in managed folder a file has changed. A workaround is to add this line `os.environ["IDF_COMPONENT_OVERWRITE_MANAGED_COMPONENTS"] = "1"` around line 50 in `espidf.py` in folder  `builder/frameworks`
/espidf.py

HamzaHajeir, 2024-09-02:
> The component manager has a changed behaviour. It stops when in managed folder a file has changed. A workaround is to add this line `os.environ["IDF_COMPONENT_OVERWRITE_MANAGED_COMPONENTS"] = "1"` around line 50 in `espidf.py` in folder `builder/frameworks`

I see, that now bypasses this error, and stops on a next one that relates to esp-matter's copied configuration files: 

```cmake
set_property(TARGET ${COMPONENT_LIB} PROPERTY CXX_STANDARD 17)
target_compile_options(${COMPONENT_LIB} PRIVATE "-DCHIP_HAVE_CONFIG_H")
```

Here's the log: 
```
-- Configuring incomplete, errors occurred!
See also "/root/PlatformIO/Projects/pio-matter/.pio/build/esp32-s3-devkitc-1/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository: /root/.platformio/packages/framework-espidf/components/openthread/openthread/../../../.git/modules/components/openthread/openthread
ESP Insights Project commit: bd7b483-dirty
CMake Error at main/CMakeLists.txt:8 (set_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "CXX_STANDARD" is not allowed.

CMake Error at main/CMakeLists.txt:9 (target_compile_options):
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
```

valeros, 2024-09-02:
> Target "__idf_main" is not an executable or library.

That's exactly the error I bumped into. What's more puzzling is that I also observed the same error in the native IDF development environment:
<details>
<summary>log</summary>

```
Executing action: all (aliases: build)
Running cmake in directory /mnt/c/Users/vbox/pio-matter/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/vbox/.espressif/python_env/idf5.3_py3.12_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /mnt/c/Users/vbox/pio-matter"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.43.0")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/vbox/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/vbox/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/vbox/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
.................NOTICE: Updating lock file at /mnt/c/Users/vbox/pio-matter/dependencies.lock
Processing 13 dependencies:
[1/13] espressif/cbor (0.6.0~1)
[2/13] espressif/esp_diag_data_store (1.0.1)
[3/13] espressif/esp_diagnostics (1.0.2)
[4/13] espressif/esp_encrypted_img (2.1.0)
[5/13] espressif/esp_insights (1.0.1)
[6/13] espressif/esp_matter (1.3.0)
[7/13] espressif/esp_secure_cert_mgr (2.4.1)
[8/13] espressif/jsmn (1.1.0)
[9/13] espressif/json_generator (1.1.2)
[10/13] espressif/json_parser (1.0.3)
[11/13] espressif/mdns (1.4.0)
[12/13] espressif/rmaker_common (1.4.6)
[13/13] idf (5.3.0)
-- Found Python3: /home/vbox/.espressif/python_env/idf5.3_py3.12_env/bin/python3 (found version "3.12.3") found components: Interpreter
-- Project sdkconfig file /mnt/c/Users/vbox/pio-matter/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /mnt/c/Users/vbox/pio-matter/managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, /mnt/c/Users/vbox/pio-matter/managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /mnt/c/Users/vbox/pio-matter/managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:929, /mnt/c/Users/vbox/pio-matter/managed_components/espressif__esp_matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
Loading defaults file /mnt/c/Users/vbox/pio-matter/sdkconfig.defaults...
warning: unknown kconfig symbol 'BUTTON_PERIOD_TIME_MS' assigned to '20' in /mnt/c/Users/vbox/pio-matter/sdkconfig.defaults
warning: unknown kconfig symbol 'BUTTON_LONG_PRESS_TIME_MS' assigned to '5000' in /mnt/c/Users/vbox/pio-matter/sdkconfig.defaults
warning: unknown kconfig symbol 'ESP_MATTER_NVS_USE_COMPACT_ATTR_STORAGE' assigned to 'y' in /mnt/c/Users/vbox/pio-matter/sdkconfig.defaults
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /home/vbox/.espressif/python_env/idf5.3_py3.12_env/bin/python (found version "3.12.3") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "pio-matter" version: 1.0
-- Adding linker script /mnt/c/Users/vbox/pio-matter/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /mnt/c/Users/vbox/pio-matter/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /mnt/d/IDF/5.3.0/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /mnt/d/IDF/5.3.0/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /mnt/d/IDF/5.3.0/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /mnt/d/IDF/5.3.0/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /mnt/d/IDF/5.3.0/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /mnt/d/IDF/5.3.0/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /mnt/d/IDF/5.3.0/components/soc/esp32/ld/esp32.peripherals.ld
ESP Insights Project commit: bd7b483-dirty
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_h [... cut ...]
    $<TARGET_FILE:__idf_main>

  Target "__idf_main" is not an executable or library.

-- Generating done (39.5s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the /mnt/c/Users/vbox/pio-matter/build/log/idf_py_stderr_output_1444 and /mnt/c/Users/vbox/pio-matter/build/log/idf_py_stdout_output_1444
```
</details>

@HamzaHajeir Could you please provide a bit more details how you managed to compile this project using the IDF toolchain?

HamzaHajeir, 2024-09-02:
I've started from an example a blink example, modified by adding the exact
additions added in the demo repository: idf_component.yml, both CMakeLists,
and sdkconfig.default.

This might be an issue of renaming src to main?

* If more details is needed I'll provide once back home

HamzaHajeir, 2024-09-02:
> > Target "__idf_main" is not an executable or library.
> 
> That's exactly the error I bumped into. What's more puzzling is that I also observed the same error in the native IDF development environment:
> 
> log
> @HamzaHajeir Could you please provide a bit more details how you managed to compile this project using the IDF toolchain?

Here's a copy of the blink example with added esp-matter on it: https://github.com/HamzaHajeir/esp-matter-test

valeros, 2024-09-05:
> Target "__idf_main" is not an executable or library.

This issue is probably caused by a typo in the `src/CMakeLists.txt` file. The folder with sources was renamed to `main`, but the list of sources in that file was still pointing to `src`.

Anyway, after some trial and error I managed to compile your project, although it requires the latest version of the `espressif32` dev-platform with some fixes from this repository. I'm attaching a project example that successfully compiles on my machine.

[pio-matter-updated.zip](https://github.com/user-attachments/files/16895950/pio-matter-updated.zip)

HamzaHajeir, 2024-09-05:
> This issue is probably caused by a typo in the `src/CMakeLists.txt` file. The folder with sources was renamed to `main`, but the list of sources in that file was still pointing to `src`.
>
> Anyway, [...]

Thanks! We have missed the directory renaming effects backthen. 

This demo now just do compile!

closed 2024-09-05 by HamzaHajeir