platformio/platform-espressif32 #1456: Help with finding the reason for a non-working code ESP-WROOM-32

brightproject, 2024-09-04:
My project compiles, flashes and works fine with version `3.5.0`.
As soon as I change the version to `4.0.0` or `5.0.0` or `6.8.1`

> Processing stabilo-pcb (platform: espressif32@^4.3.0; board: esp32dev; framework: arduino)----------------------------------------------
> Verbose mode can be enabled via -v, --verbose option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB FlashDEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jta
> g-tiny, tumpa)PACKAGES:
>  - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3) - tool-esptoolpy @ 1.30300.0 (3.3.0)
>  - tool-mkfatfs @ 2.0.1 - tool-mklittlefs @ 1.203.210628 (2.3)
>  - tool-mkspiffs @ 2.230.0 (2.30) - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

The output to the serial port is like this:

> 13:02:59.893 > [     4][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
> 
> 13:02:59.924 > [    32][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

And the code does not work.
I cannot understand the reason for this behavior.
I found that `UART2` on pins `16` and `17`, and the `IP` address for connecting to `Wi-Fi` can affect, but this cannot affect me, and the same code works on `3.5.0,` but does not work on versions higher than this.

Jason2866, 2024-09-04:
github issues is **not** a place for asking individual programming problems

valeros, 2024-09-05:
Hi @brightproject, as @Jason2866  correctly pointed out, the issue tracker in this repository is not the right place for project-specific questions. Anyway, versions >3.5.0 switched from Arduino core v1 to Arduino core v2 which had a great number of changes, so I'd recommend checking if your code was not affected.

closed 2024-09-05 by valeros