platformio/platform-espressif32 #1463: support for sparkfun esp32-s3 thing plus board

Vigeant, 2024-09-19:
Good day, still exploring how to do this myself but though I would mention it here.

Please add support for the sparkfun esp32-s3 thing plus board. I noticed that you have many esp32-s3 boards already so should be a quick addition.

Vigeant, 2024-09-27:
@valeros I started doing the work hand have written the json file, the entry in the boards.txt and the variant folder for the arduino framework. It all compiles and uploads but the code does not run. Any clue how to debug this. My arduino framework additions work with the arduino ide 2.0 (I can compile, upload and execute).

valeros, 2024-09-30:
Hi @Vigeant, share here your board manifest. What code do you use to verify the board is working?

Vigeant, 2024-10-09:
hi @valeros 
attached is my current json file.
[sparkfun_esp32s3_thing_plus.json](https://github.com/user-attachments/files/17300278/sparkfun_esp32s3_thing_plus.json)

I am using a simple blink demo
```C
/*
 * Blink
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */

#include <Arduino.h>

// Set LED_BUILTIN if it is not defined by Arduino framework
#ifndef LED_BUILTIN
    #define LED_BUILTIN 46
#endif

#define LED_BUILTIN 0

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.println("LED on");
  // wait for a second
  delay(1000);
  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);
  Serial.println("LED off");
   // wait for a second
  delay(1000);
}

the json file refers to the arduino board I wrote. I did a pull request for it but will take a while for it to propagate.
```

valeros, 2024-10-09:
I see you've used `sparkfun_esp32s3_thing_plus` in the variant field. There is no such variant folder in the Arduino core v2.x which is used by PlatformIO at the moment. This makes me think you're using a custom dev-platform or a fork? Does the board print anything to the console?

Vigeant, 2024-10-09:
hi @valeros thats correct, I wrote my own based on the generic esp32-s3 dev board and yes I had the demos working. Yesterday I tried pointing my platformio.ini since my pull request was merged in the arduino @master recently but still having trouble.

Here is my platformio.ini
```
[env:sparkfun_esp32s3_thing_plus]
platform = https://github.com/platformio/platform-espressif32.git
;platform = espressif32
board = sparkfun_esp32s3_thing_plus
framework = arduino
;may need to update lib link to latest version https://github.com/espressif/esp32-arduino-lib-builder/releases
; see this for additional details https://github.com/platformio/platform-espressif32/issues/1225
platform_packages = 
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
    framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.1/esp32-arduino-libs-idf-release_v5.1-33fbade6.zip
monitor_speed = 115200
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
```

Vigeant, 2024-10-09:
not sure how to go about officially adding a board and use platformio. I guess I could fork the last 2.0 build and add my board and point to my fork. Is there a timeline for platformIO to move to core V3.x?

valeros, 2024-10-10:
I'm afraid I cannot help you here with third-party implementations. As for support for v3.x, see #1225.

Jason2866, 2024-10-10:
@Vigeant Your platform_packages is wrong use
```
platform = https://github.com/platformio/platform-espressif32.git
platform_packages = 
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.5
  framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.1/esp32-arduino-libs-idf-release_v5.1-33fbade6.zip
```
to use core 3.0.5

jeremiahrose, 2024-12-20:
What are the blockers for resolving this issue - can I help in any way? I have this board and would love to get it working. Are there any workarounds?

Vigeant, 2024-12-20:
Espressif dropping support for platformio...

Jason2866, 2024-12-20:
@jeremiahrose If you look https://github.com/platformio/platform-espressif32/issues/1225#issuecomment-2529060276
you have a solution.

jeremiahrose, 2024-12-22:
> Espressif dropping support for platformio...

@Vigeant I'm not sure I understand. Can you explain to me why them dropping platformio support means that we can't proceed? Did they stop paying platformio developers? If so, then perhaps the community can contribute - I'd be happy to. Other than Espressif's support, what the technical blockers are for enabling this board?

@Jason2866 in the comment you linked to, @kylefmohr recommends pulling in https://github.com/pioarduino/platform-espressif32, which is a fork of this repo. So what's preventing us from merging the community contributed code in this fork back into this repo?

kylefmohr, 2024-12-22:
@jeremiahrose I don't understand why you can't proceed, just use the pioarduino fork and that's that.

As to the reasons for the falling out between Espressif and PlatformIO, this is extensively discussed in #1225, and nothing any of us say here is likely to change the situation.

In my experience, the new fork is very actively maintained, works well, and it's extremely easy to switch to using it.

jeremiahrose, 2024-12-22:
@kylemohr thanks for the reply. The pioarduino fork doesn't have a sparkfun_esp32s3_thing_plus board either! So what would it take to solve this issue, either in this repo or the fork? Should we transfer this issue over to the fork? Is it documented somewhere that Espressif board requests should be posted there, and not here?

kylefmohr, 2024-12-22:
Right, you have to remember that the people running this fork are volunteers - 

I would recommend opening a PR in [pioarduino/platform-espressif32](https://github.com/pioarduino/platform-espressif32), take a look at merged PRs to see how others have been adding boards: https://github.com/pioarduino/platform-espressif32/pulls?q=is%3Apr+is%3Aclosed

jeremiahrose, 2024-12-22:
Roger that, thanks @kylefmohr . 

For reference, here is @Vigeant 's PR in the Arduino core: https://github.com/espressif/arduino-esp32/pull/10382

kylefmohr, 2024-12-22:
@jeremiahrose if you'd like to get started immediately, use this as your `platformio.ini` file

```
; Flash: 4MB QD, PSRAM: 2MB QD
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino

board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM
```

I sourced this from [this repo](https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations?tab=readme-ov-file#esp32-s3-mini-11u-n4r2) and [this hardware description of the board](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3/blob/c9fc69fc1d79e1f3e45f7f2e50d5e357b4dd9242/docs/hardware_overview.md#:~:text=Thing%20Plus%20uses%20the%20ESP32%2DS3%2DMINI1%2DN4R2)

jeremiahrose, 2024-12-22:
Thanks so much @kylefmohr. The above environment didn't quite work, but by doing a diff between @Vigeant's `.json` and the `esp32-s3-devkitc-1.json`, I was able to fill in the missing settings. This environment is working perfectly for me so far:

```
[env:sparkfun_esp32s3_thing_plus]
board = esp32-s3-devkitc-1
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino
board_build.arduino.memory_type = qio_qspi
board_build.arduino.partitions = default.csv
board_build.flash_mode = qio
board_build.psram_type = qio
board_build.cdc_on_boot = 1
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
board_build.extra_flags =
  -DARDUINO_USB_CDC_ON_BOOT=1
```

jeremiahrose, 2024-12-22:
Have added this workaround to the Sparkfun documentation: https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3/pull/3

jeremiahrose, 2024-12-22:
> the json file refers to the arduino board I wrote. I did a pull request for it

@Vigeant I couldn't find the pull request for your board manifest, so I have created a new one over at `pioarduino/platform-espressif32`: https://github.com/pioarduino/platform-espressif32/pull/89

Vigeant, 2024-12-22:
> pioarduino/platform-espressif32: pioarduino#89

Sorry, I gave up on platformio and moved to esp-idf.

jeremiahrose, 2024-12-22:
https://github.com/pioarduino/platform-espressif32/pull/89 has been merged, so we don't need the workardound anymore. Just pull the development version of `pioarduino/platform-espressif32` like so:
```
[env:sparkfun_esp32s3_thing_plus]
framework = arduino
platform = https://github.com/pioarduino/platform-espressif32.git#develop
board = sparkfun_esp32s3_thing_plus
```

jeremiahrose, 2024-12-22:
I've updated the PR to Sparkfun's documentation to reflect this. I guess we can close this issue now, unless we want to wait for `pioarduino/platform-espressif32` to cut a new release.

closed 2026-10-16 by valeros