platformio/platform-espressif32 #1473: Compile error building project for Espressif ESP32-C6-DevKitC-1

znmeb, 2024-10-15:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 22.04 LTS

**PlatformIO Version** (`platformio --version`):
```
❯ pio --version
PlatformIO Core, version 6.1.16
```

### Description of problem

Cannot build project on Espressif ESP32-C6-DevKitC-1

#### Steps to Reproduce

1. Clone the project repository <https://github.com/MitchBradley/cforth.git>.
2. Append the following to the repository `platformio.ini` file.

    ```
   
    [env:esp32-c6-devkitc-1]
    extends = target
    platform = platformio/espressif32
    framework = espidf
    board = esp32-c6-devkitc-1
    ; 320 KiB of RAM
    build_flags = ${target.target_build_flags} -DMAXDICT=0x50000
    ```

3. Run `pio run --verbose`. This generates the "host" version of `cforth`, which provides data needed by the embedded `cforth` built in the next step. The output of this is in logfile `3_build_host_cforth_log.txt`.
4. Run

    ```
    /usr/bin/time pio run --verbose \
        --environment esp32-c6-devkitc-1 \
        --target upload \
        >> $LOGFILE 2>&1
    ```

    This is supposed to compile, link and upload the `cforth` firmware to the board but the compile step crashes. See the logfile `esp32-c6-devkitc-1_log.txt`.

    
### Actual Results

Compile step fails.

### Expected Results

Compile, link and upload to board.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[env:esp32-c6-devkitc-1]
extends = target
platform = platformio/espressif32
framework = espidf
board = esp32-c6-devkitc-1
; 320 KiB of RAM
build_flags = ${target.target_build_flags} -DMAXDICT=0x50000
```

**Source file to reproduce issue:**
It's a whole project - <https://github.com/MitchBradley/cforth.git>.

### Additional info
I have this as a set of scripts. If you'd like, I can make it self-contained and post a `zip` archive. I'm still not convinced I didn't misread the PlatformIO documentation, but I've tried a few different versions of the `.ini` and haven't found one that worked.

valeros, 2024-10-16:
Hi @znmeb, according to the logs you attached, the problem is that the `Arduino.h` is missing which is expected because you specified ESP-IDF as the framework.

closed 2024-10-16 by valeros

znmeb, 2024-10-16:
Which framework should I use then?

valeros, 2024-10-16:
Unfortunately, you cannot use this board with PlatformIO if your project depends on Arduino API.

znmeb, 2024-10-16:
That is indeed unfortunate.