platformio/platform-espressif32 #1486: Fix Wireless Stick V3 definition

BrentonPoke, 2024-11-08:
The Wireless Stick and Wireless Stick Lite from Heltec have the wrong MCU listed in their board definitions and I think it's interfering with the build process.
```
=================================================================================================================================================
ID                                   MCU      Frequency    Flash    RAM     Name

heltec_wireless_stick                ESP32    240MHz       8MB      320KB   Heltec Wireless Stick
heltec_wireless_stick_lite           ESP32    240MHz       4MB      320KB   Heltec Wireless Stick Lite
```
This is what it shows right now, when the MCU should be `esp32s3`. Using the board definition for the Heltec Wifi LoRa V3 seems to work for just getting the board running since it uses the same MCU, but there's something off about the display.

valeros, 2024-11-08:
> The Wireless Stick and Wireless Stick Lite from Heltec have the wrong MCU listed in their board definitions

They are not wrong, those IDs were created for previous revisions of the boards which were based on a generic ESP32.
`Heltec Wireless Stick Lite v3` is a completely different board based on ESP32-S3. 

I slightly changed the title of the issue and labeled it as a board request.

BrentonPoke, 2024-11-08:
What are all the differences between the two? If it amounts to just busy work of copying the existing profile and editing a few fields, I could do it.

valeros, 2024-11-11:
> What are all the differences between the two? If it amounts to just busy work of copying the existing profile and editing a few fields, I could do it.

TBH, I haven't looked into the differences, but I'd recommend using the `esp32-s3-devkitc-1` manifest as a basis, since the values for ESP32-S3 targets are quite different. Keep in mind that this only makes sense if there are proper variant files in the Arduino core v2.x for those V3 boards.

BrentonPoke, 2024-11-16:
I think this might be too much for me at the moment. I'd have to figure out how to set this up and test it.

closed 2026-10-16 by valeros