platformio/platform-espressif32 #1500: Program size incorrectly checked when building for ESP32

vladkorotnev, 2024-12-02:
I have a [partitions.csv](https://github.com/vladkorotnev/plasma-clock/blob/develop/partitions.csv) with two 0x180000 partitions, and platformio.ini configured [like this](https://github.com/vladkorotnev/plasma-clock/blob/develop/platformio.ini), running on Windows 10. 

When building, the build pipeline finds the partition size correctly (1,572,864 bytes) and claims that the program fits:

```
Checking size .pio\build\AKI_K875\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.7% (used 64588 bytes from 327680 bytes)
Flash: [==========]  99.7% (used 1568765 bytes from 1572864 bytes)
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM4
```

However the actual file created is 1,575,056 bytes (file size, not size on disk) — that is, 0x180890 bytes, which is much bigger than the 0x180000 byte partition. And so esptool tries to flash exactly that, which leads to the firmware crashing on startup:

```
...
Flash will be erased from 0x00010000 to 0x00190fff...
...
Compressed 1575056 bytes to 952743...
Writing at 0x00010000... (1 %)
...
Writing at 0x0018fa51... (100 %)
Wrote 1575056 bytes (952743 compressed) at 0x00010000 in 16.4 seconds (effective 768.8 kbit/s)...
Hash of data verified.
...
--- Terminal on COM4 | 115200 8-N-1
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46

... many more of the same thing ...
```

Which is kind of expected, since the flash was written correctly, but is being mapped into the memory according to the partition table — so part of the program ends up being not available, the code jumps somewhere into the uninitialized RAM, and things explodify.

Why does it think the program is smaller than it actually is? Could this be a Windows-specific issue? (I don't have a Linux or Mac machine to test)

I'm not 100% sure if this belongs in here or in some other part of the toolchain, please redirect me if needed.

valeros, 2024-12-04:
Hi @vladkorotnev, there is probably an inaccuracy in a regular expression used for parsing the output from the size tool. Since we use the [same regex](https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L193) as the Arduino IDE does, I'd recommend reproducing the problem using Arduino environment and opening an issue in their repository.

closed 2024-12-04 by valeros

vladkorotnev, 2024-12-04:
According to the post [here at the ESP32 forum](https://esp32.com/viewtopic.php?f=19&t=43303&sid=57a889e4b5b8de8e0cbcad2fba28ce62#p141312), the `size` tool does not produce a correct result at all, so the regexp is probably fine, but the concept of measuring the size like that is not.

I will copy this issue into the Arduino repository for the time being.

In the mean time, for anyone dropping by from Google etc, here is a workaround I did in the mean time that can be placed in the post-actions in platformio.ini: https://github.com/vladkorotnev/plasma-clock/blob/develop/helper/check-size-correctly.py . This is sufficient to derail the build pipeline if the produced flash image is going to definitely be unbootable.