platformio/platform-espressif32 #1501: `merge_bin` fails in Github Actions when called from `pkg exec --package "platformio/tool-esptoolpy" -- esptool`

CocoDico78, 2024-12-04:
Calling `pio pkg exec --package "platformio/tool-esptoolpy" -- esptool --chip esp32 merge_bin --output output.bin 0x1000 test1.bin 0x2000 test2.bin` fails in Github Actions runner with a `UserSideException: [Errno 13] Permission denied: 'esptool'` error.

Calling the equivalent `python -m esptool --chip esp32 merge_bin --output output.bin 0x1000 test1.bin 0x2000 test2.bin` works in Github Actions runner.

The error appears in Windows and Ubuntu. The error doesn't appear outside of Github Actions runner (ie it "works on my machine").

For repro and logs, you can check [the Actions logs](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions) of [this repo](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail). The failed run shows the error with `pio pkg exec ...`, the successful run shows the `python -m ...` call.

Jason2866, 2024-12-04:
Wrong way to use the Platformio installed python. See https://github.com/platformio/platform-espressif32/blob/c6a8c9a08a8de8aa83e6d15eb1dc9ae751056c35/builder/main.py#L283
You have to use the Platformio env var for PYTHON to get the path and call

Jason2866, 2024-12-04:
Here is an example to use esptool.py to merge binaries
https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/post_esp32.py

CocoDico78, 2024-12-04:
I know how to merge binaries using esptool. The issue is that the command fails in a Github Actions runner when called through the `pio pkg exec` paradigm, see [this Actions log](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12155987517/job/33898674084#step:5:24).

To prove that it's not an issue with the underlying `merge_bin` command, I also provide [a reference Actions log](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12155987521/job/33898674128#step:5:1) that works, through a non-platformio way.

The repo I'm linking is specifically made as a minimum repo that reproduces the issue. It's not doing anything except calling `merge_bin` in two different ways and showing that it fails with `pio pkg exec`.

Jason2866, 2024-12-04:
The snippet shown does work in github actions too.

CocoDico78, 2024-12-04:
The snippet shown doesn't use `pio pkg exec`; it is using `esptool.main` to call into esptool. It's similar to what I demonstrate [here](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12155987521/job/33898674128#step:5:1), so I'm not suprised it works.

When using `pio pkg exec`, it fails in Github Actions however, as demonstrated [here](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12155987517/job/33898674084#step:5:24).

Jason2866, 2024-12-04:
I guess `pio` calls not the correct Platformio. Try with pinned path

CocoDico78, 2024-12-04:
> I guess pio calls not the correct Platformio

What is the correct platformio? The Github Actions runs `pip install --upgrade platformio` and then calls `pio`. What other platformio is there that it could be mistaken?

> Try with pinned path

Pinned path for what? For pio? Also I reproduced the issue with the much simpler `esptool version` instead of `esptool merge_bin [...]`. This is not a `merge_bin` issue but as you say a wiring issue between pio and its tools.

Jason2866, 2024-12-04:
Yes, path for calling `pio`

CocoDico78, 2024-12-04:
> You have probably two Platformio versions installed

So you're saying that there is an issue with the `pip install --upgrade platformio` command itself? As you can see [in the workflow file](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12162143930/workflow), I don't explicitly install two versions of platformio.

Or is it by design from platformio that there are always two versions that get installed?

CocoDico78, 2024-12-04:
And by design `pio` will point to the wrong one?

CocoDico78, 2024-12-04:
> What is your reason for installing the later needed tool manually?

That is a good question. In my automated build, I use pio for the heavy-lifting and later I need to invoke esptool commands. Since I already have pio installed and `"platformio/tool-esptoolpy"` is already installed, I figured it'd be better to call into `pio pkg exec --package "platformio/tool-esptoolpy" -- esptool` instead of installing `esptool` again with python directly. It turns out it doesn't work, as this issue shows.

So I don't need to install the tool manually. I just need to call into it when it's already installed.

Jason2866, 2024-12-04:
> So I don't need to install the tool manually. I just need to call into it when it's already installed. Which doesn't work (in GH runner).

That looks like my statement is correct. There is no path set and so just calling `esptool.py` fails, since not found

CocoDico78, 2024-12-04:
> so just calling esptool.py fails, since not found.

I don't call `esptool.py`, do I? I call into a tool using `pio pkg`. There is something I don't see?

> Solution find the path and store "somewhere" to call it later

I don't follow. Why is the point of this tool "platformio/tool-esptoolpy" if it doesn't run as standalone?

> Easier to do all needed post compile steps in an Platformio script

It's easier to do `pip install esptool` and call `python -m esptool` if you ask me. Don't want to be messing with Python. But that's opinion-based :)

Jason2866, 2024-12-04:
Anyways looks like no issue from Platformio here.

CocoDico78, 2024-12-04:
> looks like no issue from Platformio here.

There is the issue that the `"platformio/tool-esptoolpy"` package doesn't work in GH Actions. But as you say that's not a general platformio issue (`pio pkg exec` works just fine in general), that's an issue with the `"platformio/tool-esptoolpy"` package. Is there a better place for me than this repo to open an issue?

Jason2866, 2024-12-04:
I wouldn't say there is an issue. Call the command with path and it will work.

CocoDico78, 2024-12-04:
Sure, I've [added the full path](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12163941569/workflow#L25) so now the full command is `/opt/hostedtoolcache/Python/3.12.7/x64/bin/pio pkg exec --package "platformio/tool-esptoolpy" -- esptool version` (on Ubuntu).

The [new GH logs]
(https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12163941569/job/33924374160#step:5:24) show the same error.

If that's not the correct path, then what is the correct path? And why would pio get installed in the wrong path?

Jason2866, 2024-12-04:
This is not the path of the venv!
`/opt/hostedtoolcache/Python/3.12.7/x64/bin/pio`

CocoDico78, 2024-12-05:
As you may notice [in my repro `.yaml` file](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/12163941569/workflow#L22), I am using the [Python Package Manager installation method](https://docs.platformio.org/en/latest/core/installation/methods/pypi.html#python-package-manager). From the docs, it says that it _doesn't_ use a venv. So that location I'm linking is really the only one with platformio.

in the `~/.platformio` folder, there is nothing else than `appstate.json` and a  `packages` subfolder.

CocoDico78, 2024-12-05:
Hey @Jason2866 and thanks for your involvement yesterday, I feel like we're now touching on the heart of the issue and I've tried your suggestion to

> Call the command with path and it will work

with the `venv` but it still doesn't work. Let me know if you need any more evidence of the issue.

CocoDico78, 2025-03-07:
Just a message to prevent this issue to go stale. The issue is still present in Github Actions.

valeros, 2025-03-07:
I believe you need to specify the `.py` extension when invoking a python script:
```bash
~/.platformio/penv/bin/pio pkg exec --package "platformio/tool-esptoolpy" -- esptool.py version
```
Note that esptoolpy depends on the `intelhex` package, so you need to install it beforehand.

CocoDico78, 2025-03-07:
@valeros your insight "fixed" it, as demonstrated in [this GH log](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/13725193300/job/38389831629#step:5:25), ie this is working:
> ~/.platformio/penv/bin/pip install intelhex
> ~/.platformio/penv/bin/pio pkg exec --package "platformio/tool-esptoolpy" -- esptool.py version

I think it is problematic that `intelhex` has to be installed by hand. I also think it's an issue in itself that calling `esptool` directly doesn't work and we are forced to call `esptool.py`. I also think it's an issue that we have to make multiple calls into esoteric paths to get a simple package command working.

valeros, 2025-03-10:
Hi @CocoDico78, thanks for feedback.

> I think it is problematic that intelhex has to be installed by hand.

Support for the latest esptoolpy v4.8.1 is experimental. The current stable version of the `espressif32` dev-platform uses 4.5.1 that doesn't depend on the intelhex package. I recommend using it.

> I also think it's an issue in itself that calling esptool directly doesn't work and we are forced to call esptool.py.

It doesn't work mainly because there is simply no such file `esptool`, only `esptoolpy`. Besides, CLI commands that deal with Python scripts are handled in a special way, so the `.py` extension is mandatory.  

> I also think it's an issue that we have to make multiple calls into esoteric paths to get a simple package command working.

Here is the recommended way without esoteric parts and intelhex dependency:
```yaml
jobs:
   ...
      - name: Install PlatformIO
         run: python3 -m pip install -U platformio
      
      - name: Run esptoolpy
         run: pio pkg exec --package "platformio/tool-esptoolpy@~1.40501.0" -- esptool.py version
```

CocoDico78, 2025-03-10:
@valeros absolutely brilliant, thanks! This is exactly the kind of resolution I could have hoped for. Thanks @Jason2866 for your early involvement as well.

Final [GH log](https://github.com/CocoDico78/PlatformIoEsptoolActionsFail/actions/runs/13764975514/job/38489210633#step:5:25) to show the output after applying resolution step.

closed 2025-03-10 by CocoDico78