platformio/platform-espressif32 #1502: Request for Espressif ESP32-S2-DevKitC-1-N8R2 Board

dustinlaroche, 2024-12-04:
What kind of issue is this?

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

------------------------------------------------------------------

### Description of problem

I would like to see the Espressif ESP32-S2-DevKitC-1-N8R2 board being supported for development using PlatformIO. Thanks!

ForrestFire0, 2024-12-17:
I am also interested in developing with a similar board on platformio. 

Here's the board: [ESP32-S2-DevKitM-1U-N4R2](https://www.amazon.com/dp/B0B97H9D4F?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1)

ForrestFire0, 2024-12-17:
I modified the S2 Solad board definition to work with my dev board.

Here's the relevant bits
```ini
board = esp32-s2-saola-1

board_build.name = Espressif ESP32-S2-DevKitM-1U-N4R2
board_build.url = https://www.espressif.com/en/products/devkits/esp32-s2-devkitm-1

board_build.extra_flags =
    -DARDUINO_ESP32S2_DEV
    -DARDUINO_USB_MODE=1
    -DARDUINO_RUNNING_CORE=1
    -DARDUINO_EVENT_RUNNING_CORE=1
board_build.psram = enabled
```

With my entire config being
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[common]

[env:esp32dev]
monitor_speed = 115200
board = esp32-s2-saola-1

board_build.name = Espressif ESP32-S2-DevKitM-1U-N4R2
board_build.url = https://www.espressif.com/en/products/devkits/esp32-s2-devkitm-1

board_build.extra_flags =
    -DARDUINO_ESP32S2_DEV
    -DARDUINO_USB_MODE=1
    -DARDUINO_RUNNING_CORE=1
    -DARDUINO_EVENT_RUNNING_CORE=1
board_build.psram = enabled

platform = espressif32 @ 6.7.0
framework = arduino
; Below are github packages but are frozen with a release.
platform_packages=
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/3.0.2/esp32-3.0.2.zip
    framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.2/esp32-arduino-libs-3.0.2.zip
    toolchain-riscv32-esp @ 12.2.0+20230208
    toolchain-xtensa-esp32s2 @ 12.2.0+20230208

lib_deps =
    ArduinoJson
monitor_filters = esp32_exception_decoder
```

I have found the platform_packages version pinning to crucial to getting this to work. But this just uploaded! IDK about PSRAM yet or any other peripherals.

dustinlaroche, 2025-01-07:
Thank you for your time answering this question. I was wondering if it is possible to have the board work with ESP-IDF?