platformio/platform-espressif32 #1507: Request: ESP32-C6 ULP support in PlatformIO.

morcibacsi, 2024-12-07:
ULP example code does not build for ESP32-C6 under PlatformIO.

Steps to reproduce:

Download official LP Core example from here:
https://github.com/espressif/esp-idf/tree/v5.3.2/examples/system/ulp/lp_core/gpio

I added the following platformio.ini:
```
[platformio]
core_dir = C:\.platformio
src_dir = main

[env:esp32-c6-devkitm-1]
platform = espressif32@6.9.0
board = esp32-c6-devkitm-1
framework = espidf
upload_port = COM29
build_flags = -v
```

The error I got is the following:

```main/lp_core_gpio_example_main.c:20:10: fatal error: ulp_main.h: No such file or directory```

This file should be autogenerated at this location:
```
 .pio\build\esp32-c6-devkitm-1\esp-idf\main\ulp_main\
```
The issue is similar to this: #1144  but downloading the test files from there reveals ulp programs in .s files.

Also checked the layout of the ESP32-S2 and S3 examples here, but those have CONFIG_ULP_COPROC_TYPE_RISCV=y in the sdkconfig however the C6 has CONFIG_ULP_COPROC_TYPE_LP_CORE=y
https://github.com/Jason2866/platform-espressif32/tree/Arduino/IDF5/examples/espidf-ulp-riscv

<details>
<summary>Full log</summary>

```
Executing task: C:\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-c6-devkitm-1 (platform: espressif32@6.9.0; board: esp32-c6-devkitm-1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitm-1.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-C6-DevKitM-1
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-c6-devkitm-1\main\lp_core_gpio_example_main.c.o
*** [.pio\build\esp32-c6-devkitm-1\.pio\build\esp32-c6-devkitm-1\ulp_main.bin.S.o] Source `.pio\build\esp32-c6-devkitm-1\ulp_main.bin.S' not found, needed by target `.pio\build\esp32-c6-devkitm-1\.pio\build\esp32-c6-devkitm-1\ulp_main.bin.S.o'.
Using built-in specs.
[... 178 lines cut ...]
 C:/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/include
 C:/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/include-fixed
 C:/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/include
End of search list.
Compiler executable checksum: 14ee240e1ec1d8c9d8c826325e8e087c
main/lp_core_gpio_example_main.c:20:10: fatal error: ulp_main.h: No such file or directory

******************************************************************
* Looking for ulp_main.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ulp_main.h"
* Web  > https://registry.platformio.org/search?q=header:ulp_main.h
*
******************************************************************

   20 | #include "ulp_main.h"
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32-c6-devkitm-1\main\lp_core_gpio_example_main.c.o] Error 1
=========================================================================== [FAILED] Took 3.43 seconds ===========================================================================

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```
</details>

Jason2866, 2024-12-07:
@morcibacsi  You have to use my branch https://github.com/Jason2866/platform-espressif32/tree/Arduino/IDF53 for support of all LP cores

morcibacsi, 2024-12-07:
Thanks for the quick reply. Unfortunately after updating platformio.ini to point your branch, I still have the same error.

```
[platformio]
core_dir = C:\.platformio
src_dir = main

[env:esp32-c6-devkitm-1]
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53
board = esp32-c6-devkitm-1
framework = espidf
upload_port = COM29
build_flags = -v
```

<details>
<summary>Full log</summary>

```
 *  Executing task: C:\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-c6-devkitm-1 (platform: https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53; board: esp32-c6-devkitm-1; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitm-1.html
PLATFORM: Espressif 32 (2024.11.31+sha.b66496e) > Espressif ESP32-C6-DevKitM-1
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.3 
 - tool-mklittlefs @ 3.2.0
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-scons @ 4.40801.0 (4.8.1)
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Generating assembly for certificate bundle...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-c6-devkitm-1\main\lp_core_gpio_example_main.c.o
*** [.pio\build\esp32-c6-devkitm-1\.pio\build\esp32-c6-devkitm-1\ulp_main.bin.S.o] Source `.pio\build\esp32-c6-devkitm-1\ulp_main.bin.S' not found, needed by target `.pio\build\esp32-c6-devkitm-1\.pio\build\esp32-c6-devkitm-1\ulp_main.bin.S.o'.
Using built-in specs.
COLLECT_GCC=riscv32-esp-elf-gcc
Target: riscv32-esp-elf
Configured with: /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-host_w64-mingw32 --target=riscv32-esp-elf --prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf --exec_prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf/riscv32-esp-elf --with-sysroot=/builds/idf/crosstool-NG/builds/riscv32-esp-elf/riscv32-esp-elf --with-native-system-header-dir=/include --with-newlib --enable-threads=no --disable-shared --with-arch=rv32gc --with-abi=ilp32 --with-pkgversion='crosstool-NG esp-13.2.0_20240530' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-libstdcxx-verbose --with-gmp=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/buildtools/complibs-host --with-mpfr=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/buildtools/complibs-host --with-mpc=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/buildtools/complibs-host --with-isl=/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/buildtools/complibs-host --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes --with-gnu-ld
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (crosstool-NG esp-13.2.0_20240530)
COLLECT_GCC_OPTIONS='-o' '.pio/build/esp32-c6-devkitm-1/main/lp_core_gpio_example_main.c.o' '-c' '-Og' '-Wno-error=deprecated-declarations' '-Wno-error=unused-but-set-variable' '-Wno-error=unused-function' '-Wno-error=unused-variable' '-fmacro-prefix-map=C:/.platformio/packages/framework-espidf=/IDF' '-fmacro-prefix-map=C:/GIT/espidf/gpio_platformio=.' '-gdwarf-4' '-ggdb' '-nostartfiles' '-std=gnu17' '-v' '-march=rv32imac_zicsr_zifencei' '-ffunction-sections' '-fdata-sections' '-Wno-error=unused-function' '-Wno-error=unused-variable' '-Wno-error=unused-but-set-variable' '-Wno-error=deprecated-declarations' '-Wno-unused-parameter' '-Wno-sign-compare' '-Wno-enum-conversion' '-gdwarf-4' '-ggdb' '-nostartfiles' '-Og' '-fno-shrink-wrap' '-fmacro-prefix-map=C:/GIT/espidf/gpio_platformio=.' '-fmacro-prefix-map=C:/.platformio/packages/framework-espidf=/IDF' '-fno-lto' '-fstrict-volatile-bitfields' '-fno-jump-tables' '-fno-tree-switch-conversion' '-Wno-old-style-declaration' '-D' '_POSIX_READER_WRITER_LOCKS' '-D' '_GNU_SOURCE' '-D' '_GLIBCXX_USE_POSIX_SEMAPHORE' '-D' '_GLIBCXX_HAVE_POSIX_SEMAPHORE' '-D' 'SOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ' '-D' 'SOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE' '-D' 'IDF_VER="5.3.2"' '-D' 'ESP_PLATFORM' '-D' 'PLATFORMIO=60116' '-I' 'C:/.platformio/packages/framework-espidf/components/riscv/include' '-I' '.pio/build/esp32-c6-devkitm-1/config' '-I' 'C:/.pl [... cut ...]
   20 | #include "ulp_main.h"
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32-c6-devkitm-1\main\lp_core_gpio_example_main.c.o] Error 1
============================================================ [FAILED] Took 20.46 seconds ============================================================

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

</details>

Jason2866, 2024-12-08:
Looks like you hit the Windows path length limit. Sorry I have no solution for this issue with Windows.

morcibacsi, 2024-12-08:
That's very unfortunate 🫤 I'll try to use a VM or WSL then. I couldn't have figured that out from the log though...

Thanks.

morcibacsi, 2024-12-09:
@Jason2866 

I just installed Ubuntu 24.10 in a VM, and unfortunately the issue is the same.

<details>
<summary>Full log</summary>

```
 *  Executing task: platformio run 

Processing esp32-c6-devkitm-1 (platform: https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53; board: esp32-c6-devkitm-1; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitm-1.html
PLATFORM: Espressif 32 (2024.11.31+sha.b66496e) > Espressif ESP32-C6-DevKitM-1
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50302.0 (5.3.2) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.3 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/esp32-c6-devkitm-1/main/lp_core_gpio_example_main.c.o
*** [.pio/build/esp32-c6-devkitm-1/.pio/build/esp32-c6-devkitm-1/ulp_main.bin.S.o] Source `.pio/build/esp32-c6-devkitm-1/ulp_main.bin.S' not found, needed by target `.pio/build/esp32-c6-devkitm-1/.pio/build/esp32-c6-devkitm-1/ulp_main.bin.S.o'.
Using built-in specs.
COLLECT_GCC=riscv32-esp-elf-gcc
Target: riscv32-esp-elf
Configured with: /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=riscv32-esp-elf --prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf --exec_prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/riscv32-esp-elf/riscv32-esp-elf --with-sysroot=/builds/idf/crosstool-NG/builds/riscv32-esp-elf/riscv32-esp-elf --with-native-system-header-dir=/include --with-newlib --enable-threads=no --disable-shared --with-arch=rv32gc --with-abi=ilp32 --with-pkgversion='crosstool-NG esp-13.2.0_20240530' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-libstdcxx-verbose --with-gmp=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/riscv32-esp-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (crosstool-NG esp-13.2.0_20240530) 
COLLECT_GCC_OPTIONS='-o' '.pio/build/esp32-c6-devkitm-1/main/lp_core_gpio_example_main.c.o' '-c' '-Og' '-Wno-error=deprecated-declarations' '-Wno-error=unused-but-set-variable' '-Wno-error=unused-function' '-Wno-error=unused-variable' '-fmacro-prefix-map=/home/p/.platformio/packages/framework-espidf=/IDF' '-fmacro-prefix-map=/home/p/Documents/gpio_platformio=.' '-gdwarf-4' '-ggdb' '-nostartfiles' '-std=gnu17' '-v' '-march=rv32imac_zicsr_zifencei' '-ffunction-sections' '-fdata-sections' '-Wno-error=unused-function' '-Wno-error=unused-variable' '-Wno-error=unused-but-set-variable' '-Wno-error=deprecated-declarations' '-Wno-unused-parameter' '-Wno-sign-compare' '-Wno-enum-conversion' '-gdwarf-4' '-ggdb' '-nostartfiles' '-Og' '-fno-shrink-wrap' '-fmacro-prefix-map=/home/p/Documents/gpio_platformio=.' '-fmacro-prefix-map=/home/p/.platformio/packages/framework-espidf=/IDF' '-fno-lto' '-fstrict-volatile-bitfields' '-fno-jump-tables' '-fno-tree-switch-conversion' '-Wno-old-style-declaration' '-D' '_POSIX_READER_WRITER_LOCKS' '-D' '_GNU_SOURCE' '-D' '_GLIBCXX_USE_POSIX_SEMAPHORE' '-D' '_GLIBCXX_HAVE_POSIX_SEMAPHORE' '-D' 'SOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ' '-D' 'SOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE' '-D' 'IDF_VER="5.3.2"' '-D' 'ESP_PLATFORM' '-D' 'PLATFORMIO=60116' '-I' '/home/p/.platformio/packages/framework-espidf/components/riscv/include' '-I' '.pio/build/esp32-c6-devkitm-1/config' '-I' '/home/p/.platformio/packages/framework-espidf/components/newlib/platform_include' '-I' '/home/p/.platformio/packages/framework-espidf/components/freertos/config/include' '-I' '/home/p/.platformio/packages/framework-espidf/components/freertos/config/include/freertos' '-I' '/home/p/.platformio/packages/framework-espidf/components/freertos/config/riscv/include' '-I' '/home/p/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/include' '-I' '/home/p/.platformio/packages/framework-espidf/component [... cut ...]
******************************************************************

   20 | #include "ulp_main.h"
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio/build/esp32-c6-devkitm-1/main/lp_core_gpio_example_main.c.o] Error 1
=============================================================== [FAILED] Took 2.50 seconds ===============================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```
</details>

Jason2866, 2024-12-09:
Mhh, this [example](https://github.com/pioarduino/platform-espressif32/tree/Arduino/IDF53/examples/espidf-ulp-lp) compiles and is verified working.

morcibacsi, 2024-12-09:
Thanks, this works! Even from Windows 11. Comparing the files from there with mine I guess either the original CMakeLists.txt files are the cause as yours are slightly different (even though I didn't modify the files from original repo) or the fact that I tried to use a different folder structure.  Anyways, thanks for the help, now I can build my stuff based on your example.

Jason2866, 2024-12-10:
The folder structure is the reason. You can't freely choose.

commit 1442a35a (2025-01-13): Update ULP build script

- Added support for ESP32-C6 ULP
- Updates according to changes in IDF v5.4.0

Resolves #1507
closed 2025-01-13 by valeros