platformio/platform-espressif32 #1510: Setting `platform_packages` per board

and3rson, 2024-12-09:
We have recently merged a backport of an important SPI bus fix into ESP32 Arduino core (version 2.x): https://github.com/espressif/arduino-esp32/pull/10697

This fix targets v2.x of ESP32 Arduino core. However, since version 2.x is no longer supported/released, the only way to use the latest 2.x version (as pointed by me-no-dev and Jason2866 [here](https://github.com/espressif/arduino-esp32/issues/9221#issuecomment-2527698052)) is to target `release/2.x` branch as such:

```
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/archive/refs/heads/release/v2.x.zip
```

This is error-prone since it overrides the default framework version from platformio. So it seems there are two solutions:

1. Define `platform_packages` on a board level (specifically we're interested in [Lilka v2](https://github.com/platformio/platform-espressif32/blob/develop/boards/lilka_v2.json) which is severely affected and would benefit a lot from the SPI bus fix).
   I'm not sure if this is generally possible though.
2. Update espressif32 platform to target `release/v2.x` branch. This would be the preferred way to go, since it would solve SPI bus contention issues for everyone who's using PlatformIO with ESP32-S3/C3.

Unfortunately, due to v2.x not being maintained anymore, there won't be v2.0.19 release - the last one is v2.0.18 which does not include our SPI fix.

Please let me know what you think. Any thoughts are welcome!

valeros, 2024-12-12:
Hi @and3rson, thanks for reporting. I've updated the default `framework-arduinoespressif32` package with some cherry-picked patches from the `release/v2.x` branch. You can run `pio pkg update -g -p espressif32` on your machine to pull the latest changes.

and3rson, 2024-12-12:
@valeros This is wonderful, thank you!