platformio/platform-espressif32 #1514: esp32 managed components are not re-downloaded if removed

fariouche, 2024-12-12:
Hello,

When using managed_components (for example to ass tinyusb support to an esp32S3), it is downloading the components sources in managed_components and everything compiles just fine.
However, if the folder is deleted and we relaunch "pio run", it will not re-download de sources and will fail compilation about missing dependencies. Removing the build.ninja find or deleting .pio fixes the issue.

In my case I have 2 envs, one esp32 and one esp32s3 and only esp32s3 has the components.
If I switch to esp32 env, the folder content gets deleted, and this is how I discovered the issue.

I think something is missing in the dependencies or the management of managed_components.

Thanks!

Jason2866, 2024-12-16:
Have you deleted the `dependencies lock` file?

fariouche, 2024-12-16:
no, the file remain in src/idf_component.yml and the lock file is still in the root project directory

Jason2866, 2024-12-16:
Does it download when you delete the `dependencies.lock` file?

fariouche, 2024-12-17:
no, it does not.
The only way I've found it to retrigger a cmake configure (by deleting the build.ninja file)

valeros, 2024-12-18:
Hi @fariouche, 

> If I switch to esp32 env, the folder content gets deleted, and this is how I discovered the issue.

That's strange, this folder shouldn't be deleted automatically. Could you please provide a minimal example to reproduce the issue?

fariouche, 2024-12-18:
Sure! Here is a very small project that reproduce the issue:
[pio_idf_component.zip](https://github.com/user-attachments/files/18186068/pio_idf_component.zip)

First, build the Sample_esp32_s3_usb_otg env, then see that managed_component folder is created.
Then build Sample_esp32_devkit_c, and see that the content of the managed_component folder is deleted.
If you build the esp32_s3 again, it will fail this time.

My project supports esp32 and s3, and when we build the s3 variant the project adds usb support (that the esp32 does not support)

valeros, 2024-12-19:
Thanks @fariouche for reporting, should be fixed in the development branch.

closed 2024-12-19 by valeros

fariouche, 2024-12-19:
great! thanks!

commit 492d38ba (2024-12-19): Reconfigure IDF project if dependencies changed

Resolves #1514