platformio/platform-espressif32 #1518: esp32-s3-devkitc-1-n16r8v board missing

CEHansen, 2024-12-13:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [x] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- **[ ] **Development Platform or Board**.**
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Win 11
**PlatformIO Version** (`platformio --version`):
Version: 1.96.0 (user setup)
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Windows_NT x64 10.0.22631

### Description of problem
esp32-s3-devkitc-1-n16r8v board missing

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

Can be found here:

https://community.platformio.org/t/how-do-you-set-up-a-new-board-esp32-s3-n16r8/32306/3

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

valeros, 2024-12-16:
Duplicate https://github.com/platformio/platform-espressif32/issues/989

closed 2024-12-16 by valeros

Jason2866, 2024-12-16:
Here is a nice page for setting up S3 boards https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations?tab=readme-ov-file#esp32-s3-wroom-11u-n16r8