platformio/platform-espressif32 #1519: captive portals don't work

BeaverUI, 2024-12-17:
posted this issue just for reference; please see my issue posted at https://github.com/pioarduino/platform-espressif32/issues/78

vortigont, 2024-12-18:
Have you checked [CaptivePortal](https://github.com/espressif/arduino-esp32/tree/master/libraries/DNSServer/examples/CaptivePortal) example? it works fine without DHCP option manupulation. DHCP option captive portal is not used much and can't be relied on from the application side. Most of those use to "ping" some known urls.

valeros, 2024-12-18:
> It's not project-specific, I'm requesting a feature to the library to allow setting dhcps options by making the esp_netifs variable available for reading (perhaps through a ::getNetIfs function) by the main code.

@BeaverUI PlatformIO is a development environment, it has nothing to do with libraries you're using in your project. I'd recommend forwarding your issue to a proper repository, probably this one https://github.com/espressif/arduino-esp32

BeaverUI, 2024-12-19:
Now I feel stupid. Please close this one, I will report there 👍