platformio/platform-espressif32 #1520: Compilation error when using PlatformIO and Ethernet.h

MicSG-dev, 2024-12-19:
When using PlatformIO with the following platformio.ini file:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    arduino-libraries/Ethernet@^2.0.2
```
And running the [WebServer](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino) example from Ethernet.h, several compilation errors occur:
```
src/main.cpp: In function 'void loop()':
src/main.cpp:75:43: error: cannot allocate an object of abstract type 'EthernetClient'
   75 |   EthernetClient client = server.available();
      |                           ~~~~~~~~~~~~~~~~^~
In file included from src/main.cpp:22:
.pio/libdeps/esp32dev/Ethernet/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
  214 | class EthernetClient : public Client {
      |       ^~~~~~~~~~~~~~
In file included from C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Arduino.h:197,
                 from src/main.cpp:20:
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
src/main.cpp:75:18: error: cannot declare variable 'client' to be of abstract type 'EthernetClient'
   75 |   EthernetClient client = server.available();
      |                  ^~~~~~
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:40:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::available()'
   40 | EthernetClient EthernetServer::available()
      |                ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:22:
.pio/libdeps/esp32dev/Ethernet/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
  214 | class EthernetClient : public Client {
      |       ^~~~~~~~~~~~~~
In file included from C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Arduino.h:197,
                 from .pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:21:
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32@src-e5a7fd5d9117bd8a322277fde59ac3d3/cores/esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::available()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:47:54: error: invalid cast to abstract class type 'EthernetClient'
   47 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:72:40: error: invalid cast to abstract class type 'EthernetClient'
   72 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: At global scope:
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:75:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::accept()'
   75 | EthernetClient EthernetServer::accept()
      |                ^~~~~~~~~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::accept()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:82:54: error: invalid cast to abstract class type 'EthernetClient'
   82 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:104:40: error: invalid cast to abstract class type 'EthernetClient'
  104 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'virtual size_t EthernetServer::write(const uint8_t*, size_t)':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:170:18: error: cannot allocate an object of abstract type 'EthernetClient'
  170 |         available();
      |         ~~~~~~~~~^~
*** [.pio\build\esp32dev\libe72\Ethernet\EthernetServer.cpp.o] Error 1
```

Jason2866, 2024-12-20:
The library you try to use is not compatible. This is not a Platformio nor an Arduino core issue.

MicSG-dev, 2024-12-20:
> The library you try to use is not compatible. This is not a Platformio nor an Arduino core issue.

See my response to this incorrect statement at: https://github.com/espressif/arduino-esp32/issues/10760#issuecomment-2557417697

MicSG-dev, 2024-12-20:
> This is not a Platformio nor an Arduino core issue.

I don't know what happened with the [Client.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Client.h) file (from the esp32 core), but according to @per1234 (https://github.com/arduino-libraries/Ethernet/issues/277#issuecomment-2555774494) there seems to have been some mishap that causes the problem I'm facing.

---
On my local machine I made the changes as per the pull request https://github.com/espressif/arduino-esp32/pull/3191 and it temporarily resolved the https://github.com/espressif/arduino-esp32/issues/10760#issue-2751313443 that I was facing.

Changes made: https://github.com/arduino-libraries/Ethernet/issues/277#issuecomment-2555815432

Jason2866, 2024-12-21:
@MicSG-dev The issue is not caused from the Platformio build system. It is either in Arduino Core or in the used library. In both cases this repo is not responsible for.

valeros, 2024-12-23:
Closing as not related to PlatformIO.