platformio/platform-espressif32 #1522: Windows: Generated command exceeds windows 32k limit and then fails

zackees, 2024-12-27:
Hi there, I'm a dev on FastLED. I'm cannot build esp32c6 on windows due to command limit of 32k being exceeded:

# Repro steps.

  * `git clone https://github.com/fastled/fastled`
  * checkout commit `5442a57c6603baac823bac885f2f10753bca2ff9`
  * `pio run`

# Result

```
Compiling .pio\build\dev\lib354\fastled\FastLED.cpp.o
Compiling .pio\build\dev\lib354\fastled\bitswap.cpp.o
Compiling .pio\build\dev\lib354\fastled\cled_controller.cpp.o
Compiling .pio\build\dev\lib354\fastled\colorpalettes.cpp.o
Compiling .pio\build\dev\lib354\fastled\colorutils.cpp.o
Compiling .pio\build\dev\lib354\fastled\crgb.cpp.o
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
Compiling .pio\build\dev\lib354\fastled\five_bit_hd_gamma.cpp.o
*** [.pio\build\dev\lib354\fastled\bitswap.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\colorpalettes.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\FastLED.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\cled_controller.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\five_bit_hd_gamma.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\crgb.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/niteris/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\dev\lib354\fastled\colorutils.cpp.o] Error 1
```

This is the generated command, running this with `wc -m` shows that the size is 34k, larger than the 32k windows limit.

It would be helpful if a lot of these same path prefixes were instead folded into an environmental variable.

```
-o .pio/build/dev/lib354/fastled/bitswap.cpp.o -c -march=rv32imac_zicsr_zifencei -std=gnu++2b -fexceptions -fno-rtti -std=gnu++17 -Werror=return-type -Os -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -Wno-error=uninitialized -Wno-error=maybe-uninitialized -Wno-format-nonliteral -Wno-missing-field-initializers -Wno-error=array-bounds -Wno-write-strings "-DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=<lib/address_resolve/AddressResolve_DefaultImpl.h>" -MMD -DPLATFORMIO=60116 -DESP32_ARDUINO_LIB_BUILDER -DESP_PLATFORM -DIDF_VER=\"v5.3.2-174-g083aad99cf-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMD5_ENABLED=1 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DOPENTHREAD_FTD=1 -DOPENTHREAD_PROJECT_LIB_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DSERIAL_FLASHER_BOOT_HOLD_TIME_MS=50 -DSERIAL_FLASHER_RESET_HOLD_TIME_MS=100 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -DTF_LITE_STATIC_MEMORY -DCHIP_CONFIG_SOFTWARE_VERSION_NUMBER=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=0 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_DNSSD_DEFAULT_MINIMAL=1 -DARDUINO_ARCH_ESP32 -DESP32=ESP32 -DF_CPU=160000000L -DARDUINO=10812 -DARDUINO_VARIANT=\"esp32c6\" "-DARDUINO_BOARD=\"Espressif ESP32-C6-DevKitC-1\"" -DARDUINO_PARTITION_default -Isrc -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/newlib/platform_include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/config/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/config/include/freertos -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/config/riscv/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/FreeRTOS-Kernel/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/FreeRTOS-Kernel/portable/riscv/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/freertos/esp_additions/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/include/soc -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/include/soc/esp32c6 -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/dma/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/ldo/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/port/esp32c6 -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/port/esp32c6/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_hw_support/port/esp32c6/private_include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/heap/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/log/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/soc/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/soc/esp32c6 -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/soc/esp32c6/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/hal/platform_port/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/hal/esp32c6/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/hal/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_rom/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_rom/include/esp32c6 -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_rom/esp32c6 -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_common/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_system/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/esp_system/port/soc -IC:/Users/niteris/.platformio/pack [... cut ...] s/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/joltwallet__littlefs/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/fb_gfx/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/qio_qspi/include -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32/cores/esp32 -IC:/Users/niteris/.platformio/packages/framework-arduinoespressif32/variants/esp32c6 src/bitswap.cpp
```

zackees, 2025-01-09:
I'm constantly hitting this every day.

zackees, 2025-01-09:
We have users. github.com/fastled/fastled

Jason2866, 2025-01-09:
agree with @zackees it is a Platformio issue. When providing for Windows it has to work.
As it is now Windows is simply broken. No big project can be compiled when using Windows as OS

joba-1, 2025-01-09:
Today I read somewhere platformio is „just“ a package manager, now it is a build system :)

AFAIK platformio uses scons for that.
Scons has TempFileMunge command to shorten long command lines.
Maybe look into that?

zackees, 2025-01-10:
I built all of FastLED's massive testing infrastructure on platformio. I built a web compiler for FastLED using platformio.

I've tuned platformio nearly to the maximum performance on linux with through docker. It's so much faster than running on windows. Especially when you throw on ccache, then pre-warm it, then when it's ready to go full steam, the image get's frozen, and shipped to the cloud for distribution.

It's an amazing piece of technology - way better than Arduino in pretty much every conceivable way. I make no hesitation to say that FastLED would not be where it is today if it weren't for platformio. I tried to get an arduino build going for some parity and gave up.

zackees, 2025-01-14:
> agree with @zackees it is a Platformio issue. When providing for Windows it has to work.
> 
> As it is now Windows is simply broken. No big project can be compiled when using Windows as OS
> 
> Despite the fact this example is from a fork (which is not the reason for failure) here Windows tells it has a problem, Normally unrelated error messages appear.
> 
> https://github.com/pioarduino/platform-espressif32/actions/runs/12690624596/job/35371889939

@Jason2866 

Since espressif doesn't want to pay Ivan for espressif support I think this issue will remain a constant problem.

What would the possibility be that espressif can fix this and push a patch?

Jason2866, 2025-01-14:
@zackees I can't speak for espressif nor for Platformio. If you want to see this fixed, you have to do yourself.

valeros, 2025-01-14:
It's a WinAPI limitation. The `CreateProcess` function imposes a limit on the length of the CLI string that can be passed to a new process. Historically, this limit has been 32,768 characters (bytes), which includes the program name and the arguments.

The only workaround I'm aware of is to use special include flags that utilize a common include prefix. To test it out you can create an extra script, for example `shorten_includes.py` with the following content:

```python
import os

from platformio import fs

Import("env")

platform = env.PioPlatform()

FRAMEWORK_DIR = fs.to_unix_path(
    platform.get_package_dir("framework-arduinoespressif32")
)

IS_INTEGRATION_DUMP = env.IsIntegrationDump()

def shorthen_includes(env, node):
    if IS_INTEGRATION_DUMP:
       # Don't shorten include paths for IDE integrations
       return node
       
    includes = [fs.to_unix_path(inc) for inc in env.get("CPPPATH", [])]
    shortened_includes = []
    generic_includes = []
    for inc in includes:
        if inc.startswith(FRAMEWORK_DIR):
            shortened_includes.append(
                "-iwithprefix/"
                + fs.to_unix_path(os.path.relpath(inc, FRAMEWORK_DIR))
            )
        else:
            generic_includes.append(inc)
 
    return env.Object(
        node,
        CPPPATH=generic_includes,
        CCFLAGS=env["CCFLAGS"] + ["-iprefix", FRAMEWORK_DIR] + shortened_includes
    )

env.AddBuildMiddleware(shorthen_includes)
```

Then enable this script in your `platformio.ini`, for example:
```ini
[env:lolin32]
platform = espressif32
framework = arduino
board = lolin32
monitor_speed = 115200
extra_scripts = 
    pre:shorten_includes.py
```

Jason2866, 2025-01-15:
@valeros Thx for the code snipplet. The approach is to invasive. More complex includes in libs are not working anymore.

valeros, 2025-01-15:
> More complex includes in libs are not working anymore.

@Jason2866 Mind sharing an example of a complex include that breaks the code above?

Jason2866, 2025-01-15:
@valeros  The include `sdkconfig.h` is not found anymore here: https://github.com/arendst/Tasmota/blob/development/lib/libesp32_lvgl/lvgl/src/lv_conf_kconfig.h

valeros, 2025-01-16:
@Jason2866 I believe I know what went wrong, but it's a bit hard to configrm based only on scarce error message from the log, especially when a big project like Tasmota is compiled. Any chance you could put together a minimal example to reproduce the issue?

Jason2866, 2025-01-16:
@valeros  Here is a small example https://github.com/Jason2866/pio-shortpath

valeros, 2025-01-16:
@Jason2866 Thanks for the example, helped a lot. Here is an improved version of the script:
```python
import os

from platformio import fs

Import("env")

platform = env.PioPlatform()
target = env.BoardConfig().get("build.mcu", "esp32")

FRAMEWORK_SDK_DIR = fs.to_unix_path(
    os.path.join(
        platform.get_package_dir("framework-arduinoespressif32"),
        "tools",
        "sdk",
        target,
        "include",
    )
)

IS_INTEGRATION_DUMP = env.IsIntegrationDump()

def is_framework_subfolder(potential_subfolder):
    if (
        os.path.splitdrive(FRAMEWORK_SDK_DIR)[0]
        != os.path.splitdrive(potential_subfolder)[0]
    ):
        return False
    return os.path.commonpath([FRAMEWORK_SDK_DIR]) == os.path.commonpath(
        [FRAMEWORK_SDK_DIR, potential_subfolder]
    )

def shorthen_includes(env, node):
    if IS_INTEGRATION_DUMP:
        # Don't shorten include paths for IDE integrations
        return node

    includes = [fs.to_unix_path(inc) for inc in env.get("CPPPATH", [])]
    shortened_includes = []
    generic_includes = []
    for inc in includes:
        if is_framework_subfolder(inc):
            shortened_includes.append(
                "-iwithprefix/"
                + fs.to_unix_path(os.path.relpath(inc, FRAMEWORK_SDK_DIR))
            )
        else:
            generic_includes.append(inc)

    return env.Object(
        node,
        CPPPATH=generic_includes,
        CCFLAGS=env["CCFLAGS"]
        + ["-iprefix", FRAMEWORK_SDK_DIR]
        + shortened_includes,
        ASFLAGS=env["ASFLAGS"]
        + ["-iprefix", FRAMEWORK_SDK_DIR]
        + shortened_includes,
    )

env.AddBuildMiddleware(shorthen_includes)

```

Jason2866, 2025-01-16:
@valeros Thx for the advanced script. Included directly in the platform of my test fork branch.
It generates an error when compiling Tasmota in CI
```
ValueError: Can't mix absolute and relative paths:
  File "/opt/hostedtoolcache/Python/3.11.11/x64/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
```
The code where the error happens
```
  File "/home/runner/.platformio/platforms/espressif32/builder/frameworks/arduino.py", line 234:
    if is_framework_subfolder(inc):
  File "/home/runner/.platformio/platforms/espressif32/builder/frameworks/arduino.py", line 220:
    return os.path.commonpath([FRAMEWORK_SDK_DIR]) == os.path.commonpath(
```
https://github.com/Jason2866/platform-espressif32/actions/runs/12816490475/job/35737551124

File: https://github.com/Jason2866/platform-espressif32/blob/short_path/builder/frameworks/arduino.py

valeros, 2025-01-16:
@Jason2866  I'm not sure how that relative path looks like, but a simple check should suffice:
```python
def is_framework_subfolder(potential_subfolder):
    if not os.path.isabs(potential_subfolder):
        return False
    if (
        os.path.splitdrive(FRAMEWORK_SDK_DIR)[0]
        != os.path.splitdrive(potential_subfolder)[0]
    ):
        return False
    return os.path.commonpath([FRAMEWORK_SDK_DIR]) == os.path.commonpath(
        [FRAMEWORK_SDK_DIR, potential_subfolder]
    )
```

Jason2866, 2025-01-16:
@valeros Great! Tasmota does compile successfully with the script! Going to test the Windows example where compile always failed with the script! Thx so far. Will come back when i have done more tests!

Jason2866, 2025-01-17:
All tests successfull. Now code compiles on Windows where it failed before using this script.
Big thank you @valeros 
The complete working shorten include path Snipplet.
```python
import os

from platformio import fs

Import("env")

platform = env.PioPlatform()
target = env.BoardConfig().get("build.mcu", "esp32")

FRAMEWORK_SDK_DIR = fs.to_unix_path(
    os.path.join(
        platform.get_package_dir("framework-arduinoespressif32"),
        "tools",
        "sdk",
        target,
        "include",
    )
)

IS_INTEGRATION_DUMP = env.IsIntegrationDump()

def is_framework_subfolder(potential_subfolder):
    if not os.path.isabs(potential_subfolder):
        return False
    if (
        os.path.splitdrive(FRAMEWORK_SDK_DIR)[0]
        != os.path.splitdrive(potential_subfolder)[0]
    ):
        return False
    return os.path.commonpath([FRAMEWORK_SDK_DIR]) == os.path.commonpath(
        [FRAMEWORK_SDK_DIR, potential_subfolder]
    )

def shorthen_includes(env, node):
    if IS_INTEGRATION_DUMP:
        # Don't shorten include paths for IDE integrations
        return node

    includes = [fs.to_unix_path(inc) for inc in env.get("CPPPATH", [])]
    shortened_includes = []
    generic_includes = []
    for inc in includes:
        if is_framework_subfolder(inc):
            shortened_includes.append(
                "-iwithprefix/"
                + fs.to_unix_path(os.path.relpath(inc, FRAMEWORK_SDK_DIR))
            )
        else:
            generic_includes.append(inc)

    return env.Object(
        node,
        CPPPATH=generic_includes,
        CCFLAGS=env["CCFLAGS"]
        + ["-iprefix", FRAMEWORK_SDK_DIR]
        + shortened_includes,
        ASFLAGS=env["ASFLAGS"]
        + ["-iprefix", FRAMEWORK_SDK_DIR]
        + shortened_includes,
    )

env.AddBuildMiddleware(shorthen_includes)
```

Jason2866, 2025-01-25:
Use the script only for Windows. On Mac i see a dramatic increase in compile time. Probably the same happens under Linux (did not test).

zackees, 2025-03-08:
Espressif probably needs to consolidate their considerably complex header include logic.

Or else ESP32C6 is likely going to stop working with the latest idf 5.4.

This will probably be an issue with Arduino IDE soon too.

The 32k limit for windows sucks but it is what it is.

Jason2866, 2025-03-08:
@zackees the C6 works perfectly fine with ArduinoIDE and with Platformio

zackees, 2025-03-10:
hmm interesting

cpatulea, 2026-03-20:
Hi, I have been tracking down a problem with pioarduino build on Windows for weeks. I found that pioarduino has some Windows max command length middleware, first added January 2025 (https://github.com/pioarduino/platform-espressif32/commit/53117e04e80ec3f804fa7de42fdd7249a5746cf3), inspired by the ideas and code examples in this issue.

But, I also see that platformio has built-in support for very long command lines https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/piomaxlen.py using standard compiler `@file` syntax.

Any idea why the built-in platformio `@file` support was not sufficient here?

(There is a bad interaction between the current middleware in pioarduino and Meshtastic extra_scripts, but first I'm trying to see if the middleware is needed at all. When I disabled the middleware, the problem went away, platformio puts the long command-line in `@file`, and the build works successfully)