platformio/platform-espressif32 #1524: Conditional CMakeLists.txt SRC compilation

hanssbtn, 2024-12-31:
I am trying to choose which folder to compile based on a build flag in ```platformio.ini```:
```
[env]
platform = platformio/espressif32
board = esp32doit-devkit-v1
framework = arduino, espidf
monitor_raw = yes
monitor_speed = 115200
extra_scripts = pre:extra_scripts.py
build_flags = -v

[env:server]
build_flags = 
    -D ENVIRONMENT='"SERVER"'

[env:client]
build_flags = 
    -D ENVIRONMENT='"CLIENT"'
```

```CMakeLists.txt```:
```
if (DEFINED ENVIRONMENT AND ENVIRONMENT STREQUAL "SERVER")
    file(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/src/server/*.cpp)
elseif (DEFINED ENVIRONMENT AND ENVIRONMENT STREQUAL "CLIENT")
    file(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/src/client/*.cpp)
else()
    message(FATAL_ERROR "no environment set.")
endif()
```
However I can't seem to pass the ```ENVIRONMENT``` define flag to CMake. Is there a way to do this? I tried using ```extra_scripts.py``` but I can't figure out how to set the variable from the script. Any help would be appreciated.

valeros, 2025-01-02:
Hi @hanssbtn, your approach with `build_flags` won't work because this option is used for adding compiler or linker flags. Instead you can try to use an undocumented option `board_build.cmake_extra_args` that's used to pass additional settings for CMake, for example:

```ini
[env:client]
platform = espressif32
framework = espidf
board = esp32dev
monitor_speed = 115200

board_build.cmake_extra_args = -DCLIENT_APP=ON
```

closed 2025-01-02 by valeros

hanssbtn, 2025-01-08:
Right, thanks for the help @valeros