platformio/platform-espressif32 #1528: Wrong CONFIG_ESPTOOLPY_FLASHSIZE and CONFIG_ESPTOOLPY_FLASHSIZE_xxx in generated configs

puzrin, 2025-01-09:
**How to reproduce:**

1. Take any project with `sdkconfig.defaults`, and without env-specific full configs.
    - https://github.com/puzrin/reflow_micro/tree/master/firmware
2. Run build, pio will create full configs in root. See value of CONFIG_ESPTOOLPY_FLASHSIZE and CONFIG_ESPTOOLPY_FLASHSIZE_*. Those will be 2M instead of board real value.
    - I used `lolin_c3_mini` config.

What's wrong:

- Every build shows **Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB! Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!`**
- I did not found yet a real problems, but such warning is very annoying.
- Also see topic on your forum https://community.platformio.org/t/platformio-with-esp32-qualia-s3-rgb66-board-flash-size/37609

What's expected:

- Those values could be taken from `platform.ini` board info.

When project has multiple boards with the same flash size - this can be overriden in `sdkconfig.defaults`. But when flash sizes are different - things become tricky. Also, pinning that value via defaults is very fragile for future board changes.

valeros, 2025-01-09:
Hi @puzrin, even though I understand your frustration, but the native IDF development environment was set up that way by design. 

We're doing our best to keep things backward compatible with the native build system and to minimize any interference with the default configuration process implemented in the IDF environment. Passing any implicit default configuration flags doesn't seem right to me, that's why a warning is shown if there's a mismatch between the IDF and our tool settings. That warning won't appear if you properly configure a project via `sdkconfig.defaults` or via `menuconfig`. 

As a workaround, you can explicitly set a default IDF configuration per environment by passing the `SDKCONFIG_DEFAULTS` option, for example:

```ini
[env:esp32-s2-kaluga-1]
board = esp32-s2-kaluga-1
board_build.cmake_extra_args = 
    -DSDKCONFIG_DEFAULTS="sdkconfig.defaults.first"

[env:esp32-c6-devkitc-1]
board = esp32-c6-devkitc-1
board_build.cmake_extra_args = 
    -DSDKCONFIG_DEFAULTS="sdkconfig.defaults.second"
```

puzrin, 2025-01-10:
Ok. I migrated to SDK only a few days ago and may not be familiar with "right/expected" use. My expectations were based on these things:

1. All examples I've seen for IDF SDK have only `sdkconfig.defaults` file with significant keys.
2. I always expected PlatformIO to automate all setup steps after the project opened (users need only build/upload).

If you think it's better stay as is - let's close issue. Thank you for example with `cmake_extra_args`

---

Could you clarify similar point about partitions. Should I customize those via menuconfig or via platform.ini or both?

closed 2025-01-10 by puzrin

valeros, 2025-01-10:
> Could you clarify similar point about partitions. Should I customize those via menuconfig or via platform.ini or both?

The workflow with the partition files is quite similar to Arduino:

```ini
[env:esp32-s2-kaluga-1]
...
board_build.partitions = partitions_two_ota_large.csv
```