platformio/platform-espressif32 #1529: ESP32-S3-Zero (ESP32-S3FH4R2) flash/psram issue - board config unknown for platfromio.ini

Queequack, 2025-01-09:
I have three ESP32-S3-Zero (ESP32-S3FH4R2, should probably have Flash: 4MB QD and PSRAM: 2MB QD) from [Aliexpress](https://de.aliexpress.com/item/1005006862681936.html?spm=a2g0o.order_list.order_list_main.22.49585c5fMUk2hN&gatewayAdapt=glo2deu). Seems to be the same as mentioned [in thread #1318](https://github.com/platformio/platform-espressif32/issues/1318#issuecomment-1976970633). I have the same problem with all of them. I usually work with ESP32-S3 (ESP32 and others), I'm using PlatformIO and I run hundreds of applications with dozens of different boards.

First of all: The upload works as expected. After restarting the board, the serial monitor displays the following error message.
At the end of the error message: It seems that the PSRAM is not recognized. And if there is a partition problem like [in thread #1318](https://github.com/platformio/platform-espressif32/issues/1318#issuecomment-1979962600), I don't know how to solve it.

```
E (246) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377ac4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (196) opi psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (197) spiram: SPI RAM enabled but initialization failed. Bailing out.
E (215) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)
```

I've also tried [this setup from thread #1318](https://github.com/platformio/platform-espressif32/issues/1318#issuecomment-2490960327), [this setup](https://community.platformio.org/t/esp32-s3-zero-does-not-work-on-platformio/40297/10) and [this file](https://community.platformio.org/t/esp32-s3-zero-does-not-work-on-platformio/40297/12) without success.
This is my current platformio.ini, which also leads to the error (from [thread #1318](https://github.com/platformio/platform-espressif32/issues/1318#issuecomment-2345802175)).
```
; Flash: 4MB QD, PSRAM: 2MB QD
[env:ESP32-S3-ZERO-TEST]
platform = espressif32
framework = arduino
monitor_speed = 115200
board = esp32-s3-devkitc-1
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM

board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
```

The sketch doesn't really play a role. I just want a simple serial output for test purposes.
```
#include <Arduino.h>

void setup() 
{
  Serial.begin(115200);
  Serial.println("STARTED");
}

void loop() 
{
  Serial.println("TEST");   // not nice, only a test
}
```

Thank you and regards

valeros, 2025-01-10:
Hi @Queequack, I see you've changed a lot of settings regarding flash memory, but not the partition table filename. The `esp32-s3-devkitc-1` board uses `default_8MB.csv` by default. Maybe that's your problem? Try something like this:

```ini
board_build.partitions = default.csv
```

Queequack, 2025-01-11:
Hi @valeros, many thanks for your support. Very good advice, thank you very much! The upload works, it's up and running! :)

So that I can see something on the serial monitor, I have also added these build_flags:
```
   -D ARDUINO_USB_CDC_ON_BOOT=1
    ;-D ARDUINO_USB_MODE=1 ; This flag had no effect
```

However, there is still a small problem that I can't figure out. I have added a timer to the code. 
- First so that 2 serial messages are output alternately. 
- Then so that the on board RGB LED is switched on or off (library: FastLED).

FastLED works reliably and from time zero. Only the serial communication does not run smoothly (at the beginning).

After the successful upload:
**Good**: I open the serial monitor and see ON/OFF alternately, analogous to the flashing LED.
**Bad**: If I reset the board, no more serial messages are received.
**Good**: If I restart the serial monitor in PlatformIo, I see ON/OFF alternating again.
**Bad**: STARTED, which should be output in setup(), is not visible.

To narrow down the problem, I added a few `delays()` and `while(!Serial)`.
All without success. I'm not sure if it's a problem with the board (it's the same with all three), or if it's possibly due to an incomplete configuration in platformio.ini.

In the following tests it was irrelevant whether I used `delay(1000)` or `while(!Serial)`.

Test 1
If I reset the board, the monitor no longer displays anything new. It simply stops after the last ON/OFF (PlatformIO task still running). However, the board restarts, the LED first lights up blue as in the code and then continues to flash green.

Test 2
Is the board running or have I just reset it (status after test 1):
If I restart the serial monitor in PlatformIO, I see ON/OFF alternating again.

Test 3
Is the board running or have I just reset it (status after test 1):
If I restart the serial monitor in PlatformIO, press Reset on the board and only release it again when the monitor is open, I also see STARTED.

Regards René

Full main.cpp:

```
#include <Arduino.h>
#include <FastLED.h>

bool timerExpired(unsigned long, unsigned long);

CRGB rgbLed[1];
#define ledPin 21
#define numLeds 1

bool ledState             = true;
unsigned long lastTimeLed =    0;
unsigned long delayLed    =  500;

void setup() 
{
  //delay(1000);
  Serial.begin(115200); 
  //while(!Serial);

  FastLED.addLeds<SK6812, ledPin, GRB>(rgbLed, numLeds);
  FastLED.setBrightness(25);
  rgbLed[0] = CRGB::Blue; FastLED.show();
  ledState = true;

  delay(1000);

  Serial.println("STARTED");
}

void loop() 
{
  if (timerExpired(lastTimeLed, delayLed))
  {
    lastTimeLed = millis();

    if(ledState) 
    { 
      rgbLed[0] = CRGB::Black; FastLED.show(); 
      ledState = false; Serial.println("OFF"); 
    }
    else         
    { 
      rgbLed[0] = CRGB::Green; FastLED.show(); 
      ledState =  true; Serial.println("ON"); 
    }
  }
}

bool timerExpired(unsigned long lastTime, unsigned long delay)
{
  if (millis() - lastTime >= delay) return true;
  return false;
}
```

Full platformio.ini:
```
[env:ESP32-S3-ZERO-TEST]
platform = espressif32
framework = arduino
monitor_speed = 115200

board = esp32-s3-devkitc-1
board_build.partitions=default.csv
board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304

build_flags =
    -DBOARD_HAS_PSRAM
    -D ARDUINO_USB_CDC_ON_BOOT=1
   ;-D ARDUINO_USB_MODE=1 ; This flag had no effect

lib_deps =
    fastled/FastLED @ ^3.6.0
```

valeros, 2025-01-13:
I believe the internal USB peripheral is used to implement a virtual COM port? If so, then such behavior is somewhat expected, because your OS needs some time to recognize your board and represent in as a COM port. You can try another serial monitor to make sure the behavior is quite similar.

Queequack, 2025-01-19:
> Hi [@Queequack](https://github.com/Queequack), I see you've changed a lot of settings regarding flash memory, but not the partition table filename. The `esp32-s3-devkitc-1` board uses `default_8MB.csv` by default. Maybe that's your problem? Try something like this:
> 
> board_build.partitions = default.csv

Setup works, serial monitor delay is not a problem. Can be closed.

closed 2025-01-20 by Queequack