platformio/platform-espressif32 #1531: Compiling debug version using Pre-Debug option causes line ending ParseException

hanssbtn, 2025-01-12:
I tried to compile a debug version of my project using the Pre-Debug option, but I kept getting this error message:
```
Traceback (most recent call last):
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\entity.py", line 163, in _get_infos_from_file
    results = parser.parseString(info.content, parseAll=True)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\penv\.espidf-4.4.7\Lib\site-packages\pyparsing.py", line 1814, in parseString
    raise exc
  File "%USERPROFILE%\.platformio\penv\.espidf-4.4.7\Lib\site-packages\pyparsing.py", line 1808, in parseString
    se._parse( instring, loc )
  File "%USERPROFILE%\.platformio\penv\.espidf-4.4.7\Lib\site-packages\pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\penv\.espidf-4.4.7\Lib\site-packages\pyparsing.py", line 3722, in parseImpl
    loc, exprtokens = e._parse( instring, loc, doActions )
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\penv\.espidf-4.4.7\Lib\site-packages\pyparsing.py", line 1552, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\penv\.espidf-4.4.7\Lib\site-packages\pyparsing.py", line 3502, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected end of text (at char 163098), (line:2551, col:50)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\ldgen.py", line 178, in <module>
    main()
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\ldgen.py", line 154, in main
    mapping_rules = generation_model.generate(sections_infos)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 520, in generate
    root_node.insert(entity, sections, target, flags, entities)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 251, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 243, in child_placement
    child.insert(entity, sections, target, flags, sections_db)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 251, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 243, in child_placement
    child.insert(entity, sections, target, flags, sections_db)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 251, in insert
    self.child_placement(entity, sections, target, flags, sections_db)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\generation.py", line 304, in child_placement
    found_sections = sections_db.get_sections(self.parent.name, self.name)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\entity.py", line 206, in get_sections
    obj = self._match_obj(archive, obj)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\entity.py", line 194, in _match_obj
    objs = self.get_objects(archive)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\entity.py", line 187, in get_objects
    self._process_archive(archive)
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\entity.py", line 175, in _process_archive
    parsed = self._get_infos_from_file(stored)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "%USERPROFILE%\.platformio\packages\framework-espidf@3.40407.240606\tools\ldgen\entity.py", line 165, in _get_infos_from_file
    raise ParseException('Unable to parse section info file ' + info.filename + '. ' + p.msg)
pyparsing.ParseException: Unable to parse section info file %USERPROFILE%/Desktop/ping-pong/.pio/build/server/esp-idf/esp_event/libesp_event.a. Expected end of text (at char 0), (line:1, col:1)
Compiling .pio\build\server\bootloader\esp_hw_support\port\esp32\chip_info.c.o
*** [.pio\build\server\sections.ld] Error 1
========================================================================================== [FAILED] Took 439.96 seconds ==========================================================================================

Environment    Status    Duration
-------------  --------  ------------
server         FAILED    00:07:19.959
====================================================================================== 1 failed, 0 succeeded in 00:07:19.959 ======================================================================================
```

Building the project normally compiles with a few warnings. The error happens at the same file every single time. I tried reinstalling the extension, removing the ```.platformio``` and the ```.pio``` folder, but nothing worked. As usual, any help is greatly appreciated.

Here is my project level ```CMakeLists.txt``` file:
```
cmake_minimum_required(VERSION 3.16.0)
string(TOLOWER ${CURRENT_ENV} CURRENT_ENV)
set(ENV{CURR_ENV} "${CURRENT_ENV}")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(proj_name)
```

Component level ```CMakeLists.txt``` file:
```
FILE(GLOB_RECURSE app_sources ${CMAKE_SOURCE_DIR}/src/$ENV{CURR_ENV}/*.cpp)
idf_component_register(SRCS ${app_sources})
```

Here is my ```sdkconfig.defaults``` file:
```
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
```

Here is my ```platformio.ini``` file:
```
[env]
platform = espressif32
board = esp32doit-devkit-v1
monitor_raw = yes
monitor_speed = 115200
framework = arduino, espidf

build_flags = 
    -DCORE_DEBUG_LEVEL=3
    -DCONFIG_ESP_SYSTEM_PANIC_HALT=y
    -DCONFIG_FREERTOS_VTASKLIST_INCLUDE_COREID=y
    -DCONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=y

debug_build_flags =
    -Os 
    -g 
    -ggdb3
    -DCONFIG_ESP_COREDUMP_DECODE_INFO=y

[env:server]
debug_extra_cmds =
    break main
    break recv_callback
    break send_callback
    break notify_client

board_build.cmake_extra_args = 
    -DCURRENT_ENV=SERVER

[env:client]
debug_extra_cmds =
    break listen_table
    break listen_goal
    break send_packet
    break recv_callback
    break send_callback

board_build.cmake_extra_args = 
    -DCURRENT_ENV=CLIENT
```

valeros, 2025-01-13:
Hi @hanssbtn, I believe the `-ggdb3` causes the problem, probably because it generates extra debug information that is not supported. Try lowering the debug level to `-ggdb2`.

hanssbtn, 2025-01-14:
It worked. Thank you for the answer. I am trying to debug my code since it kept crashing at a certain point of the program, presumably due to a memory corruption in the code, but I can't seem to figure out where it errs. Should I make another issue or can I post the code here? @valeros

valeros, 2025-01-14:
@hanssbtn the issue tracker is this repository is focused on addressing specific problems related to the PlatformIO ecosystem. If you have general coding questions, project-specific issues or problems with third-party software, you might find better help in forums or communities dedicated to those topics.

closed 2025-01-14 by valeros