platformio/platform-espressif32 #1535: ESP32S3-EYE stuck in a crash loop

AlexC-0806, 2025-01-29:
Hi, I'm currently trying to do some projects with the [ESP32S3EYE from Espressif](https://www.espressif.com/en/taxonomy/term/890). I didn't find any platformio.ini file for it, neither I found any precise specs of the board, since there is no specific board option for this board in PlatformIO. I tried to configure it setting the board to `esp32s3camlcd` but it didn't work. I tried to set it to a generic ESP32 S3 DevKit, but still nothing. This is my platformio.ini:

```
[env:esp32s3camlcd]
platform = espressif32
board = esp32s3camlcd
framework = arduino

; Flash settings
board_build.flash_mode = qio
board_build.f_flash = 80000000L

; Enable octal flash if needed
build_flags =
  -DCONFIG_ESPTOOLPY_FLASHMODE_OPI=y
  -DCONFIG_ESPTOOLPY_OCT_FLASH=y

; PSRAM configuration
board_build.arduino.memory_type = qio_opi
```

I tried to upload a simple sketch to test WiFi connectivity, something like this:

```
#include <Arduino.h>
#include <Wifi.h>

#define WIFI "SSID"
#define PASS "PASSWORD"

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Hello World");
  WiFi.begin(WIFI, PASS);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
}

void loop()
{
  // put your main code here, to run repeatedly:
  Serial.println("Connected to WiFi");
  Serial.print("IP Address: " + WiFi.localIP().toString());
  delay(1000);
}
```

but on the serial monitor I get this:

```
ELF file SHA256: 23530ed39e3827cb

E (195) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376fa8
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3808,len:0x41c
load:0x403c9700,len:0x9a8
load:0x403cc700,len:0x28d0
entry 0x403c98b8
E (171) cpu_start: Octal Flash option selected, but EFUSE not configured!

abort() was called at PC 0x40376b89 on core 0

Backtrace: 0x4037739a:0x3fceb260 0x4037c615:0x3fceb280 0x4038228d:0x3fceb2a0 0x40376b89:0x3fceb320 0x403cd72f:0x3fceb350 0x403cd9ee:0x3fceb380 0x403c990d:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
```

I tried troubleshooting with DeepSeek, but didn't find any useful conclusion. If someone could help, it could be very helpful

valeros, 2025-01-31:
Hi @AlexC-0806, Fist of all, it's an issue tracker, not a support forum. Anyway, where did you get that settings? For example:
```
build_flags =
  -DCONFIG_ESPTOOLPY_FLASHMODE_OPI=y
  -DCONFIG_ESPTOOLPY_OCT_FLASH=y
```

These settings are not even applicable to the Arduino framework. 

It seems that the `ESP32S3EYE` board uses `ESP32-S3-WROOM-1U-N8R8` (QSPI Flash, OSPI PSRAM), but you chose the `esp32s3camlcd` board as the basis which uses OSPI FLASH and corresponding bootloader.

What happens if you overwrite it with the following option:

```ini
[env:esp32s3camlcd]
platform = espressif32
board = esp32s3camlcd
framework = arduino

; Flash settings
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.arduino.memory_type = qio_opi

; Set default boot type
board_build.boot = qspi
```

AlexC-0806, 2025-01-31:
I tried some troubleshooting with Deepseek since I wasn't able to find any additional info on the use of this board with PlatformIO. Anyway, I'm gonna try your modification and give you an update on this issue. Thank you

AlexC-0806, 2025-01-31:
With the boot option it doesn't even compile:

```
*  Executing task: platformio run --environment esp32s3camlcd 

Processing esp32s3camlcd (platform: espressif32; board: esp32s3camlcd; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32s3camlcd.html
PLATFORM: Espressif 32 (6.9.0) > ESP32S3 CAM LCD
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFi @ 2.0.0
Building in release mode
*** [.pio/build/esp32s3camlcd/bootloader.bin] Source `/home/alexc/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/bin/bootloader_qspi_80m.elf' not found, needed by target `.pio/build/esp32s3camlcd/bootloader.bin'.
=========================================================================== [FAILED] Took 0.94 seconds ===========================================================================

```

platformio.ini:

```
[env:esp32s3camlcd]
platform = espressif32
board = esp32s3camlcd
framework = arduino

; Flash settings
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.arduino.memory_type = qio_opi

; Set default boot type
board_build.boot = qspi
```

valeros, 2025-01-31:
My bad, it should be `board_build.boot = qio`.

AlexC-0806, 2025-01-31:
After uploading, now I get nothing on the serial port, so I don't know if the code is running correctly...

valeros, 2025-01-31:
> After uploading, now I get nothing on the serial port, so I don't know if the code is running correctly...

What happens when you push the reset button?

AlexC-0806, 2025-01-31:
Nothing happens, I get no response at all

valeros, 2025-01-31:
> Nothing happens, I get no response at all

You probably also need to enable CDC on boot:
```ini
[env:esp32s3camlcd]
platform = espressif32
board = esp32s3camlcd
framework = arduino

; Flash settings
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.arduino.memory_type = qio_opi

; Set default boot type
board_build.boot = qspi

; Enable CDC
build_flags = 
  -DARDUINO_USB_CDC_ON_BOOT=1
```

AlexC-0806, 2025-01-31:
IT WORKS! Thank you so much mate, finally I can start testing camera functionality and other stuff. Thank you very much!!!

closed 2025-01-31 by AlexC-0806

valeros, 2025-01-31:
> IT WORKS! Thank you so much mate, finally I can start testing camera functionality and other stuff. Thank you very much!!!

Take into account that the pinout of the `esp32s3camlcd` board probably doesn't match with your board, so be cautious when using predefined pins like `I2C_SDA`, `VSYNC_GPIO_NUM`, etc.