platformio/platform-espressif32 #1537: Arduino Nano ESP32 and -DBOARD_HAS_PIN_REMAP flag compilation issue

Nobiscum, 2025-02-02:
Trying to run an example from here: https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink

There are two types of errors raising during the build:
1. With all default flags there is an error _'Serial' was not declared in this scope_:

```
src/Blink.cpp: In function 'void setup()':
src/Blink.cpp:68:5: error: 'Serial' was not declared in this scope
     Serial.begin(115200);
     ^~~~~~
src/Blink.cpp:68:5: note: suggested alternative: 'Serial2'
     Serial.begin(115200);
     ^~~~~~
     Serial2
src/Blink.cpp: In function 'void loop()':
src/Blink.cpp:74:5: error: 'Serial' was not declared in this scope
     Serial.println("Hello!");
     ^~~~~~
Compiling .pio/build/arduino_nano_esp32/bootloader_support/src/bootloader_flash.c.o
src/Blink.cpp:74:5: note: suggested alternative: 'Serial2'
     Serial.println("Hello!");
     ^~~~~~
     Serial2 
```
2. If I do `#define Serial Serial0` I'm getting following error:

```
In file included from /Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:235,
                 from /Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.cpp:9:
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:44:69: error: variable or field 'attachInterrupt' declared void
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.cpp:28:6: note: in expansion of macro 'attachInterrupt'
 void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode)
      ^~~~~~~~~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.cpp:28:30: error: expected primary-expression before 'pin'
 void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode)
                              ^~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:44:92: note: in definition of macro 'attachInterrupt'
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                                            ^~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.cpp:28:61: error: expected primary-expression before 'intRoutine'
 void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode)
                                                             ^~~~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:44:98: note: in definition of macro 'attachInterrupt'
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                                                  ^~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/FunctionalInterrupt.cpp:28:73: error: expected primary-expression before 'int'
 void attachInterrupt(uint8_t pin, std::function<void(void)> intRoutine, int mode)
                                                                         ^~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:44:103: note: in definition of macro 'attachInterrupt'
 #define attachInterrupt(pin, fcn, mode)             attachInterrupt(digitalPinToGPIONumber(pin), fcn, mode)
                                                                                                       ^~~~
*** [.pio/build/arduino_nano_esp32/FunctionalInterrupt.cpp.o] Error 1
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp: In function 'void serialEventRun()':
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:51:23: error: 'Serial' was not declared in this scope
     if(serialEvent && Serial.available()) serialEvent();
                       ^~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:51:23: note: suggested alternative: 'Serial2'
     if(serialEvent && Serial.available()) serialEvent();
                       ^~~~~~
                       Serial2
*** [.pio/build/arduino_nano_esp32/HardwareSerial.cpp.o] Error 1
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp: In function 'void app_main()':
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:58:5: error: 'Serial' was not declared in this scope
     Serial.begin();
     ^~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:58:5: note: suggested alternative: 'Serial2'
     Serial.begin();
     ^~~~~~
     Serial2
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:64:5: error: 'USB' was not declared in this scope
     USB.enableDFU();
     ^~~
*** [.pio/build/arduino_nano_esp32/main.cpp.o] Error 1
In file included from /Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:235,
                 from /Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:1:
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:20:44: error: variable or field 'noTone' declared void
 #define noTone(_pin)                noTone(digitalPinToGPIONumber(_pin))
                                            ^~~~~~~~~~~~~~~~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:104:6: note: in expansion of macro 'noTone'
 void noTone(uint8_t _pin){
      ^~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:104:21: error: expected primary-expression before '_pin'
 void noTone(uint8_t _pin){
                     ^~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:20:67: note: in definition of macro 'noTone'
 #define noTone(_pin)                noTone(digitalPinToGPIONumber(_pin))
                                                                   ^~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:21:42: error: variable or field 'tone' declared void
 #define tone(_pin, args...)         tone(digitalPinToGPIONumber(_pin), args)
                                          ^~~~~~~~~~~~~~~~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:123:6: note: in expansion of macro 'tone'
 void tone(uint8_t _pin, unsigned int frequency, unsigned long duration){
      ^~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:123:19: error: expected primary-expression before '_pin'
 void tone(uint8_t _pin, unsigned int frequency, unsigned long duration){
                   ^~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:21:65: note: in definition of macro 'tone'
 #define tone(_pin, args...)         tone(digitalPinToGPIONumber(_pin), args)
                                                                 ^~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:123:25: error: expected primary-expression before 'unsigned'
 void tone(uint8_t _pin, unsigned int frequency, unsigned long duration){
                         ^~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:21:72: note: in definition of macro 'tone'
 #define tone(_pin, args...)         tone(digitalPinToGPIONumber(_pin), args)
                                                                        ^~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/Tone.cpp:123:49: error: expected primary-expression before 'unsigned'
 void tone(uint8_t _pin, unsigned int frequency, unsigned long duration){
                                                 ^~~~~~~~
/Users/ahrys/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:21:72: note: in definition of macro 'tone'
 #define tone(_pin, args...)         tone(digitalPinToGPIONumber(_pin), args)
                                                                        ^~~~
*** [.pio/build/arduino_nano_esp32/Tone.cpp.o] Error 1
```
 
platformio.ini:
```
[env]
platform = espressif32@6.10.0
framework = arduino, espidf
monitor_speed = 115200
build_flags =
    -D CONFIG_BLINK_GPIO=2
build_unflags = -Werror=all
build_type = debug

[env:arduino_nano_esp32]
board = arduino_nano_esp32
board_build.f_cpu = 80000000L
board_build.filesystem = littlefs
board_build.flash_mode = dio
```

Nobiscum, 2025-02-04:
@valeros Yesterday I played a lot with configuration and was able to compile and run the code. However I had to change settings (unset flag BOARD_HAS_PIN_REMAP) to this:
`; Налаштування для Arduino Nano ESP32
[env:arduino_nano_esp32]
board = arduino_nano_esp32
board_build.f_cpu = 80000000L
board_build.filesystem = littlefs
board_build.flash_mode = dio
;board_build.partitions = default_16MB.csv
build_flags = 
    -DARDUINO_USB_MODE=1
    -DARDUINO_USB_CDC_ON_BOOT=1

build_unflags = 
    -DBOARD_HAS_PIN_REMAP
    -DARDUINO_USB_DFU_ON_BOOT
    -Werror=all`

With native IDF I faced another one issue with absent libraries like LittleFS (I imported my arduino project there):
` .pio/build/arduino_nano_esp32/src/main.cpp.o:(.literal._Z5setupv+0x38): undefined reference to `LittleFS'`
now I'm looking for solution how to attach this library to the project.

valeros, 2025-02-04:
The `-DBOARD_HAS_PIN_REMAP` is definitely the culprit here, but the reason why it breaks the build is probably off topic for this repository.

Nobiscum, 2025-02-04:
@valeros >but the reason why it breaks the build is probably off topic for this repository.
Not sure I understand why it is off topic? 
`
[env]
platform = espressif32@6.10.0
framework = arduino, espidf
upload_protocol = esptool
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
lib_deps =
    teemuatlut/TMCStepper@^0.7.3
    waspinator/AccelStepper@^1.64
    workloads/SCServo@^1.0.1
    powerbroker2/SerialTransfer@^3.1.3
    mathieucarbou/ESPAsyncWebServer@^3.6.0
lib_extra_dirs = lib
build_unflags = -Werror=all
`
The platform is espressif32 with two frameworks: arduino, espidf. 
Maybe there is somewhere a repository for arduino submodule?

valeros, 2025-02-04:
@Nobiscum PlatformIO is a development environment that simply puts all parts of your application together, it cannot be held responsible for the internal the issues related to IDF or Arduino frameworks. I'd recommend asking for help in this repo https://github.com/espressif/arduino-esp32

Nobiscum, 2025-02-04:
> [@Nobiscum](https://github.com/Nobiscum) PlatformIO is a development environment that simply puts all parts of your application together, it cannot be held responsible for the internal the issues [...]

Ah, I thought you somehow customize idf and arduino-esp32 frameworks because there is still no support of idf 5.x and arduino-esp32 framework 3.x. 

Ok, tnx. Will dig into that problem from another side.

valeros, 2025-02-04:
> Ah, I thought you somehow customize idf and arduino-esp32 frameworks because there is still no support of idf 5.x and arduino-esp32 framework 3.x.

We use the v2.x branch https://github.com/espressif/arduino-esp32/tree/release/v2.x

closed 2025-02-04 by valeros