platformio/platform-espressif32 #1538: Packaged CMake version is outdated

RandomSpaceship, 2025-02-05:
This toolchain defaults to using CMake 3.16.0, with only up to 3.21.3 even packaged. These are both _many_ years out of date, and missing support for modern CMake features such as file sets, significantly limiting interoperability with other CMake projects. Would it be possible to get an updated version of `tool-cmake` published on the PlatformIO registry, or even set as the toolchain default?

valeros, 2025-02-06:
Hi @RandomSpaceship, I've just uploaded v3.30.2 to the registry. You can specify it directly in your `platformio.ini`:

```ini
[env:esp32dev]
platform = espressif32
framework = espidf
monitor_speed = 115200
board = esp32dev
platform_packages = 
    tool-cmake @ ~3.30.2

```

closed 2025-02-06 by valeros

RandomSpaceship, 2025-02-06:
Thank you so much! That'll make my life a lot easier.