platformio/platform-espressif32 #1546: About ESP32-S3  & ETH_PHY_W5500

hzhh110, 2025-02-21:
SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);

I don't know why, when I use the latest version, the option ETH_PHY_W5500 cannot be selected for network cable connection. I want to use ESP32-S3-WROOM-1 to connect network cable, but it cannot be realized

hzhh110, 2025-02-21:
This won't compile with platformio
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_W5500_Arduino_SPI/ETH_W5500_Arduino_SPI.ino

Jason2866, 2025-02-21:
@hzhh110 As expected since Platformio does not support actual espressif Arduino core 3.x.x

valeros, 2025-02-21:
Depends on https://github.com/platformio/platform-espressif32/issues/1225

closed 2025-02-21 by valeros