platformio/platform-espressif32 #1550: Webserver calls wrong Callback

ams-hh, 2025-03-10:
Good morning!

After several years of "works as expected" I now have crashes inside the Webserver. 
(WebServer.h. "Now" means: Change for framework-arduinoespressif32 @ 3.20014 to framework-arduinoespressif32 @ 3.20017)

WbServer.h offers the method

```
void on(const Uri &uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn); //ufn handles file uploads

```  
to define two callbacks "Function" and "Uploads Function".

As said, until  V 3.20014 everything worked as expected. From  3.20017 on, when exceuting a POST to the fn, THE UFN is called. NOT THE FN. (And my system then crashes because of wrong handling ...)

To demonstrate I provide:
1) Demo-Program
2) platformio.ini
3) HTML Test-File

DEMO PROGRAM
```
#include    <Arduino.h>
#include    <WiFi.h>
#include    <WebServer.h>

WebServer                   clWebServer(80);

// Change this!!
const char* ssid     = "???";   //  <----------- YOUR SSID
const char* password = "???";   //  <----------- YOUR PW

void    CB_FN()
{
    Serial.println("Hello! I am the FN-Callback");
}

void    CB_UFN()
{
    Serial.println("Hello! I am the UFN-Callback");
}

void    NotFound()
{
    Serial.println("Sorry ... did not found");
}

boolean connectWifi(){
  boolean state = true;
  int i = 0;
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 20){
      state = false; break;
    }
    i++;
  }
  Serial.println("");
  if (state){
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("Connection failed.");
  }
  return state;
}

void setup()
{
    Serial.begin(115200);
    Serial.println("\n\nSTART");

    if (connectWifi())
    {
        clWebServer.on("/test"  , 
                       HTTP_POST,
                       CB_FN    ,
                       CB_UFN   );

        clWebServer.onNotFound(NotFound);  

        clWebServer.begin();
    }
    else
    {
        while(1)
        {
            Serial.print("?..");
            delay(1000);
        }
    }
}
 
void loop()
{
    clWebServer.handleClient();
}

```

PLATFORMIO.INI
(I use a custom board file. Substitude it by an own board for ESP32-S3)

```
[platformio]
default_envs = esp32S3-with-6-10-0

[env]
framework = arduino
monitor_speed = 115200
upload_speed = 921600
monitor_filters = time, default

[env:esp32S3-with-6-10-0]
build_flags = 
  -D CORE_DEBUG_LEVEL=0

platform = espressif32@6.10.0
board = esp32s3_4MB_NOPSRAM_CDC

[env:esp32S3-with-6-6-0]
build_flags = 
  -D CORE_DEBUG_LEVEL=0
platform = espressif32@6.6.0
board = esp32s3_4MB_NOPSRAM_CDC

```  

HTML FILE
```
<!DOCTYPE html>
<html>
	<head>
		<meta content="text/html;charset=ISO-8859-1" http-equiv="content-type">
		<title>CALL FN</title>
	</head>

	<body>
	</body>

	<script>
		alert("Now excecute FN Call");

	  	var xhttp = new XMLHttpRequest();

		//CAUTION: Change to your Webserver IP Address!!
		xhttp.open("POST", "http:/192.168.0.105/test", true); 
	  	xhttp.send();
	</script>
</html>

```  

1.) Build the project with project environment esp32S3-with-6-6-0
(Substitude SSID/PW for your needs!)
2.) Run and see the connected IP
3.) Substitude IP in HTML-File
4.) Open Brower and execute HTML-File
5.) See in PIO Terminal, that Function FN ist called
---> Works as expected

6.) Change to project environment esp32S3-with-6-10-0 and build
... rest the same as before

---> See in PIO Terminal, that Function UFN ist called first, than FN

Why?

Can someone verify? Or provide a solution?

Thank you !

P.S.
There is a change in RequestHandlersImpl.h. in the class FunctionRequestHandler.
There appears two new methods canRaw and raw that do not exist in Version
 3.20014. And they do something with the _ufn parameter. But I did not digg into deeper ...

valeros, 2025-03-10:
Hi @ams-hh, the issue tracker in this repository is not the right place for project-specific questions.

closed 2025-03-10 by valeros

ams-hh, 2025-03-10:
Hello Valeros,
this is not a project-specific questions.
This is an issue, because the Webserver changed its behavier by change of the framework-arduinoespressif32.
It calls THE WRONG Callback.
Please read again, what I#ve written.

valeros, 2025-03-11:
@ams-hh What I was trying to say is that PlatformIO is a development environment, it has nothing to do with third-party components like Arduino framework, Webserver library or your code.

ams-hh, 2025-03-11:
Thanks for clarification

ams-hh, 2025-03-11:
... sorry to bother you again...
I am a little bit confused.

Is https://community.platformio.org the right place to ask that question?
Or if not, can direct me to that place?

Regards
A.

valeros, 2025-03-11:
I'd recommend this repo https://github.com/espressif/arduino-esp32, but keep in mind that PlatformIO currently uses Arduino core v2.x. The latest core v3.x may contain a fix for your problem, but it's not yet supported by PlatformIO.

ams-hh, 2025-03-11:
Yes, I know that PIO probably is a dead end concerning V2 / V3.
Maybe the pioArduino fork comes into mind in the future but until now I am missing nothing that comes exclusivly with V3.

Thank you again