platformio/platform-espressif32 #1559: Use uv for package installation if available

grischard, 2025-03-31:
The framework currently installs a few packages by calling `pip` directly:

https://github.com/platformio/platform-espressif32/blob/master/builder/frameworks/espidf.py

The new-ish [`uv`](https://docs.astral.sh/uv/) is much faster than pip. This matters especially when compiling on low power devices like a Raspberry Pi.

If `uv` is installed, `pip` should be called as `uv pip` instead.

Jason2866, 2025-04-01:
Pip installs needed packages one time only. Same as for tools and toolchains. There is no speed increase when compile is done the 2nd time.

grischard, 2025-04-01:
Yes, but in a build system like esphome, it happens every time a compile environment gets created, for every single device. This environment gets re-created every new version. The savings over time would therefore be significant. 

(Yes, some caching there would improve thing seriously; both of these things are true)

Spectre5, 2025-07-24:
I have `esphome` installed using `uv`, but now when compiling I get an error since the `uv` python environments do no contain `pip` at all:

`/home/USER/.local/share/uv/tools/esphome/bin/python3: No module named pip`.

@grischard, what did you do to get around this issue?  I guess I'll have to install pip into that `uv` environment...or did you do something else?