platformio/platform-espressif32 #1564: SPI crashes on lilygo T-embed-s3

ricdijk, 2025-04-04:
_I don't know if this is the right place to enter the issue and maybe I am doing something wrong myself. _

Since espressif32@6.7.0 a call to tft.init() (from the bodmer/TFT_eSPI library latest version as of today) crashes for the lilygo T-embed-s3:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4200a2f5  PS      : 0x00060a30  A0      : 0x8200a50c  A1      : 0x3fcebe30  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x0000000b  
A6      : 0x000000ff  A7      : 0x00000001  A8      : 0x08000000  A9      : 0x3fcebe00  
A10     : 0x3fca3bb0  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000200  
A14     : 0x00000000  A15     : 0x3fc9ab1c  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x4209d498  LEND    : 0x4209d4fc  LCOUNT  : 0x00000000  

Backtrace: 0x4200a2f2:0x3fcebe30 0x4200a509:0x3fcebe60 0x42007e52:0x3fcebe80 0x42006510:0x3fcebeb0 0x4202628a:0x3fcebf20
```

I traced back the issue and found the board crashes on: `SET_BUS_WRITE_MODE;` (TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c)

Traceback of `SET_BUS_WRITE_MODE;`:
1: `#define SET_BUS_WRITE_MODE *_spi_user = SPI_USR_MOSI` (TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h)
2: `#define _spi_user               (volatile uint32_t*)(SPI_USER_REG(SPI_PORT))`
3: `#define SPI_USER_REG(i)          (REG_SPI_BASE(i) + 0x10)`
4: `#define REG_SPI_BASE(i)         (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))    // GPSPI2 and GPSPI3` (.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:122 (included from sensor.h))

Since the board defines SPI_PORT=0, _spi_user is assigned a NULL pointer and the board crashes.
_in 6.6.0 the definition is:  `#define REG_SPI_BASE(i) (((i)>1) ? (DR_REG_SPI3_BASE) : (DR_REG_SPI2_BASE))`, no NULL pointer and works fine_

I solved the issue by using the flag `-D USE_HSPI_PORT` => this sets SPI_PORT=3 and the sketch runs fine. However the definition or REG_SPI_BASE(i) seems wrong of the board should not use SPI_PORT=0. Maybe someone with more knowledge can can look int this (or point me to where I m wrong).

Jason2866, 2025-04-04:
For sure the issue is wrong placed in this github. The issue is known. https://github.com/espressif/arduino-esp32/issues/10115
Best would be if it is fixed in the place where it is wrong -> IDF
A fix could be done in the bodmer/TFT_eSPI too

valeros, 2025-04-08:
Thanks @Jason2866 for the clarification.
Closing as the issue has nothing to do with PlatformIO.

closed 2025-04-08 by valeros