platformio/platform-espressif32 #1566: ULP ADC fails on Espressif32 platform version 6.5.0 and up

ayavilevich, 2025-04-15:
Using an ADC read instruction in the ULP (ultra low power) processor will cause the ULP to "get stuck" in newer SDK versions.

I saw this in an Platform IO project but not sure what the root case is.

I created a minimal reproduce project at https://github.com/ayavilevich/ulp_adc_issue . 

Last working set is: Espressif32 6.4.0, ESP Arduino: 2.0.11, ESP-IDF: 4.4.5  
The first non working set is: Espressif32 6.5.0, ESP Arduino: 2.0.14, ESP-IDF: 4.4.6

What is the process to troubleshot this further?

Can others reproduce and confirm this please?

Jason2866, 2025-04-15:
Wrong repo for the issue since not related to Platformio. Open the issue here https://github.com/espressif/esp-idf

ayavilevich, 2025-04-15:
> Wrong repo for the issue since not related to Platformio. Open the issue here https://github.com/espressif/esp-idf

I suppose it is more likely to be related to esp-idf or to ESP Arduino. What is the right process to confirm that?

Jason2866, 2025-04-15:
ULP support code is in IDF. Running ULP code in Arduino is more or less a hack.
So the espressif IDF repo is the place for.

ayavilevich, 2025-04-16:
Ok, I will try to test this with esp-idf and will create an issue in the esp-idf repo if I am able to reproduce.

ayavilevich, 2025-07-06:
For the record, this doesn't seem to happen in a pure esp-idf case. Something specific to arduino-esp32.
Created: https://github.com/espressif/arduino-esp32/issues/11557
POC: https://github.com/ayavilevich/ulp_adc_issue_ard_min