platformio/platform-espressif32 #1570: Feature Request: Add support for ESP32-P4 series to espressif32 platform

LShang001, 2025-04-21:
## Feature Description

The ESP32-P4 is Espressif's latest high-performance RISC-V SoC, featuring advanced processing power, AI acceleration, and a rich set of peripherals (e.g., MIPI CSI/DSI, USB 3.0 OTG, Ethernet). As this chip becomes available and gains traction among developers for next-generation applications, integrating native support into the PlatformIO ecosystem would provide significant value.

This feature request seeks to add full support for the **ESP32-P4 series** SoCs and related development boards (e.g., the official **ESP32-P4-DevKitC-1.0**) to the `espressif32` platform in PlatformIO.

## Motivation

Currently, developers targeting the ESP32-P4 must use the native ESP-IDF build system outside of PlatformIO, which lacks the streamlined experience PlatformIO offers. Adding ESP32-P4 support would enable developers to:

1. **Leverage the PlatformIO workflow:** Benefit from PlatformIO's intuitive project management, library management, build system, and IDE integration (e.g., VSCode).
2. **Simplify development:** Eliminate the need to manually configure toolchains and build environments, saving time and effort.
3. **Utilize PlatformIO features:** Take advantage of tools like Unit Testing, Static Code Analysis, Firmware Inspection, and simplified debugging setups.
4. **Maintain consistency:** Ensure a uniform development experience across Espressif's chip lineup (ESP32, ESP32-S2/S3/C3, and now P4).

## Proposed Solution Components

To implement ESP32-P4 support, the following components are suggested:

1. **Toolchain Integration:** Incorporate the RISC-V GCC toolchain required for ESP32-P4 development.
2. **Framework Support:**
   - **ESP-IDF:** Support ESP-IDF versions that include ESP32-P4 compatibility (e.g., ESP-IDF v5.3 or later).
   - **Arduino Core:** Integrate the Arduino core for ESP32-P4 once it is released and stable.
3. **Board Definitions:** Add definitions for official boards like `ESP32-P4-DevKitC-1.0` and potentially popular third-party boards, including memory layouts, pin mappings, and default upload/monitor settings.
4. **Debugging Support:** Enable compatibility with debugging probes (e.g., JTAG via ESP-Prog or J-Link) using OpenOCD or other debug servers.
5. **Documentation:** Update PlatformIO documentation with ESP32-P4-specific guidance, including examples and configuration details.

## Relevant Information

- **ESP32-P4 Overview:** [https://www.espressif.com/en/products/socs/esp32-p4](https://www.espressif.com/en/products/socs/esp32-p4)
- **ESP-IDF Documentation (Ensure version supports P4):** [https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/)
- **Example Dev Kit:** ESP32-P4-DevKitC-1.0 (or adjust based on official naming when confirmed)

## Current Status/Workaround

At present, the ESP32-P4 is not supported within the `espressif32` platform in PlatformIO. Developers must resort to using the ESP-IDF build system directly as a workaround.

## Conclusion

Incorporating ESP32-P4 support into PlatformIO would be a impactful enhancement, aligning the platform with Espressif's latest hardware innovations. This addition would empower developers to create cutting-edge embedded applications within their preferred development environment.

Thank you for considering this feature request!

Jason2866, 2025-07-02:
There is full support for the P4 in the pioarduino Platformio fork. For IDF 5.3 and 5.4 and Arduino.

miguelfernando55, 2025-09-24:
Please add support for FireBeetle 2 ESP32-P4

MichelReij, 2026-01-07:
What is the current status on this? Is there a beta-version for testing the P4-with-Arduino on PlatformIO?

nicklasb, 2026-01-07:
I am a little bit worried about there not being progress on what feels like an important chip, is there anything the community can do to help? (With regards to financial support, it would be great with a lower-tier support package, $3K/year is steep for lowest tier of most things)

With regards to using pioarduino, I am having issues with the pioarduino VS Code extension together with PlatformIO, which while impressive, it isn't as polished and I am getting conflicts with other projects (my workaround it vanilla ESP-IDF, but I'd rather use PlatformIO to be able to change MCU:s easier).

Pieter-van-Wyk, 2026-09-09:
ESP32-P4 for Platformio - Is it possible? Yes it is.
What are the politics behind the screens - who is possibly discouraging this and for what? Is this device a threat for others?
I think if we can firstly find out who are capable of establishing the ESP32-P4 for Platformio, to correspond, for us to find out from them what is required for this addition?

nicklasb, 2026-09-09:
> ESP32-P4 for Platformio - Is it possible? Yes it is. What are the politics behind the screens - who is possibly discouraging this and for what? Is this device a threat for others? I think if we can [...]

Well, with this lack of information and what seems like a sudden stop of continued support for a very popular MCU brand, your speculation becomes understandable. 
As the P4s are such an important step forward for Espressif, and I use them all over for their performance, my workaround has now been ESP-IDF for so long that I am starting to give up on PlatformIO altogether. I want to be able to move parts to STM variants, but now that becomes more difficult.