platformio/platform-espressif32 #1574: Request to support ESP IDF v5.4.1

ascillato, 2025-04-29:
The new ESP-IDF v5.4.1 has been released last month: https://github.com/espressif/esp-idf/releases/tag/v5.4.1

Please, can this be added to platformio since it has several bugfixing including some for BLE?

Thanks a lot for your time.

valeros, 2025-04-30:
ESP-IDF v5.4.1 is already available in the development branch.

closed 2025-04-30 by valeros

ascillato, 2025-04-30:
Awesome! Thanks a lot @valeros 

I just tested the development branch and works fine. No issues. Besides, I tested some of the fixes of the ESP-IDF v5.4.1 and it solves some issues I was facing.

Just to know, when do you expect to release the new version of platform-espressif32?

valeros, 2025-04-30:
> Just to know, when do you expect to release the new version of platform-espressif32?

We're aiming for release within the next month, assuming no critical issues arise.