platformio/platform-espressif32 #1576: ESP32-C3 5.4.1  uart: GPIO 20 is not usable, maybe used by others

kscheff, 2025-05-01:
When switching to the new 5.4.1 version I get this warning when dynamically assigning different GPIO Pins as RX.
In my code I am dynamically changing from GPIO2 and GPIO20:

```
        int rxd_pin = port->use_rxd_alt ? port->rxd_pin_alt : port->rxd_pin;
        uart_set_pin(port->port, UART_PIN_NO_CHANGE, rxd_pin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
```

The console output is disabled and when compiling and running with ESPIDF Version 5.4.0. I don't get any warning.
The UART is working fine. I still have to verify if this is just a warning issue or if it's not working with 5.4.1.

valeros, 2025-05-02:
HI @kscheff, what's the warning exactly?

kscheff, 2025-05-02:
The warning comes form the uart driver:
` uart: GPIO 20 is not usable, maybe used by others`

valeros, 2025-05-02:
> The warning comes form the uart driver

I see, so this warning comes directly from ESP-IDF, hence it has nothing to do with PlatformIO
You'd better ask for a clarification here https://github.com/espressif/esp-idf

closed 2025-05-02 by valeros