platformio/platform-espressif32 #1590: Arduino: Unable to use ADC continous mode

PaulChristopher, 2025-05-23:
Hi all and thanks for the great work!

I'm totally new to PlatformIO, but I like it much more the than the Arduino IDE. Great job!

I am having trouble to get this example work:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogReadContinuous/AnalogReadContinuous.ino

Building fails with the following error:
`
```
Processing esp32dev (platform: espressif32 @ 6.10.0; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20017.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@3.20017.241212+sha.dcc1105b has been installed!
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Building .pio\build\esp32dev\bootloader.bin
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\HardwareSerial.cpp.o
src/main.cpp:18:1: error: 'adc_continuous_data_t' does not name a type
 adc_continuous_data_t *result = NULL;
 ^~~~~~~~~~~~~~~~~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:30:3: error: 'analogContinuousSetWidth' was not declared in this scope
   analogContinuousSetWidth(12);
   ^~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:30:3: note: suggested alternative: 'analogSetWidth'
   analogContinuousSetWidth(12);
   ^~~~~~~~~~~~~~~~~~~~~~~~
   analogSetWidth
src/main.cpp:37:3: error: 'analogContinuous' was not declared in this scope
   analogContinuous(adc_pins, adc_pins_count, CONVERSIONS_PER_PIN, 20000, &adcComplete);
   ^~~~~~~~~~~~~~~~
src/main.cpp:37:3: note: suggested alternative: 'analogInPinToBit'
   analogContinuous(adc_pins, adc_pins_count, CONVERSIONS_PER_PIN, 20000, &adcComplete);
   ^~~~~~~~~~~~~~~~
   analogInPinToBit
src/main.cpp:40:3: error: 'analogContinuousStart' was not declared in this scope
   analogContinuousStart();
   ^~~~~~~~~~~~~~~~~~~~~
src/main.cpp:40:3: note: suggested alternative: 'analogInPinToBit'
   analogContinuousStart();
   ^~~~~~~~~~~~~~~~~~~~~
   analogInPinToBit
src/main.cpp: In function 'void loop()':
src/main.cpp:49:31: error: 'result' was not declared in this scope
     if (analogContinuousRead(&result, 0)) {
                               ^~~~~~
src/main.cpp:49:9: error: 'analogContinuousRead' was not declared in this scope
     if (analogContinuousRead(&result, 0)) {
         ^~~~~~~~~~~~~~~~~~~~
src/main.cpp:49:9: note: suggested alternative: 'analogRead'
     if (analogContinuousRead(&result, 0)) {
         ^~~~~~~~~~~~~~~~~~~~
         analogRead
src/main.cpp:52:7: error: 'analogContinuousStop' was not declared in this scope
       analogContinuousStop();
       ^~~~~~~~~~~~~~~~~~~~
src/main.cpp:52:7: note: suggested alternative: 'analogInPinToBit'
       analogContinuousStop();
       ^~~~~~~~~~~~~~~~~~~~
       analogInPinToBit
src/main.cpp:64:7: error: 'analogContinuousStart' was not declared in this scope
       analogContinuousStart();
       ^~~~~~~~~~~~~~~~~~~~~
src/main.cpp:64:7: note: suggested alternative: 'analogInPinToBit'
       analogContinuousStart();
       ^~~~~~~~~~~~~~~~~~~~~
       analogInPinToBit
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
=========================================================================== [FAILED] Took 212.78 seconds ===========================================================================
```

`

Indeed, when looking into .platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-adc.h, the continous mode for adc is not implemented.

How can I resolve the issue and update the ESP core?

I have a fresh install of Visual Studio Code 1.100.2 and PlatformIO Core 6.1.18.

"PIO Home -> Platforms -> Updates" shows "No results"...

No way to update?

Before, I had tried to add

platform = platformio/espressif32
board = esp32dev
framework = arduino
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

to platform.ini as suggested [here](https://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version).

This triggered the following errors:
```

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Configuring toolchain packages from a remote source...
Tool Manager: Installing git+https://github.com/espressif/arduino-esp32.git
git version 2.44.0.windows.1
Cloning into 'C:\Users\XXX\.platformio\.cache\tmp\pkg-installing-pczsx5y1'...
remote: Enumerating objects: 2496, done.
remote: Counting objects: 100% (2496/2496), done.
remote: Compressing objects: 100% (1729/1729), done.
remote: Total 2496 (delta 616), reused 1421 (delta 418), pack-reused 0 (from 0)
Receiving objects: 100% (2496/2496), 23.73 MiB | 6.57 MiB/s, done.
Resolving deltas: 100% (616/616), done.
Updating files: 100% (1914/1914), done.
Tool Manager: framework-arduinoespressif32@3.2.0+sha.d85b75c has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.2.0+sha.d85b75c
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

*** missing SConscript file 'C:\\Users\\XXX\\.platformio\\packages\\framework-arduinoespressif32\\tools\\platformio-build.py'
File "C:\Users\XXX\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 36, in <module>
=========================================================================== [FAILED] Took 65.75 seconds ===========================================================================
```

Sorry, I am stuck...

Thanks a keep up the good work!
Paul

PaulChristopher, 2025-05-23:
Test case

[main.cpp.txt](https://github.com/user-attachments/files/20418089/main.cpp.txt)

Jason2866, 2025-05-23:
This is not possible with actual Arduino core 3.x 
```
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

valeros, 2025-05-27:
Hi @PaulChristopher, the API you're trying to use was added in Arduino core v3.x. PlatformIO currently supports v2.x only, see #1225.

closed 2025-05-27 by valeros