platformio/platform-espressif32 #1600: PlatformIO cannot upload to a ESP32 device: Wrong boot mode detected (0x13)

MarusGradinaru, 2025-06-09:
Hi,
I’am new to Platform IO. I switched from Arduino IDE because PIO build faster.

**The Setup:**
I have a laptop with Windows 10, and a ESP32-DevKitC-VE with WROVER-E N8R8. I installed VSCode and then added Platform IO extension. I also updated the esptool.py to the latest version (v4.8.1). I installed the official CP210x_VCP_Windows drivers and the board is successfully detected on COM3. The board is original from Espressif and it has auto-upload capabilities (using DTR/RTS). If I try to program the board from Arduino IDE, everything works fine ! But I can't do it from Platform IO.

**The Problem:**
I cannot convince the esptool.py to automatically connect to my board when I press the Upload button from PlatformIO. Or if I run in PlatformIO CLI commands like "python esptool.py flash_id". I receive the same error. There were a few times when the connection succeeded, but most of the time it doesn’t work. I suspect that is someting from the "custom_reset_sequence", because if I manually hold the BOOT button while connecting, it succeeds every time.

**The Error**
```
esptool.py v4.8.1
Serial port COM3
Connecting......................................
A fatal error occurred: Failed to connect to Espressif device: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
```

So, It seems that the esptool cannot put the board in download mode by itself. But why it works fine if esptool is executed from Arduino IDE ? It is the same version ! I don’t want to press the BOOT button everytime I make an upload. I switched to PIO because Arduino IDE was slow. Pressing the BOOT button everytime is even slower… 🙄 

I extracted the esptool command line use by both IDE's:

**Arduino IDE**
```
"C:\Users\Marus\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3/esptool.exe"
--chip esp32 --port "COM3" --baud 921600 --before default_reset --after hard_reset write_flash -z 
--flash_mode keep --flash_freq keep --flash_size keep 
0x1000 "C:\Users\Marus\AppData\Local\arduino\sketches\1D792A9CE841D1DB0E6D4E382C64C4AF/BluetoothAudio.ino.bootloader.bin" 
0x8000 "C:\Users\Marus\AppData\Local\arduino\sketches\1D792A9CE841D1DB0E6D4E382C64C4AF/BluetoothAudio.ino.partitions.bin" 
0xe000 "C:\Users\Marus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0/tools/partitions/boot_app0.bin" 
0x10000 "C:\Users\Marus\AppData\Local\arduino\sketches\1D792A9CE841D1DB0E6D4E382C64C4AF/BluetoothAudio.ino.bin"
```

**Platform IO**
```
"C:\Users\Marus\.platformio\penv\Scripts\python.exe" "C:\Users\Marus\.platformio\packages\tool-esptoolpy\esptool.py"
--chip esp32 --port "COM3" --baud 921600 --before default_reset --after hard_reset write_flash -z 
--flash_mode dio --flash_freq 80m --flash_size 8MB 
0x1000 "D:\ESP Audio\Soft\Bluetooth Audio PIO\.pio\build\esp32dev\bootloader.bin" 
0x8000 "D:\ESP Audio\Soft\Bluetooth Audio PIO\.pio\build\esp32dev\partitions.bin" 
0xe000 "C:\Users\Marus\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin" 
0x10000 .pio\build\esp32dev\firmware.bin
```

There don't seem to be any significant differences... But it may be more than that. The "esptool" can be run with a [custom configuration](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html), and maybe Arduino IDE changed someting there. 
I made a "esptool.cfg" file with extended delays, which I can confirm that is recognised and used, but it has no effect. It still cannot connect to bootloader.
```
[esptool]
custom_reset_sequence = D0|R1|W1|D1|R0|W1|D0
```

It's such a shame that something that was intended to be a fun hobby, it started with a big nightmare... 😢

MarusGradinaru, 2025-06-10:
> pioarduino use an actual verified working esptool.py v4.8.1+ Not the original v4.8.1!

I don't understand what you mean here... 
When I first installed the Platform IO in VSCode, the esptool.py version was someting lower (i don't remember which). And the upload didn't work. So I thought to update it to the latest version v4.8.1. But this not helped either...

Jason2866, 2025-06-10:
clone the example project i have linked and try if this works

joba-1, 2025-06-10:
> More than that, I don't know what to do... 😢

I do - follow Jasons advice 🤓: clone his repo, add your upload port if you have to but change nothing else, build and upload 

If that doesn‘t work, post the upload output

You have done so many tests and changes that by now it is impossible to see from outside what is part of a solution and what is part of the problem

closed 2025-06-12 by valeros