platformio/platform-espressif32 #1603: Windows 11 esp-idf  cmake fails with "Command too long"

rw950431, 2025-06-22:
Hi
I’m a new platformio user with a familiar problem: The length of a compiler command exceeds the maximum allowable command length for my windows 11 system. 
I started with the [espidf-ulp-pulse](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-ulp-pulse) example and tried to add the bits for wifi and mqtt as follows

CMakeLists.txt

```
cmake_minimum_required(VERSION 3.16.0)

list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(espidf-ulp-pulse-new)
```
src/CMakeLists.txt

```
idf_component_register(SRCS "ulp_example_main.c"
                    INCLUDE_DIRS ""
                    REQUIRES driver soc nvs_flash ulp 
                    PRIV_REQUIRES mqtt esp_wifi
                    )
#
# ULP support additions to component CMakeLists.txt.
#
# 1. The ULP app name must be unique (if multiple components use ULP).
set(ulp_app_name ulp_main)
#
# 2. Specify all assembly source files.
#    Files should be placed into a separate directory (in this case, ulp/),
#    which should not be added to COMPONENT_SRCS.
set(ulp_s_sources "../ulp/pulse_cnt.S" "../ulp/wake_up.S")
#
# 3. List all the component source files which include automatically
#    generated ULP export file, ${ulp_app_name}.h:
set(ulp_exp_dep_srcs "ulp_example_main.c")
#
# 4. Call function to build ULP binary and embed in project using the argument
#    values above.
ulp_embed_binary(${ulp_app_name} "${ulp_s_sources}" "${ulp_exp_dep_srcs}")

target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
```

By my count the command is just over 9K in length. Interestingly removing mqtt from the REQUIRES line shortens it just enough to come in under the 8K windows command line limit.

I tried adding the script suggested as a workaround at 
https://github.com/platformio/platformio-core/issues/3792#issuecomment-759617616

but that python script would not run on my system.

I’m using latest versions of all systems as installed by the VSCode extension system:
Windows 11 Home Edition, fully patched
VSCode version 1.101.1
Platform IO version 3.3.4
ESP-IDF v5.1.2

Complete output at gist [pio run -v](https://gist.github.com/rw950431/fa8f58019dfe7ded1cdfb461c9fd0541)

valeros, 2025-06-30:
Hi @rw950431, could you please put together a minimal example to reproduce the issue?

rw950431, 2025-07-01:
Thanks for your interest @valeros .  Happy to put together a minimal example but how do I post it here?  Do I put it up as a repo or attach as a zip file?  Sorry to ask newbie questions but I've not been involved in the issues process before.

valeros, 2025-07-01:
Just put your project example into a zip archive and attach to your next comment.

rw950431, 2025-07-05:
Please find attached example of the build failure.   The output of **pio run -v**  is saved to the **build.log** file for reference

[issue_1603_espidf-ulp-pulse.zip](https://github.com/user-attachments/files/21071842/issue_1603_espidf-ulp-pulse.zip)

Apologies for the delay in posting this.

valeros, 2025-07-07:
That's strange, I can build your project without any issue even though build commands on my machine are even longer than yours. Have you enabled [support for long paths](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry#enable-long-paths-in-windows-10-version-1607-and-later) on your machine? I also noticed you're using almost a two-year-old version v6.5.0, is there a reason for that? Have you tried the latest version of the `espressif32` dev platform v6.11?