platformio/platform-espressif32 #1610: Platform IO why everytime build the application checking the partition table at offset 0x8000 why?

ept-taseen, 2025-07-17:
1. Just building the application with platform.io and found that this application is not working with any other partition table having different offset other then 0x8000

The application should be independent of this hard coded 0x8000 address, because if we want to change the address of our partition table then the application unable to locate the partition table and failed to load the nvs 

The app eveytime check the nvs at partition offset 0x8000 why?

Jason2866, 2025-07-17:
Why, probably because it is hardcoded in ArduinoIDE too. [Changed this](https://github.com/espressif/arduino-esp32/pull/11534) recently to make it configurable. This possibility is only available with latest released Arduino core 3.2.1. If you want to use you have to use the fork pioarduino.

valeros, 2025-07-18:
@ept-taseen what framework are you using?

ept-taseen, 2025-07-19:
@valeros what if i want to change the offset?

i want to build the application from the platform io arduino framework

1. Is there any way or we need to build the arduino core again?

valeros, 2025-07-21:
@ept-taseen Are we still talking about a native Arduino project or it's a mixed Arduino/IDF project?

closed 2026-10-31 by stale[bot]