platformio/platform-espressif32 #1629: ESP32-C3 compiler error

SimoDolcio, 2025-08-02:
I'm trying to compile a project for ESP32-C3 using VS Code and PlatformIO, but I'm getting the following error:

`rocessing sonoff_r4 (platform: espressif32; board: esp32-c3-devkitm-1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (53.3.11) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.1
 - framework-arduinoespressif32-libs @ 5.3.0+sha.cfea4f7c98
 - tool-esptoolpy @ 4.8.5
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-riscv32-esp @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 54 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.4.2
|-- AsyncTCP @ 3.4.6
|-- Adafruit AHTX0 @ 2.0.5
|-- Adafruit BMP280 Library @ 2.6.8
|-- TcpIpServerClient @ 0.0.0+20250802030153
|-- LittleFS @ 3.1.1
|-- WiFi @ 3.1.1
Building in release mode
Compiling .pio\build\sonoff_r4\src\main.cpp.o
Archiving .pio\build\sonoff_r4\lib33c\libNetwork.a
Compiling .pio\build\sonoff_r4\lib4b1\AsyncTCP\AsyncTCP.cpp.o
Compiling .pio\build\sonoff_r4\libc60\Wire\Wire.cpp.o
Compiling .pio\build\sonoff_r4\libcd9\SPI\SPI.cpp.o
Archiving .pio\build\sonoff_r4\libc60\libWire.a
Archiving .pio\build\sonoff_r4\lib4b1\libAsyncTCP.a
Compiling .pio\build\sonoff_r4\libe2b\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
Archiving .pio\build\sonoff_r4\libcd9\libSPI.a
Compiling .pio\build\sonoff_r4\libe2b\Adafruit BusIO\Adafruit_GenericDevice.cpp.o
Compiling .pio\build\sonoff_r4\libe2b\Adafruit BusIO\Adafruit_I2CDevice.cpp.o
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/simon/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\sonoff_r4\src\main.cpp.o] Error 1
==================================================================== [FAILED] Took 47.22 seconds ====================================================================

Environment    Status    Duration
-------------  --------  ------------
sonoff_r4      FAILED    00:00:47.218
=============================================================== 1 failed, 0 succeeded in 00:00:47.218 ===============================================================

 *  The terminal process "C:\Users\simon\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'sonoff_r4'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

 Below is my PlatformIO.ini file:

`[env:sonoff_r4]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino

upload_speed = 921600
monitor_speed = 115200

; Usa di default il filesystem LittleFS
board_build.filesystem = littlefs

build_flags =
  -DRELAY_MODE
  -DESP_PLATFORM_ESP32C3

lib_deps =
  bblanchon/ArduinoJson @ ^7.4.1
  esp32async/AsyncTCP @ ^3.4.6
  adafruit/Adafruit AHTX0 @ ^2.0.5
  adafruit/Adafruit BMP280 Library @ ^2.6.8
  ../libs/TcpIpServerClient/`

 Thank you for your help.

SimoDolcio, 2025-08-04:
I fixed it by changing the platform from platform = espressif32 to platform = espressif32 @ 6.12.0. I don't understand why the latest release isn't downloaded automatically.

valeros, 2025-08-04:
Closing, looks like a corrupted package.