platformio/platform-espressif32 #1631: Compiling unity unit tests with `-Wpedantic` fails after upgrade to `platform = espressif32@6.12.0`

grisharav, 2025-08-04:
After upgrading to `platform = espressif32@6.12.0` compilation of Unit Tests fails with:
```
In file included from .pio/libdeps/gpt_rev2/Unity/src/unity_internals.h:308,
                 from .pio/libdeps/gpt_rev2/Unity/src/unity.h:21,
                 from test/test_on_board/src/upload_fw_utils_tests.cpp:1:
/home/greg/.platformio/packages/framework-espidf/components/newlib/platform_include/stdio.h:10:2: error: #include_next is a GCC extension [-Werror]
   10 | #include_next <stdio.h>
      |  ^~~~~~~~~~~~
```
due to the pedantic flag, seems like the platformio is not using system include directive for including unity…
Is there a way to either disable `pedantic` only for unit tests or some other solution?

valeros, 2025-08-04:
Hi @grisharav, could you please put together a simple project to reproduce the issue?

grisharav, 2025-08-04:
It's very trivial, attached. Just `#include <unity.h>` causes errors:
```
In file included from .pio/build/pico32/unity_config/unity_config.c:37:
C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/newlib/platform_include/stdio.h:10:2: error: #include_next is a GCC extension [-Werro ]
   10 | #include_next <stdio.h>
      |  ^~~~~~~~~~~~
.pio/build/pico32/unity_config/unity_config.c: In function 'unityOutputChar':
.pio/build/pico32/unity_config/unity_config.c:39:48: error: conversion to 'int' from 'unsigned int' may change the sign of the result [-Werror=sign-conversion]
   39 | void unityOutputChar(unsigned int c) { putchar(c); }
      |                                                ^
cc1.exe: all warnings being treated as errors
```

when compiling with 
```
build_src_flags =
    -Wpedantic
```

[TestProj.zip](https://github.com/user-attachments/files/21577065/TestProj.zip)

valeros, 2025-08-04:
I see. So even if we fix the second `sign-conversion` error, the first issue with the `GCC extension` is out of our control as it comes from the framework itself. As a workaround you can remove troublesome flags solely for the `unity_config.c` file via an extra script, something like this:

```ini
[env:pico32]
platform = espressif32
board = pico32
framework = espidf
extra_scripts = pre:disable_warnings_as_errors_for_unity_config.py
...
```
where the `disable_warnings_as_errors_for_unity_config.py` file is the following:

```python
Import("env")

def disable_warnings_as_errors_for_unity_config(env, node):

    if "unity_config" not in node.name:
        return node
    
    return env.Object(
        node,
        CCFLAGS=[
            f for f in env["CCFLAGS"] if f not in ("-Werror", "-Wpedantic")
        ]
    )[0]

env.AddBuildMiddleware(disable_warnings_as_errors_for_unity_config, "**/unity_config.c")

```

closed 2026-10-20 by valeros