platformio/platform-espressif32 #1632: tool-esptoolpy@2.40900.0 unconditionally imports intelhex, causing ModuleNotFoundError for ESP32 builds

ryandtracy, 2025-08-04:
Summary
After updating to platform = espressif32@6.12.0, PlatformIO installs tool-esptoolpy@2.40900.0. This version of esptool.py contains the following unconditional import:

    from intelhex import HexRecordError, IntelHex

This causes builds to fail with:

    ModuleNotFoundError: No module named 'intelhex'

—even when the project doesn't use .hex files and doesn't require intelhex.

Environment Details
- PlatformIO Core: 6.1.18
- Platform: espressif32@6.12.0
- Tool: tool-esptoolpy@2.40900.0
- Python used by PlatformIO: "~\Local\Programs\Python\Python311\python.exe"
- Host OS: Windows 10 (Git Bash + VSCode)
- Framework: Arduino
- Board: nodemcu-32s

How to Reproduce
1. Use a basic PlatformIO platformio.ini:
   
   platform = espressif32@6.12.0
   board = nodemcu-32s
   framework = arduino

2. Ensure intelhex is not installed in the Python environment used by PlatformIO

3. Run platformio run or call a build script

4. Observe this crash:

   File "...tool-esptoolpy\esptool\bin_image.py", line 16, in <module>
       from intelhex import HexRecordError, IntelHex
   ModuleNotFoundError: No module named 'intelhex'

Expected Behavior
- If .hex functionality is unused, intelhex should not be required
- Builds should succeed without forcing intelhex installation

Actual Behavior
- Even .bin-only ESP32 builds fail due to an import in bin_image.py that is not guarded by a try-except

Suggested Fixes
1. Move the from intelhex import ... inside the function that needs it  
2. Or wrap in a try-except ImportError block and emit a clear message if .hex usage is triggered  
3. Or add intelhex to tool-esptoolpy's install-time requirements

Workaround
Manually install intelhex into the Python environment used by PlatformIO:

    "~\Local\Programs\Python\Python311\python.exe" -m pip install intelhex

Impact
This silently breaks ESP32 builds for users after updating to espressif32@6.12.0, even for projects that don't need .hex support. The crash is non-obvious and unrelated to the user's code or libraries.

rotorman, 2025-08-05:
I can confirm this to be an issue as listed above (tested building https://github.com/ExpressLRS/ExpressLRS/tree/master/src )

BlueAndi, 2025-08-07:
> [@rotorman](https://github.com/rotorman) So the error is the same? Could you try removing the `tool-esptoolpy` package manually? It should be somewhere here `c:\Users\YOUR_USER\.platformio\packages\tool-esptoolpy` or `/home/YOUR_USER/.platformio/packages/tool-esptoolpy` if you're on Linux. If possible, please also attach the logs of the first run after removing esptoolpy.

Removing the `tool-esptoolpy` was the solution for me as well.

sonu27, 2025-08-10:
I had the same issue. Went back to 6.11

rotorman, 2025-08-10:
Works fine now with v6.12 for me.

vidplace7, 2025-08-20:
We are still experiencing this issue at the Meshtastic project, when using esptool inside an `extra_script`.
```py
    # IntelHex workaround, remove after fixed upstream
    try:
        import intelhex
    except ImportError:
        env.Execute("$PYTHONEXE -m pip install intelhex")
    # /IntelHex workaround
    import esptool
```
^ This workaround is functional but... nasty

gbiervliet, 2025-11-04:
> 6.12, same problem here, worked after removing tool-esptoolpy folder.

Wow this worked for me! FINALLY, Thnx!

HydroPage, 2025-11-17:
Can confirm deleting the `tool-esptoolpy` folder immediately solved the problem. Attempting to upload again claimed to install it back again, but probably a correct version, and uploading worked again. I wonder why it was happening in the first place, though...

Giddy-Up224, 2026-02-06:
> [@rotorman](https://github.com/rotorman) So the error is the same? Could you try removing the `tool-esptoolpy` package manually? It should be somewhere here [...]

But why isn't the update process taking care of this issue? It can obviously pull the correct package if we force it to download a fresh copy.

hpsaturn, 2026-05-03:
I had the same error some days ago in my local machine, and I fixed it installing the missing package using the penv of PlatformIO. But today I notice that one of my GitHub workflows has the same problem. I try to do the same fix but nothing. Then I also notice that other of my workflows has not this issue, and the workflow is the same, so weird.

That means that something into the project trigger this error. And yes, **I changed the order of compiling on pio ini**, first esp32, then other esp32 like c3, s3, s2 and at the end esp8266 and atmel, like the another project that was working with the same workflow. Before, I had first the esp8266 env for build, and with that was happening the issue.

<details>
  <summary>My workflow:</summary>
  
```yml
name: PlatformIO

on:
  pull_request:
    branches:
      - devel
    paths-ignore:
      - '**/README.md'

jobs:
  build:
    runs-on: ${{ matrix.os }}
    strategy:
      max-parallel: 1
      matrix:
        os: [ubuntu-latest]
    steps:
      - uses: actions/checkout@v4
      - uses: actions/cache@v3
        with:
          path: |
            ~/.cache/pip
            ~/.platformio/.cache
          key: ${{ runner.os }}-pio
      - name: Set up Python
        uses: actions/setup-python@v5
        with:
          python-version: '3.10'
      - name: Install dependencies
        run: |
          python -m pip install --upgrade pip
          pip install -U platformio
          pio pkg update
      - name: Testing All Architectures
        run: |
          pio run
```
</details>

<details>
  <summary>My PlatformIO ini:</summary>
  
```ini
; CanAirIO Sensorlib
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = ./examples/
lib_dir = ./
extra_configs = unified-lib-deps.ini

[env]
framework = arduino
upload_speed = 1500000
monitor_speed = 115200
monitor_filters = 
	esp32_exception_decoder
	time
lib_deps = ${commonlibs.lib_deps}
build_flags = 
	-D CORE_DEBUG_LEVEL=1
	-D ARDUINO_ESP32_DEV=1
	-Wall
	-Wextra

[common]
framework = ${env.framework}
upload_speed = ${env.upload_speed}
monitor_speed = ${env.monitor_speed}
build_flags = ${env.build_flags}
lib_deps = ${env.lib_deps}

[esp32_common]
extends = common
platform = espressif32
board = esp32dev

[esp8266_common]
extends = common
platform = espressif8266
board = esp12e

[atmelsam_common]
extends = common
platform = atmelsam
board = seeed_wio_terminal

[env:esp32]
extends = esp32_common
build_src_filter = -<*> +<advanced_multivariable/>
build_flags = 
	${env.build_flags}

[env:esp32c3]
extends = esp32_common
board = esp32-c3-devkitm-1
build_src_filter = -<*> +<advanced_multivariable/>

[env:esp32s2]
extends = esp32_common
board = lolin_s2_mini
build_src_filter = -<*> +<advanced_multivariable/>
build_flags =
	${env.build_flags}
	-DSLIB_I2C_SDA=8
	-DSLIB_I2C_SCL=9
	-DARDUINO_USB_CDC_ON_BOOT=1

[env:esp32s3]
extends = esp32_common
board = seeed_xiao_esp32s3
build_src_filter = -<*> +<advanced_multivariable/>
build_flags = 
	${env.build_flags}
	-DSLIB_I2C_SDA=5
	-DSLIB_I2C_SCL=6
	-DBOARD_HAS_PSRAM=1
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1

[env:esp8266]
extends = esp8266_common
build_src_filter = -<*> +<advanced_multivariable/>
build_flags =
	-D CORE_DEBUG_LEVEL=1
	-D ARDUINO_ESP8266_ESP12E=1
	-D INCLUDE_SOFTWARE_SERIAL=1

[env:atmelsam]
extends = atmelsam_common
build_src_filter = -<*> +<wio_terminal_basic/>
```

</details>