platformio/platform-espressif32 #1633: platform  espressif32 6.12.0 fails to configure under macOS

kscheff, 2025-08-04:
I had the same issue when trying to get the end-idf framework 5.4.2 going. Now with the official release of 6.12.0 including framework esp-idf 5.5.0 I am getting this error again, se log below for details.

I already cleared out the ~/.platformio/packages and ~/.platformio/platforms

This version here works installs and works fine.
platform = https://github.com/platformio/platform-espressif32.git#5e0a22e

@6.11.0 installs also fine, but does not work for me due to some new missing framework APIs (5.4.2)

```
Resolving esp32-c3-XXX dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing esp32-c3-XXX (platform: espressif32@6.12.0; board: esp32-c3-XXX; framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-XXX.html
PLATFORM: Espressif 32 (6.12.0) > XXX-24
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50500.0 (5.5.0) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 0.0.1+20241011 
 - tool-esptoolpy @ 2.40900.0 (4.9.0) 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-riscv32-esp @ 14.2.0+20241119
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5 (Apple Git-154)")
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /Volumes)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Minimal build - OFF
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Volumes/SN770/Users/xxx/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Volumes/SN770/Users/xxx/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Volumes/SN770/Users/xxx/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any parent up to mount point /Volumes)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /Volumes/SN770/Users/xxx/.platformio/packages/framework-espidf/tools/cmake/build.cmake:629 (message):
  usage: __main__.py [-h] [--project_dir PROJECT_DIR]

                     [--interface_version {0,1,2,3}] [--lock_path LOCK_PATH]
                     {prepare_dependencies,inject_requirements,inject_requrements}
                     ...

  __main__.py: error: argument --interface_version: invalid choice: 4 (choose
  from 0, 1, 2, 3)

Call Stack (most recent call first):
  /Volumes/SN770/Users/xxx/.platformio/packages/framework-espidf/tools/cmake/project.cmake:740 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.70 seconds ==========================

Environment       Status    Duration
----------------  --------  ------------
esp32-c3-XXX  FAILED    00:00:01.704
==================== 1 failed, 0 succeeded in 00:00:01.704 ====================
```

Jason2866, 2025-08-05:
I had the same error. Delete the espidf venv. For me the location is in folder `.platformio/penv/...` The venv gets not updated (it should, but it did not in my case) and this error happens. When the venv gets fresh installed it works as expected.

puzrin, 2026-09-06:
Seems I found the cause of the `fatal: not a git repository` error.

**Short answer**: delete the generated SDK config to generate it from scratch (I hope you keep base settings in separate file used to generate that config)

**Long answer**

- I have generated configs in `.gitignore`. So, those are not in the repo, but exist on the disk (until you delete the project folder)
- With a new SDK version, config may fail with a cryptic message
- When you try to reproduce it, you do it on a fresh clone, without that generated config, and it's created from scratch, without problems.

This happened only for SDK builds (not Arduino), and only when the SDK version changes (not investigated how much). Right now I get it when updating from 6.0.11 => 7.1.1.

@valeros, could you take a look? This seems to be the reason for multiple platform upgrade issues.