platformio/platform-espressif32 #1641: Upload Filesystem Image & Erase Flash using wrong port

mitra42, 2025-08-10:
Sorry if this is the wrong place, I'm not sure how code is split between Core and Home .

STR: 

* Plug in two devices to USB
* Select one of them as the port to use (in the bottom bar)
* Project Tasks -> Erase Flash  or Upload FIlesystem Image

Notice that Erase Flash still looks for a port and uses that. 

Why this matters ...... 
I was developing on two boards whose interaction I am working on. I was trying to set different initial configurations for testing , but my device kept looking like it was seeing a file system in a different state. It took me hours to figure out the problem wasn't in my code, but was due to the state being updated on the other device. 

I think Erase Flash & Upload FIlesystem Image should have the same behavior as the Upload from the bottom bar, i.e. use the selected port if there is one.   

System: Mac OSX 
Core: 6.1.8
Home: 3.4.4

Jason2866, 2025-08-10:
Same as #1582

valeros, 2025-08-11:
Duplicate #1582